betaflight/betaflight #7121: Matek F405 Sensor Alignment

Karl33to, 2018-11-24:
Flashed two identical quads, both with Matek F405 STD's and mounted in their default orientation (arrow pointing forwards) using the hex file: betaflight_4.0.0_MATEKF405-1248-2a748e73d.hex 

I noticed that the default sensor orientation was wrong on both of them, which required a -90 degree correction on the yaw axis.

Previously they were on v3.5.0 which required no orientation corrections.

Thanks,

diff
```

# version
# Betaflight / MATEKF405 (MKF4) 4.0.0 Nov 22 2018 / 13:34:20 (2a748e73d) MSP API: 1.41

board_name MATEKF405
manufacturer_id 

# name
name KPV - Proton I

# resources
resource MOTOR 1 C08
resource MOTOR 2 C06
resource MOTOR 3 C09
resource MOTOR 4 C07
resource MOTOR 5 NONE
resource MOTOR 6 NONE
resource MOTOR 7 NONE
resource SERIAL_TX 11 A08

# mixer

# servo

# servo mix

# feature
feature -TELEMETRY
feature AIRMODE

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -RX_SET
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CRASH_FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE

# beacon
beacon RX_LOST

# map
map TAER1234

# serial
serial 2 16384 115200 57600 0 115200
serial 30 2048 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1800 2100 0 0
aux 1 13 1 1300 2100 0 0
aux 2 35 2 1800 2100 0 0
aux 3 49 3 1300 2100 0 0

# adjrange

# rxrange

# vtx

# rxfail
rxfail 3 h

# master
set mag_hardware = NONE
set baro_hardware = NONE
set rssi_channel = 16
set blackbox_device = NONE
set dshot_idle_value = 300
set motor_pwm_protocol = DSHOT1200
set align_board_yaw = -90
set bat_capacity = 1500
set vbat_warning_cell_voltage = 34
set beeper_dshot_beacon_tone = 2
set yaw_motors_reversed = ON
set small_angle = 180
set pid_process_denom = 1
set osd_cap_alarm = 1500
set osd_tim1 = 2561
set osd_vbat_pos = 2529
set osd_rssi_pos = 33
set osd_tim_1_pos = 54
set osd_tim_2_pos = 55
set osd_flymode_pos = 333
set osd_throttle_pos = 225
set osd_vtx_channel_pos = 2049
set osd_ah_pos = 200
set osd_current_pos = 385
set osd_mah_drawn_pos = 2072
set osd_craft_name_pos = 2056
set osd_gps_speed_pos = 218
set osd_gps_lon_pos = 82
set osd_gps_lat_pos = 65
set osd_gps_sats_pos = 51
set osd_home_dir_pos = 302
set osd_home_dist_pos = 303
set osd_compass_bar_pos = 266
set osd_altitude_pos = 247
set osd_pid_roll_pos = 423
set osd_pid_pitch_pos = 455
set osd_pid_yaw_pos = 487
set osd_debug_pos = 1
set osd_power_pos = 321
set osd_pidrate_profile_pos = 345
set osd_warnings_pos = 2409
set osd_avg_cell_voltage_pos = 2552
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_battery_usage_pos = 2537
set osd_disarmed_pos = 2442
set osd_nheading_pos = 311
set osd_nvario_pos = 279
set osd_esc_tmp_pos = 82
set osd_esc_rpm_pos = 83
set osd_stat_tim_1 = ON
set osd_stat_tim_2 = OFF
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set usb_hid_cdc = ON

# profile
profile 0

set p_yaw = 40
set i_yaw = 120

# rateprofile
rateprofile 0

set roll_rc_rate = 121
set pitch_rc_rate = 121
set yaw_rc_rate = 121
set roll_expo = 17
set pitch_expo = 17
set yaw_expo = 17
set roll_srate = 50
set pitch_srate = 50
set yaw_srate = 50

# 
```

jflyper, 2018-11-24:
Prior to 4.0, gyro/acc alignment could be configured per different gyro/acc chip, which Matek (ab)used to adjust the alignment on different boards with different gyro/acc chips. With 4.0 and later, this "feature"  will be gone, and all boards with gyro/acc chip mounted off-aligned with default orientation of the board must post-flash configure the alignment. Your board is one of these.

closed 2018-11-24 by jflyper

robert-hoffmann, 2018-12-25:
Just flashed 4.0 on my matek f405 and set the yaw degrees to 270 as per the matek documentation

This made the quad flip ^^
So after some testing, 180/-180 seems to work fine ..though someone should tell matek to update their docs or/and make sure to have something in the release notes to make sure ppl don't have accidents ^^

Thanks