betaflight/betaflight #7139: Support multiple RSSI sources (FrSky redundancy)

bluehallu, 2018-11-27:
**Is your feature request related to a problem? Please describe.**
If I use two Frsky receivers in redundancy mode, there's no way for me to see the RSSI of the receiver in use. 

My quad has R-XSR (with f.port) and R9MM (without f.port). I can either choose no rssi source and I get the one from R-XSR via fport telemetry, or I can choose AUX 14 as RSSI source and then see the R9MM RSSI (This is what I'm currently doing)

Even then it's not super useful as you only see the R9MM RSSI when it's being used and because it's the slave, I only see it when the R-XSR is in failsafe, which is somewhat useful, but any time it reconnects to R-XSR the latest value is frozen in the screen.

**Describe the solution you'd like**

One of:

a) Ability to display multiple rssi sources on OSD
b) Ability to switch RSSI source on a switch
c) Ability to automatically observe two rssi sources and display whichever is higher

**Describe alternatives you've considered**

This could conceivably be worked around by injecting RSSI from TX on a channel, however the solution is not ideal and requires using 16 channels (with the added latency) if using at least one receiver that injects the rssi on channel 16.

**Additional context**

If you're in same situation as me, know that setting the rssi alarm in OSD to 48 is still useful as 49 is the default value you'll get at the beginning from the r-xsr channel 16 not being in use and thus reading at 1500. Any time you see any value different than 49 it came from R9.

etracer65, 2018-11-27:
This seems like it's something that's unlikely to be included in Betaflight as the number of users using redundant receiver configurations would be extremely small, making it unlikely to worth the development time and flash space usage. The iNav project might be a better fix for this as more people would likely use this config in long-range applications.

And the better solution would be for Frsky (in this case) to "pass through" the RSSI of the active receiver in the redundant configuration. Seems like all that would be needed is an improvement in the R-XSR firmware (FPORT in this case) to capture the RSSI from the slave and pass that instead of it's local RSSI.

albertmonzo, 2018-11-27:
@etracer65 While I agree that the use case is currently small, it is only going to increase as R9 is becoming more affordable. Everyone who had an R-XSR and bought the £30 offer in September is now looking at using it and the path of least resistance in terms of hardware changes is to just add the R9MM as I did. 

Even disregarding R9 and long range altogether, most of the newer receivers from FrSky do support redundancy and them being the most popular choice, I think it makes sense for Betaflight to fully support that use case. 

In terms of Frsky fixing this, it's not so simple. 

1. That would not be as desirable as I'd like to actually know what the status of both receivers is if possible
2. It would only work for a F.port master and a slave with rssi channel injection. Only a minority of users use F.port and some receivers don't support channel injection.

bluehallu, 2018-11-27:
@etracer65 While I agree that the use case is currently small, it is only going to increase as R9 is becoming more affordable. Everyone who had an R-XSR and bought the £30 offer in September is now looking at using it and the path of least resistance in terms of hardware changes is to just add the R9MM as I did.
Even disregarding R9 and long range altogether, most of the newer receivers from FrSky do support redundancy and them being the most popular choice, I think it makes sense for Betaflight to fully support that use case. Even from a safety perspective, it makes sense to encourage people to use redundant receivers.
In terms of Frsky fixing this, it's not so simple.
1. That would not be as desirable as I'd like to actually know what the status of both receivers is if possible
2. It would only work for a F.port master and a slave with rssi channel injection. Only a minority of users use F.port and some receivers don't support channel injection.
There's further use cases as well. You could choose two display two different RSSI sources from a single receiver (sbus frame drop and RSSI or RSSI and LQ on crossfire)

ur4qrh, 2018-12-03:
I confirm, a very necessary function, optimally see two levels of rssi on osd!

Szpero, 2018-12-05:
No matter how we call it we need at least 1 custom input which can be displayed on OSD.
Proposal:
- icon of your choice
- source - any input channel. Other ideas for sources?
- raw/scaled
- min/max - value range displayed on OSD related to scaled=on

This should fix multi RSSI/RX and RSSI/LQ problem.

Sm1thson, 2018-12-11:
For me this is the most significant part: 
"There's further use cases as well. You could choose two display two different RSSI sources from a single receiver (sbus frame drop and RSSI or RSSI and LQ on crossfire)" I'd love to be able to OSD frame drop and RSSI.

fuleo, 2019-01-01:
I'm having this problem as well. Would like to see this feature in the future too. I'm currently on R9M mini with XM+ (sbus redundancy). In mid-flight, it's hard to tell if I'm on XM+ or R9mini. Preferably with additional OSD features, e.g RSSI1 & RSSI2. My FC isn't F-port capable, thus I'm using SBUS.

Would be nice to have this setup in the future.
- XM+ broadcasting RSSI on channel 8
- R9mini broadcasting RSSI on default channel 16.

In betaflight, choose RSSI1 (Aux12 for R9Mini), choose RSSI2 (Aux4 for XM+).

tavdog, 2019-02-08:
I'd like to be able to have two analog RSSI inputs and display those two in the OSD.  For real failsafe conditions, telemetry will probably fail well before.

closed 2026-09-26 by blckmn