betaflight/betaflight #7144: Coretemperature alarm when coretemperature is negative.

jukkaaakko, 2018-11-27:
**Describe the bug**
"Core" alarm is showed when ESC is freezing cold.

**To Reproduce**
Fly in cold weather: https://youtu.be/2uDV9-pB9q8

**Expected behavior**
Alarm is not shown

**Additional context**
Unsigned int is used for temperature and that is causing the bug:
main/io/osd.c:            uint8_t coreTemperature = getCoreTemperatureCelsius();
main/io/osd.c:            if (osdWarnGetState(OSD_WARNING_CORE_TEMPERATURE) && coreTemperature >= osdConfig()->core_temp_alarm) {

mikeller, 2018-11-27:
@jukkaaakko: Please provide the information you were asked to when opening the issue.

etracer65, 2018-11-28:
This has already been fixed in 4.0

https://github.com/betaflight/betaflight/pull/6973

closed 2018-11-28 by etracer65