betaflight/betaflight #7147: KISSCC: Softserial on TLM pin not working

robertlacroix, 2018-11-28:
I’m not able to get softserial on KISSCC to work while it’s working fine on KISSFC.

It seems the TLM pin is actually connected to PB9, not PA13 as configured in the target.

I tried to configure that pin and define a timer, but unfortunately without luck.

```
diff --git a/src/main/target/KISSFC/target.c b/src/main/target/KISSFC/target.c
index ad90a65..9805798 100644
--- a/src/main/target/KISSFC/target.c
+++ b/src/main/target/KISSFC/target.c
@@ -37,6 +37,7 @@ const timerHardware_t timerHardware[USABLE_TIMER_CHANNEL_COUNT] = {
     DEF_TIM(TIM3,  CH1,  PA6,  TIM_USE_MOTOR,               0),
     DEF_TIM(TIM17, CH1,  PA7,  TIM_USE_MOTOR,               0),
     DEF_TIM(TIM16, CH1N, PA13, TIM_USE_PWM,                 0),
+    DEF_TIM(TIM4,  CH4,  PB9,  TIM_USE_PWM,                 0),
 #else
     DEF_TIM(TIM1,  CH2N, PB14, TIM_USE_MOTOR,               TIMER_OUTPUT_INVERTED),
     DEF_TIM(TIM8,  CH2N, PB0,  TIM_USE_MOTOR,               TIMER_OUTPUT_INVERTED),
diff --git a/src/main/target/KISSFC/target.h b/src/main/target/KISSFC/target.h
index c043b4a..c718dd6 100644
--- a/src/main/target/KISSFC/target.h
+++ b/src/main/target/KISSFC/target.h
@@ -82,7 +82,7 @@
 #define UART3_RX_PIN            PB11 // PB11 (AF7)
 
 #ifdef KISSCC
-#define SOFTSERIAL1_TX_PIN      PA13
+#define SOFTSERIAL1_TX_PIN      PB9
 #else
 #define SOFTSERIAL1_TX_PIN      PA13 // AUX1
 #define SOFTSERIAL2_TX_PIN      PA15 // ROLL
@@ -111,5 +111,9 @@
 #define TARGET_IO_PORTD         0xffff
 #define TARGET_IO_PORTF         (BIT(4))
 
+#ifdef KISSCC
+#define USABLE_TIMER_CHANNEL_COUNT 12
+#else
 #define USABLE_TIMER_CHANNEL_COUNT 11
+#endif
 #define USED_TIMERS             (TIM_N(1) | TIM_N(2) | TIM_N(3) | TIM_N(4) | TIM_N(8) | TIM_N(15) | TIM_N(16) | TIM_N(17))
```

Diff - a similar config works fine on KISSFC

```
# diff
# Betaflight / KISSCC (KISS) 3.5.3 Nov 14 2018 / 23:51:28 (61d5abf00) MSP API: 1.40

board_name KISSCC
manufacturer_id 

feature SOFTSERIAL
feature TELEMETRY
feature AIRMODE
feature ESC_SENSOR
beacon RX_LOST
beacon RX_SET
map TAER1234
serial 0 32 115200 57600 0 115200
serial 2 1024 115200 57600 0 115200
serial 30 2048 115200 57600 0 115200
aux 0 0 3 1900 2100 0 0
aux 1 13 2 1900 2100 0 0
aux 2 35 1 1900 2100 0 0
set acc_hardware = NONE
set min_check = 1024
set max_check = 2000
set rc_smoothing_type = FILTER
set max_aux_channels = 4
set min_throttle = 1070
set dshot_idle_value = 300
set use_unsynced_pwm = OFF
set motor_pwm_protocol = DSHOT600
set motor_pwm_rate = 480
set vbat_max_cell_voltage = 44
set current_meter = ESC
set battery_meter = ESC
set beeper_dshot_beacon_tone = 3
set yaw_motors_reversed = ON
set deadband = 4
set yaw_deadband = 4
set pid_process_denom = 1
set frsky_vfas_precision = 1
set vtx_channel = 2
set vtx_freq = 5760
set vtx_pit_mode_freq = 5436
set esc_sensor_halfduplex = ON
profile 0

set dterm_notch_cutoff = 0
set anti_gravity_gain = 10000
set feedforward_transition = 33
set iterm_rotation = OFF
set smart_feedforward = ON
set iterm_relax = RPY
set p_pitch = 60
set i_pitch = 100
set d_pitch = 32
set f_pitch = 200
set p_roll = 55
set i_roll = 90
set d_roll = 32
set f_roll = 200
set p_yaw = 70
set i_yaw = 90
set f_yaw = 150
rateprofile 0

set thr_mid = 26
set thr_expo = 46
set roll_rc_rate = 50
set pitch_rc_rate = 50
set yaw_rc_rate = 55
set roll_srate = 77
set pitch_srate = 77
set yaw_srate = 65
```

mikeller, 2018-11-28:
@ronlix: Can you shed some light on this please? What are the correct pin assignments for KISSCC?

robertlacroix, 2018-11-29:
Or maybe @fedorcomander

ronlix, 2018-11-29:
Hi,

the RX TLM pin is directly connected to PA13 on the KISSCC which is AUX1 on the KISSFC v1

regards

Felix

jflyper, 2018-11-30:
PA13 and PB9 are on the same pin position but rotated 45 degrees.

Now, if TLM pad is actually connected to PA13, then it is not capable of being assigned to receiver side of software serial, as the timer channel `TIM16 CH1N` is INPUT ONLY. It does not work as half-duplex software serial either.

You may want to try configuring the pin just like KISSFC does.
```
    DEF_TIM(TIM4,  CH3,  PA13, TIM_USE_PWM,                 0), // On KISSFC TIM16 did not work, using TIM4 works
```

Hope an external inverter is not on the way.

mikeller, 2018-11-30:
@jflyper:

> as the timer channel TIM16 CH1N is INPUT ONLY

Are these things that we should get to be auto-detected for universal targets?

jflyper, 2018-11-30:
Ah...

Right now, N-channels are detected and `TIMER_OUTPUT_N_CHANNEL` flag is set by `TIM_DEF` macro automatically. For generic target, I believe `timer` command already support N-channels.

`TIMER_OUTPUT_INVERTED` is more problematic.  We should have another option for `timer` command (or somewhere else) that corresponds to the `TIMER_OUTPUT_INVERTED`.
(Presence of the external inverter is not detectable.)

mikeller, 2018-11-30:
@jflyper: Ah, All N-channel timers are input only? I wasn't aware of that. What I meant is would it make sense to prevent these from being assigned to functions that require output timers?

robertlacroix, 2018-11-30:
@jflyper thanks, that fixed it!

closed 2018-11-30 by robertlacroix

jflyper, 2018-11-30:
And software serial has such checks already ;)

https://github.com/betaflight/betaflight/blob/8a4ea0785e09d7cd8c1a7bde9a3ffe0b7aa8565c/src/main/drivers/serial_softserial.c#L251-L257

https://github.com/betaflight/betaflight/blob/8a4ea0785e09d7cd8c1a7bde9a3ffe0b7aa8565c/src/main/drivers/serial_softserial.c#L264-L268

jflyper, 2018-11-30:
Other users of timer input I can think of are PPM and PWM. Any others?

mikeller, 2018-12-01:
@jflyper: Does sonar use timers? If so then sonar would be another case of an input timer.

PR #7149 (2018-12-27): KISSCC: Fix soft serial on TLM pad
Fixes #7147

_Legal disclaimer: I am making my contributions/submissions to this project solely in my personal capacity and am not conveying any rights to any intellectual property of any third parties._