betaflight/betaflight #7150: KISSCC: ESC telemetry not working

robertlacroix, 2018-11-30:
Similar problem to #7147, same configuration works fine on KISSFC, but not on KISSCC.

@ronlix are you aware of any differences between KISSFC and KISSCC regarding ESC telemetry?

```
# diff
# Betaflight / KISSCC (KISS) 3.5.3 Nov 14 2018 / 23:51:28 (61d5abf00) MSP API: 1.40

board_name KISSCC
manufacturer_id 

feature SOFTSERIAL
feature TELEMETRY
feature AIRMODE
feature ESC_SENSOR
beacon RX_LOST
beacon RX_SET
map TAER1234
serial 0 32 115200 57600 0 115200
serial 2 1024 115200 57600 0 115200
serial 30 2048 115200 57600 0 115200
aux 0 0 3 1900 2100 0 0
aux 1 13 2 1900 2100 0 0
aux 2 35 1 1900 2100 0 0
set acc_hardware = NONE
set min_check = 1024
set max_check = 2000
set rc_smoothing_type = FILTER
set max_aux_channels = 4
set min_throttle = 1070
set dshot_idle_value = 300
set use_unsynced_pwm = OFF
set motor_pwm_protocol = DSHOT600
set motor_pwm_rate = 480
set vbat_max_cell_voltage = 44
set current_meter = ESC
set battery_meter = ESC
set beeper_dshot_beacon_tone = 3
set yaw_motors_reversed = ON
set deadband = 4
set yaw_deadband = 4
set pid_process_denom = 1
set frsky_vfas_precision = 1
set vtx_channel = 2
set vtx_freq = 5760
set vtx_pit_mode_freq = 5436
set esc_sensor_halfduplex = ON
```

mikeller, 2018-12-01:
@robertlacroix:

> set esc_sensor_halfduplex = ON

After looking at #7149, and realising that the UART used for ESC telemetry is a SOFTserial, I am deeply wondering why we ever implemented this, and not just switched   `PA13` to be `SOFTSERIAL1_RX_PIN`?
Can you try this?

(And yes, `esc_sensor_halfduplex` was implemented specifically to support KISSFC...)

robertlacroix, 2018-12-01:
@mikeller hmm, I thought the ESC sensor is connected to UART3, not softserial on PA13, isn’t it? At least that’s the case on KISSFC

robertlacroix, 2018-12-01:
The TLM pad on KISSCC is labeled TLM because it’s for telemetry to the receiver, not from the ESC.

mikeller, 2018-12-01:
@robertlacroix: Lol, you're right, I guess looking at the two tickets one after another confused me.

robertlacroix, 2018-12-26:
@ronlix are there any differences between KISSCC and KISSFC in terms of ESC telemetry? Is it wired to UART3 as well? What pin?

closed 2019-02-04 by stale[bot]