betaflight/betaflight #7153: Crazybeef3fr RX constant failsafe, disconnect/reconnect when connected to Transmitter and Betaflight-configurator.

stefanhra, 2018-12-01:
On my Mobula 7 which uses a "Crazybee f3 pro fr"  flight controller any time I'm connected to Betaflight-configurator and have my transmitter on I get constant "telemetry lost, telemetry recovered" warnings. In addition the bind light on the quad (which should remain on solid when connected to the transmitter) flashes on and off sporadically and the channel values in the receiver tab of Betaflight-configurator jump between 1500 (when disconnected) and approximately 1470 (when connected). So far I've had the same issue with the Betflight version 4.0.0 that came preinstalled on the quad and Betaflight 3.5.3 in both d8 and d16 modes.

To reproduce this issue connect Betaflight-configurator to the flight controller via USB and turn on FrSky transmitter that is bound to the RX on the flight controller. I am aware of at least one other person with the same setup who is having the same issue. 

```
# diff
# Betaflight / CRAZYBEEF3FR (CBFR) 4.0.0 Oct 13 2018 / 08:00:14 (3b479f92d) MSP API: 1.41

name Mobula7

feature -ANTI_GRAVITY
feature -DYNAMIC_FILTER
feature AIRMODE
serial 2 2048 115200 57600 0 115200
aux 0 0 0 1350 2100 0 0
aux 1 2 0 1450 1700 0 0
aux 2 35 1 1800 2100 0 0
rxfail 4 s 975
set rx_spi_protocol = FRSKY_D
set motor_pwm_protocol = DSHOT600
set vbat_warning_cell_voltage = 29
set ibata_scale = 1175
set osd_warn_esc_fail = OFF
set osd_vbat_pos = 2102
set osd_rssi_pos = 2083
set osd_tim_1_pos = 54
set osd_tim_2_pos = 2134
set osd_flymode_pos = 2456
set osd_throttle_pos = 2423
set osd_crosshairs_pos = 2253
set osd_current_pos = 2401
set osd_mah_drawn_pos = 2434
set osd_craft_name_pos = 2445
set osd_warnings_pos = 329
set osd_disarmed_pos = 2346
set vtx_band = 5
set vtx_channel = 2
set vtx_freq = 5695
set frsky_spi_tx_id = 145,110
set frsky_spi_offset = -46
set frsky_spi_bind_hop_data = 0,80,160,5,85,165,10,91,170,15,95,175,20,100,180,25,105,185,30,110,190,35,115,195,40,120,200,45,125,205,50,130,210,55,135,215,60,140,221,65,145,225,70,150,230,75,155,0,0,0
set frsky_x_rx_num = 4
profile 0

rateprofile 3
```
```
# resource list
IO
A00: ADC_BATT
A01: ADC_CURR
A02: MOTOR 4
A03: MOTOR 3
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: RX_SPI
A09: RX_BIND
A10: LED
A11: FREE
A12: FREE
A13: FREE
A14: FREE
A15: FREE
B00: FREE
B01: OSD_CS
B02: FREE
B03: LED 1
B04: FREE
B05: FREE
B06: FREE
B07: FREE
B08: MOTOR 1
B09: MOTOR 2
B10: SERIAL_TX 3
B11: FREE
B12: RX_SPI_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C13: GYRO_EXTI
C14: FREE
C15: BEEPER
F00: FREE
F01: FREE
F04: FREE
```
```
# dma list

DMA:
DMA1 Channel 1: ADC
DMA1 Channel 2: TIMUP 2
DMA1 Channel 3: FREE
DMA1 Channel 4: FREE
DMA1 Channel 5: FREE
DMA1 Channel 6: FREE
DMA1 Channel 7: FREE
DMA2 Channel 1: TIMUP 8
DMA2 Channel 2: FREE
DMA2 Channel 3: FREE
DMA2 Channel 4: FREE
DMA2 Channel 5: FREE
```

wally-kroeker, 2018-12-01:
I can confirm I have the same problem.   I tried changing versions as well but it did not help.

mikeller, 2018-12-01:
Telemetry is not fully supported for the FrSky SPI RX, and can lead to lock-ups of the firmware. I recommend disabling telemetry for this board - you still get values sent for RSSI and VBat, so the most important data is available.

etracer65, 2018-12-01:
Yes, this has happened for a while with this board - since 3.4 I think. The theory is that F3 processor is a little overloaded when communicating with the configurator and this interferes with the onboard receiver communication. Other than being annoying when connected to the configurator it doesn't seem to affect the board in flight.

stefanhra, 2018-12-01:
It might not have an effect on flying but it makes setting up modes a hassle and adjusting channel endpoints nearly impossible. Is there anyway to make it stop temporarily (disabling other feature, reducing loop time, etc.) while setting up the quad so you can actually make these adjustments?

closed 2019-01-08 by stale[bot]