betaflight/betaflight #7164: receiver and telemetry on same UART without diode/resistor adapter (4 wire)

smarq8, 2018-12-02:
Hello.
Is it possible to use flit10 telemetry at the same time with receiver on same UART port?
I just connect receiver TX (ibus) to fc RX6 and receiver RX(telemetry) to fc TX6 (4 wires).
And as I use Serial6 for receiver only then it work, same for telemetry. But if I configure Serial6 for receiver and telemetry then only telemetry work.
I know I can connect telemetry to other UART but i do not want it. 

BF version: OMNIBUSF4 3.5.3
board: flip32 2016
receiver: flit10

mikeller, 2018-12-03:
@smarq8: I think you are overthinking here. iBus is capable of supporting RX and telemetry on the same UART, and the UART only needs to be configured for iBus serial Rx for this. Have a look at the instructions on https://www.banggood.com/Flit10-2_4G-10CH-Micro-Telemetry-Flysky-Compatible-Ibus-Receiver-for-FS-I6X-FS-i6S-Turnigy-Evolution-p-1260238.html, and in particular the diagram https://img.staticbg.com/images/oaupload/banggood/images/6F/82/fb841cab-9a11-4cc4-b087-63a2c399657c.jpg.

smarq8, 2018-12-03:
The diagram you posted is for receiver only. In normal case to get working receiver and telemetry then you need to connect it via "telemetry module" to FC TX and I want to avoid it and use UART6RX for receiver and UART6TX for telemetry but according to site below then after turn on RX and telemetry on same port then it work only through TX and I am not sure how to use RX and TX of same port.

https://github.com/betaflight/betaflight/wiki/Single-wire-FlySky-(IBus)-telemetry

mikeller, 2018-12-04:
@smarq8: Yes, you are right. I would assume that this new RX incorporates the 'diode and resistor' Y connector shown in the diagram at the top of betaflight/betaflight/wiki/Single-wire-FlySky-(IBus)-telemetry directly on the board - you'll just have to connect the 'IBUS' pin from the FC to the TX pin of the UART, and then follow the instructions in betaflight/betaflight/wiki/Single-wire-FlySky-(IBus)-telemetry.

McGiverGim, 2018-12-04:
Yes, the additional board of the photo is the diode and resistor hack. With it you only need to connect the board to a tx.

jflyper, 2018-12-04:
I think @smarq8 doesn't have diodes and resistors but wires.

Logic inside `rx/ibus.c` suggest IBUS is capable of full-duplex telemetry. It doesn't work?

smarq8, 2018-12-08:
> Logic inside `rx/ibus.c` suggest IBUS is capable of full-duplex telemetry. It doesn't work?

then how to properly configure it?

Eganwp, 2018-12-12:
If you’re using OpenTx, this is the whole purpose of F.port protocol instead of using SBUS. Works great and reduces latency as well.

jflyper, 2018-12-12:
Does any of Flysky RX support F.Port?

mikeller, 2018-12-13:
@jflyper: Not that I know of.

closed 2019-01-20 by stale[bot]