betaflight/betaflight #7165: Testing RSSI over MSP

voodoo614, 2018-12-02:
I am trying to test the RSSI over MSP that was added to version 3.3.  I have not been able to get it to work.
https://github.com/betaflight/betaflight/pull/4507

I know that I when I send RSSI back to FC on a channel, it works.  When I disable RSSI channel and turn off RSSI_ADC, I am not getting any RSSI. I know my S.Port is properly setup because I get all my telemetry on my radio.

hardware
CL Racing F4S
XSR receiver

```
# version
# Betaflight / CLRACINGF4 (CLR4) 3.5.3 Nov 14 2018 / 23:17:07 (61d5abf00) MSP API: 1.40

board_name CLRACINGF4
manufacturer_id 

# name
name -

# resources
resource BEEPER 1 B04
resource MOTOR 1 B00
resource MOTOR 2 B01
resource MOTOR 3 A03
resource MOTOR 4 A02
resource MOTOR 5 B08
resource MOTOR 6 NONE
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 NONE
resource PWM 1 NONE
resource PWM 2 NONE
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource LED_STRIP 1 B08
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 NONE
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 A00
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 C06
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 NONE
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 A01
resource SERIAL_RX 5 NONE
resource SERIAL_RX 6 C07
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource SERIAL_RX 11 NONE
resource INVERTER 1 C00
[... 692 lines cut ...]
set abs_control_gain = 0
set abs_control_limit = 90
set abs_control_error_limit = 20

# rateprofile
rateprofile 0

set thr_mid = 50
set thr_expo = 0
set rates_type = BETAFLIGHT
set roll_rc_rate = 100
set pitch_rc_rate = 100
set yaw_rc_rate = 100
set roll_expo = 0
set pitch_expo = 0
set yaw_expo = 0
set roll_srate = 70
set pitch_srate = 70
set yaw_srate = 70
set tpa_rate = 10
set tpa_breakpoint = 1650
set throttle_limit_type = OFF
set throttle_limit_percent = 100

```

etracer65, 2018-12-02:
Smartport telemetry doesn't supply RSSI to the flight controller. If you want that you'll need to update the firmware on your receiver and use F.PORT.

voodoo614, 2018-12-03:
I thought the point of RSSI over MSP is for this exact reason. To send RSSI back through MSP (Smartport). I am already aware the F. Port send RSSI through their packet. Or I can use an AUX channel to send RSSI back to the FC.

https://github.com/betaflight/betaflight-tx-lua-scripts/issues/89

mikeller, 2018-12-03:
@voodoo614: Yes, this should work. What version of the lua scripts do you have installed, and do you get values displayed for the settings when running the lua script? Also, what firmware do you use on your RX? Can you try updating to the latest version from https://www.frsky-rc.com/xsr/?

voodoo614, 2018-12-03:
I am not running Lua Script. I didn't think it is a requirement, because Lua Script has to be activated and isn't constantly running.

mikeller, 2018-12-03:
@voodoo614: The lua script runs in the background to make this possible. It won't work without the lua script sending the RSSI value.

voodoo614, 2018-12-03:
Thanks a lot. That clears up things. I'll close the issue.

closed 2018-12-03 by voodoo614