betaflight/betaflight #7171: Omnibus f7 v2 using more than four pwm ports

Doven007, 2018-12-04:
Omnibus f7 v2 using more than four pwm ports

Hi all, I bought an omnibus f7 v2 from unmanned tech and was planning to put it on a fixed wing through inav which requires 6 pwm ports, however upon checking the pinout it has 4 on the board, and another 4 for a 4 in one esc, is there any way to access another two other pwm outputs on the board. Thanks

A clear and concise description of any alternative solutions or features you've considered.
[image]

mikeller, 2018-12-04:
@Doven007:
- this is the Betaflight GitHub - our knowledge with respect to what iNav can do are limited, better open an issue here: https://github.com/iNavFlight/inav/issues;
- the OMNIBUSF7V2 has only 4 motor / servo outputs 'out of the box' - the ones on the PCB and the ones in the connector are connected in parallel;
- in Betaflight, you could remap the resources for the UART3 or UART6 pins to become additional motor / servo pins. But iNav does not implement resource remapping, so you'll have to build your own target for it.

Hope this helps.

Doven007, 2018-12-04:
OK,I understand. Thank you very much.

closed 2019-01-10 by stale[bot]