betaflight/betaflight #7173: Suggestion: Discharge Mode

StillVoidingWarranties, 2018-12-04:
If you fly with your friends and end up using only half your batteries, it would be nice being able to discharge them in the field to storage voltage without having to use a dedicated device.

This could be done by using the motors of a multicopter. Here is how it could work:
There is a new mode, discharge mode. It can only be activated if the model type is multicopter and the ESC type is not brushed. When acivated arming the copter will spin all motors in reverse at 10%, pushing the copter to the ground. As soon as the battery voltage drops below a given threshold, e.g. 3.75V per cell, the copter disarms and the buzzer/beacon is activated.

I also suggest automatically putting the VTX in pit mode if Tramp/Smartaudio if available. During discharge the copter could balance horizontally to avoid tipping over (requires accelerometer). RX signal loss must result in immediate disarm, not any other failsafe procedure.

jflyper, 2018-12-04:
Why are you putting your battery to storage voltage? Was it the last flight of the season?

mikeller, 2018-12-04:
Also, I can't imagine many ways to spend time that are worse than sitting there, doing nothing but watching your craft idle its battery down to storage charge...

StillVoidingWarranties, 2018-12-04:
> Why are you putting your battery to storage voltage? Was it the last flight of the season?

Some do, some don't. Bardwell suggested the stress of a full charge on a lipo is accumulative so the longer the lipo is at full charge the worse. If that really matters is a different story but there are pilots out there who always discharge to storage voltage, even if it is only for one day.

SchonRichtig, 2018-12-05:
Yes, leaving a LIPO full charged for a longer time will reduce the capacity and the internal resistance
will go up, ant thats forever.
As said by Mike, you are faster flying your Lipos empty, than observe your quad for 20 minutes pressing
itself to the ground :) And you have more fun.
For the lipos that i can`t fly bebause i have too little time, i use "DALRC USB Quick Charger 36W Supporting QC3.0/FCP/PE2.0 "
You can set the Endvoltage in this Adapter, than it stops.
Its an XT60 to USB Adapter.
If i have more Liops full, i put them an the paralel board and put this Adapter on.
So i can charge my Phone, Tablet,... and not wasting the energy into wind :)

mikeller, 2018-12-05:
@SchonRichtig:

> So i can charge my Phone, Tablet,... and not wasting the energy into wind :)

Wow, that is actually a very good point - it's probably not a lot of power you save overall, but when it comes to doing the right thing, it's the thought that counts.

SchonRichtig, 2018-12-05:
No, it isn't much energy.
but it is relative fast and i have a better feeling (when i start my car and wait for the windows thawing)

closed 2019-01-11 by stale[bot]