betaflight/betaflight #7192: Motors twitching before arming. Not happening in 3.5.x

skorzo, 2018-12-09:
**Describe the bug**
Motors twitch randomly and slightly before arming the quadcopter

Please see this video:
https://photos.app.goo.gl/K2TVmPptW5wZVh7N7

**To Reproduce**

Just flash Betaflight 4.0.0  # 1276 and set your basic parameters (see below my diff)

Connect the lipo and watch the motors. Do not arm, just wait for it.

It does not happen always, but sometimes does.
Not happening at all on 3.x.x  and, at least, not seen on Betaflight 4.0.0 pre - # 1276

**Expected behavior**
Motors should not move at all after boot.

**Flight controller configuration**

F3 XRacer SPI  Betaflight 4.0.0  # 1276
G-L-30 ESCs , BLHeli 16.4  (not updated due to this issue https://github.com/bitdump/BLHeli/issues/213 ) but anyway, this same build is not twitching in 3.x.x

Multishot protocol

```defaults nosave

feature -RX_PPM
feature RX_SERIAL
feature SOFTSERIAL
feature TELEMETRY
feature AIRMODE
serial 1 32 115200 57600 0 115200
serial 2 64 115200 57600 0 115200
serial 30 2048 115200 57600 0 115200
aux 0 0 1 1675 2100 0 0
aux 1 1 2 900 1200 0 0
aux 2 13 0 1750 2100 0 0
aux 3 47 2 1275 1700 0 0
adjrange 0 0 3 900 2100 12 3 0 0
set acc_calibration = -10,103,-76
set rc_interp_ch = RPY

set serialrx_provider = SBUS
set min_throttle = 1050
set motor_pwm_protocol = MULTISHOT
set small_angle = 180
set vtx_band = 5
set vtx_channel = 7
set vtx_power = 3
set vtx_low_power_disarm = ON
set vtx_freq = 5880

profile 0

rateprofile 0

set roll_rc_rate = 120
set pitch_rc_rate = 120
set yaw_rc_rate = 90
set roll_expo = 10
set pitch_expo = 10
set yaw_expo = 10
set roll_srate = 76
set pitch_srate = 78
set yaw_srate = 82
set tpa_rate = 15
rateprofile 1

set roll_rc_rate = 90
set pitch_rc_rate = 90
set yaw_rc_rate = 90
set roll_expo = 20
set pitch_expo = 20
set yaw_expo = 20
set roll_srate = 82
set pitch_srate = 82
set yaw_srate = 82
set tpa_rate = 20
rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

rateprofile 0
save
#
```

**Setup / Versions**
Original XRacer F3 SPI
Frsky XSR Sbus

**Additional context**
Same build and same DIFF has never twitched, for years, in Beta 3.x.x

etracer65, 2018-12-10:
Common problem that happens with multishot on all versions. Try setting `min_command = 950`. It's just caused by jitter in the multishot signal when min_command is very close to motor start. It's likely you have more CPU load on the F3 now with 4.0 and this is leading to more jitter.

skorzo, 2018-12-10:
Thank you, it makes sense, I will have a try

closed 2019-01-17 by stale[bot]