betaflight/betaflight #7194: Wrong units in GPS Altitude telemetry

netzwergo, 2018-12-10:
I set up a Beitan GPS BN-220 with my Betaflight quad.
FC is the Bardwell F4 V2 with the CLRACINGF4 Betaflight target. Receiver is a FrSky R-XSR.

I am getting a GPS altitude reading of 67XX m at my location. Real altitude at my location is roughly 67 m above sea level. When I fly up approximately 30 meters, this changes to 97XX m altitude. Therefore I am reasonably sure that the GPS altitude reading the telemetry is giving out is wrong by a factor of 100. 
If divided by 100 or multiplied by 0.01 respectively, the altitude should be correct.

I have taken a photo of the taranis telemetry display here: https://i.ibb.co/QCCBkMn/IMG-1375-276.jpg

mikeller, 2018-12-10:
@netzwergo: Just to make sure I get this right:
- you get the wrong GPS altitude only in telemetry (and not in OSD for example);
- you use SmartPort for telemetry.

Can you try to delete all sensors in OpenTX and then discover new sensors, to make sure this isn't a faulty sensor setup?

netzwergo, 2018-12-10:
I am not sure about OSD, since this is on my LOS racer. Telemetry gives me wrong GPS altitude though.
Is there a way to read the GPS altitude from CLI or something, other than OSD or telemetry?

Yes, I am using Smartport telemetry. 
Smartport for telemetry and SBUS for RC signal.

I did delete all sensors and did a "discover new sensors". Same problem still persists, see photo: https://i.ibb.co/W0YGftv/IMG-1377-279.jpg

etracer65, 2018-12-10:
I'm working on getting a setup with GPS and smart port set up to test, but this looks questionable:

https://github.com/betaflight/betaflight/blob/6972f3eb0163a138dc0e147d720075408423f8ed/src/main/telemetry/smartport.c#L827

etracer65, 2018-12-10:
@netzwergo Can you try this fixed version please?

[betaflight_3.5.4_CLRACINGF4.hex.zip](https://github.com/betaflight/betaflight/files/2663566/betaflight_3.5.4_CLRACINGF4.hex.zip)

netzwergo, 2018-12-10:
@etracer65 

Thanks a lot, this has solved the problem.
Telemetry is showing the right value now: https://i.ibb.co/0czf9SQ/IMG-1380-286.jpg

PR #7200 (2018-12-12): Fix smartport GPS altitude
Fixes #7194 

Value sent was erroneously scaled up by a factor of 10.

Note that this can't be directly applied to 3.5 maintenance as that version has a different bug.
closed 2018-12-12 by mikeller