betaflight/betaflight #7196: Disabling and re-enabling an element in the OSD Active Elements menu moves it one position left

pkruger, 2018-12-10:
**Describe the bug**
If an OSD element is disabled (i.e. made invisible) and then re-enabled in the OSD element menu on the OSD, the element is visible again, but it shifted one position to the left.

**To Reproduce**
1. Disable any OSD element in the Active elements menu on the OSD.
2. Exit all the OSD menus and return to the main OSD screen (with no menu.)
3. Re-enable (change visible to Yes) for the OSD element in the OSD Active Elements menu.
4. Upon returning to the main OSD screen the element is visible, but shifted one position to the left.

**Expected behavior**
The OSD element should be visible again (after being disabled), but in the same position it was in before.

**Flight controller configuration**
# Betaflight / CLRACINGF4 (CLR4) 4.0.0 Dec 10 2018 / 10:02:41 (ef459cb) MSP API: 1.41

board_name CLRACINGF4
manufacturer_id 

# name
name @Witblits@

# resources
resource MOTOR 5 NONE
resource LED_STRIP 1 B09
resource CAMERA_CONTROL 1 B08
resource SPI_PREINIT_IPU 2 B12
resource SPI_PREINIT_IPU 3 B03
resource SPI_PREINIT_IPU 4 A15
resource SPI_PREINIT_IPU 5 NONE

# mixer

# servo

# servo mix

# feature
feature MOTOR_STOP
feature TELEMETRY

# beeper
beeper -DISARMING
beeper -ARMING_GPS_FIX
beeper -BAT_LOW
beeper -RX_SET
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE

# beacon
beacon RX_LOST
beacon RX_SET

# map
map TAER1234

# serial
serial 0 64 115200 57600 0 115200
serial 2 32 115200 57600 0 115200
serial 3 8192 115200 57600 0 115200

# led

# color

# mode_color
mode_color 5 0 9
mode_color 5 1 11
mode_color 5 2 2
mode_color 5 3 13
mode_color 5 4 10
mode_color 5 5 3

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 1 900 1275 0 0
aux 2 2 1 1300 1700 0 0
aux 3 27 7 1700 2100 0 0
aux 4 13 3 1700 2100 0 0
aux 5 26 7 1300 1700 0 0
aux 6 28 1 1300 2100 0 0
aux 7 35 2 900 1300 0 0

# adjrange
adjrange 0 2 4 900 2100 12 4 0 0
adjrange 4 0 6 900 2100 29 6 0 0

# rxrange

# vtx

# rxfail

# master
set gyro_lowpass_type = PT1
set gyro_lowpass_hz = 100
set gyro_lowpass2_hz = 300
set dyn_lpf_gyro_max_hz = 400
set acc_calibration = 202,-214,-388
set min_check = 1100
set rssi_channel = 10
set serialrx_provider = SBUS
set blackbox_device = SERIAL
set dshot_idle_value = 450
set motor_pwm_protocol = DSHOT600
set failsafe_delay = 10
set failsafe_switch_mode = STAGE2
set bat_capacity = 1500
set ibata_scale = 400
set beeper_frequency = 0
set beeper_dshot_beacon_tone = 2
set small_angle = 180
set runaway_takeoff_deactivate_throttle_percent = 25
set osd_warn_core_temp = OFF
set osd_cap_alarm = 2000
set osd_alt_alarm = 120
set osd_tim1 = 2561
set osd_vbat_pos = 14817
set osd_rssi_pos = 8641
set osd_tim_1_pos = 2551
set osd_tim_2_pos = 0
set osd_remaining_time_estimate_pos = 0
set osd_flymode_pos = 2049
set osd_anti_gravity_pos = 0
set osd_g_force_pos = 0
set osd_throttle_pos = 0
set osd_vtx_channel_pos = 2520
set osd_crosshairs_pos = 2285
set osd_ah_sbar_pos = 0
set osd_ah_pos = 0
set osd_current_pos = 0
set osd_mah_drawn_pos = 2072
set osd_craft_name_pos = 10730
set osd_gps_speed_pos = 1
set osd_gps_lon_pos = 1
set osd_gps_lat_pos = 1
set osd_gps_sats_pos = 1
set osd_home_dir_pos = 1
set osd_home_dist_pos = 1
set osd_compass_bar_pos = 0
set osd_altitude_pos = 1
set osd_pid_roll_pos = 0
set osd_pid_pitch_pos = 0
set osd_pid_yaw_pos = 0
set osd_debug_pos = 0
set osd_power_pos = 0
set osd_pidrate_profile_pos = 2106
set osd_warnings_pos = 2473
set osd_avg_cell_voltage_pos = 0
set osd_pit_ang_pos = 0
set osd_rol_ang_pos = 0
set osd_battery_usage_pos = 0
set osd_disarmed_pos = 2507
set osd_nheading_pos = 0
set osd_esc_tmp_pos = 0
set osd_esc_rpm_pos = 0
set osd_rtc_date_time_pos = 0
set osd_adjustment_range_pos = 0
set osd_core_temp_pos = 0
set osd_stick_overlay_left_pos = 2337
set osd_stick_overlay_right_pos = 2358
set osd_stat_rtc_date_time = ON
set osd_stat_tim_1 = ON
set osd_stat_tim_2 = OFF
set osd_stat_max_dist = ON
set vtx_channel = 3
set vtx_freq = 5780
set vcd_video_system = PAL
set camera_control_key_delay = 125
set usb_msc_pin_pullup = OFF
set rcdevice_init_dev_attempts = 4

# profile
profile 0

set dyn_lpf_dterm_max_hz = 200
set dterm_lowpass_type = PT1
set dterm_lowpass_hz = 100
set dterm_lowpass2_hz = 200
set anti_gravity_gain = 8000
set feedforward_transition = 5
set acc_limit_yaw = 100
set iterm_relax_type = GYRO
set iterm_relax_cutoff = 10
set iterm_windup = 40
set iterm_limit = 150
set i_pitch = 70
set d_pitch = 27
set i_roll = 80
set p_yaw = 55
set i_yaw = 45
set f_yaw = 60

# rateprofile
rateprofile 2

set roll_rc_rate = 91
set pitch_rc_rate = 91
set tpa_rate = 10
set tpa_breakpoint = 1650

**Setup / Versions**
 - CL Racing F4 (from Armattan)

**Additional context**
This bug can be fixed by changing the code below in cms.c (line 866):
#ifdef USE_OSD
        case OME_VISIBLE:
            if (p->data) {
                uint16_t *val = (uint16_t *)p->data;

                if (key == CMS_KEY_RIGHT)
                    *val |= OSD_PROFILE_1_FLAG;
                else
                    *val %= ~OSD_PROFILE_1_FLAG;
                SET_PRINTVALUE(p);
            }
            break;
#endif
The line  *val %= ~OSD_PROFILE_1_FLAG; should be changed to *val &= ~OSD_PROFILE_1_FLAG;

mikeller, 2018-12-10:
Well spotted. I wonder who put that `%` there. :-D

pkruger, 2018-12-11:
:o) I thought it was me, but I'm innocent this time...

closed 2018-12-11 by pkruger