betaflight/betaflight #7197: OSD Active Elements menu update for OSD Profiles

pkruger, 2018-12-10:
**Is your feature request related to a problem? Please describe.**
Now that OSD Profiles are available on master, the OSD Active elements menu don't really make sense, i.e. it only disables an item from being visible in OSD profile 1 and leaves the other OSD profiles unchanged.  If an item is made visible using the OSD Active elements menu, on which profile should it be visible?  

**Describe the solution you'd like**
The OSD Active elements menu should OSD Profile sensitive.  In other words, if OSD Profiles aren't available in the flight controller, it operates only on OSD Profile 1 (like it is now).  If OSD Profiles are available, it operates on the currently selected OSD Profile or provides a way of selecting the OSD Profile to be changed.

**Describe alternatives you've considered**
1. Removing the OSD Active elements menu completely - not useful.
2. The OSD Active elements menu makes an item visible only in OSD Profile 1, but makes the element 
    invisible in all 3 profiles.
3. The OSD Active elements menu should be updated so that an OSD element can be made invisible or 
    visible in any of the OSD Profiles if OSD Profiles is available on the flight controller.

**Additional context**
Happy to implement it, just need some idea of which of the above options people prefer.

mikeller, 2018-12-10:
I would prefer 3 - CLI is not a good option to configure the screen layout, especially if users want to do things like arrange items so that multiple elements share the same position on screen but are only visible in different profiles.

My suggestion for 'how' would be to ditch the 'visible' sliders in the OSD tab, and replace them with a matrix of checkboxes, with the verticals used for the different profiles.

McGiverGim, 2018-12-11:
I think it's simpler if the OSD menu affects the actual OSD profile. You can change the profile if you want modify the elements.

mikeller, 2018-12-11:
@McGiverGim: That would be an alternative - probably a little simpler, and easier to do with the UI widgets we already use. What I like about my proposal above is that it allows you to compare different profiles in an easy way.

McGiverGim, 2018-12-12:
I'm not too sure if I have understood it. If we talk about configurator, yes, better to have the checkboxes. But I thought that we were talking about the CMS menu (this thread is an issue in the firmware part, not the configurator part). I can't see an easy way to add tons of checkboxes in the CMS menu.

mikeller, 2018-12-12:
Sorry, yes, my bad - I misunderstood this to be configurator. For CMS, I agree with @McGiverGim, the best solution would be

'The OSD Active elements menu makes an item visible/invisible in the currently active OSD profile.'

pkruger, 2018-12-13:
Pull request: https://github.com/betaflight/betaflight/pull/7215

closed 2018-12-17 by mikeller