betaflight/betaflight #7202: Beta 10.4.0 CLI exit, reboot and reconnect USB FC not working

melenefoti, 2018-12-10:
# If your issue looks like a hardware fault or a configuration problem please don't raise an issue here.

## Please consider using other user support options such as asking the manufacturer of the hardware you are using, RCGroups: https://rcgroups.com/forums/showthread.php?t=2464844, or Slack (registration at https://slack.betaflight.com/) or other user support forums & groups (e.g. facebook).

## If you believe there is an issue with the firmware itself please follow these steps:
1. Describe your problem;
2. Include ways to reproduce the problem;
3. Provide information about your flightcontroller and other components including how they are connected/wired;
4. Add the used configuration and firmware version;
5. Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/); and
6. Remove this Text :).

jflyper, 2018-12-10:
Can you concisely state how to reproduce the issue using your expertise as a developer?

melenefoti, 2018-12-10:
jflyper i am deeply sorry.
I had used Zadig before i connect the FC board on betaflight, but Zadig was not enough. 
It was usb driver problem, i google and read carefully about KAKUTEF7. 
"ImpulseRC driver fixer" is fixing this issue. 

I am closing this issue.

closed 2018-12-10 by melenefoti