betaflight/betaflight #7206: Custom VTX power levels in OSD menu

kdamon, 2018-12-10:
I think it would be nice to be able to change the power levels shown in the OSD smart audio/tramp menu. For example I believe the OSD menu shows 25, 200, 400, 600 or similar. One VTX I own has the levels as 25, 200, 500, 800. Another has 25, 200, 600, 1000. Another has 25, 50, 100, 200. I understand that choosing the highest value in the menu (600) would set the VTX to the highest level (800, or 1000, or 200). It would just be nice to have the menu state what levels are available for that VTX instead of trying to remember for every different VTX.

Perhaps there could be a way to custom set the number of power levels as well as the values for each level. Even if it's just in CLI. There may already be a way that I haven't seen.

wind0r, 2018-12-10:
There allready is a similar issue and a discussion for this. Planed but currently no one is working on it. So if you want to get you hands dirty just start ;-)

kdamon, 2018-12-10:
This this the one you're referring to?
https://github.com/betaflight/betaflight/issues/1782

I might just have to start working on some code and see what I can or can't do lol.

wind0r, 2018-12-11:
some of the discussion was here: https://github.com/betaflight/betaflight/pull/4959
with all the different protocols and stuff we do not want to have those numbers hardcoded in the source code. in the future those settings should be added after inital flash / while setup

kdamon, 2018-12-11:
I agree the numbers shouldn't be hardcoded. An ideal setup would be in the CLI or a page/tab in the  configurator where you can set number of power levels (ie. 1-4), then set the value for each level (ie. VTX_power_1 = 25, VTX_power_2 = 50, etc). I want to start looking at the code and see what I can figure out.

jflyper, 2018-12-11:
The plan is to make everything related to VTX configurable, including frequency table, so that VTX suppliers will supply necessary parameters (CLI or PG in the future) with their products. Users can amend them if so wanted.

Related:
#6935 
#4959
#4181

Eganwp, 2018-12-12:
I’d suggest checkboxes in BFConfigurator so you can check the relevant ones your specific VTx has. There’s just too many combinations otherwise. For example one brand may have these which you’d place a checkmark beside when configuring BF:
- 25mW
- 200mW
- 500mW

wind0r, 2018-12-12:
How checkboxes should do the job when there are too many combinations otherwise?

Eganwp, 2018-12-12:
Sure, I’m all for a drop down with VTx combinations, however there are so many with the Chinese releasing their own combinations now on micro Vtx’s that I’m not sure if it’s more or less work to maintain a table of values for the drop down or just select them manually via checkbox. CLI is fine for most of us here, however most new people likely wouldn’t realize the options via CLI exist.

mikeller, 2018-12-13:
I am tempted to say we should go for something like using dBm values to represent (rounded) power levels:

https://www.l-com.com/multimedia/whitepapers/dbm_chart.pdf

jflyper, 2018-12-19:
Has somebody started to work on this?

mikeller, 2018-12-19:
@jflyper: Not yet. But I will be doing some work for a new proposal from TBS for supporting custom VTX power levels soonish.

jflyper, 2018-12-19:
Oh, okay... I've hacked a cli command parser (while I'm fresh with cli parsing lol) for vtx tables. No driver changes. I'll submit a PR in a couple of hours.

mikeller, 2018-12-19:
VTX (frequency) tables are all good, they are a separate thing from what I am going to be doing, which only covers VTX power levels.

jflyper, 2018-12-19:
Argh. It handles power level/label too 😅  #7251

closed 2019-01-25 by stale[bot]