betaflight/betaflight #7208: Soft serial for crazybee f3

turboed13b, 2018-12-11:
Trying to use smartaudio v1 on crazybee f3. Hardware uart 3 is not working with SA v1 so next step is soft serial. Soft serial is disabled on the crazybee is it possible to enable it?

mikeller, 2018-12-11:
@turboed13b: There are 2 versions of the CRAZYBEEF3 boards, the first one without the extra UART, the second one with. Are you sure you've got the one with the extra UART? If so, it should be usable for SmartAudio.

If you want SOFTSERIAL you'll have to compile a custom target for it.

turboed13b, 2018-12-11:
Yeah I have uart 3 under ports tab. I enabled smart audio and it isnt working.

I will have to look up how to compile a target.

mikeller, 2018-12-11:
Maybe @githubDLG can help with using the UARTs?

turboed13b, 2018-12-12:
Hmm its the newest fr version with uart 3 breakouts. I got it working with a matekf411 but not on the crazybee. Also I am trying to enable soft serial but cant find what to define. What is soft serial called?

turboed13b, 2018-12-12:
I got softserial working and smart audio works on that. I guess it just isnt compatible with the hardware uart.

githubDLG, 2018-12-12:
hmm........  but the tx3 and rx3 are directly connect to mcu , without any component or devices. Besides, we have test sa feature on our crazybee f3 fr/fs/dx V1.2, all work well..
Can not be sure what has happend to your board.
And, aother thing is that SA can not work with rx at the same time with the same uart.

turboed13b, 2018-12-12:
I'm still using the spi rx. It's probably this cam it's not a real smart audio v1. Turbowing cyclops v2.

githubDLG, 2018-12-12:
Hm...... okay, so use the soft serial first, it can work at least , haha.

jflyper, 2018-12-12:
https://github.com/betaflight/betaflight/wiki/Unify-Smartaudio#compatibility

closed 2018-12-12 by turboed13b