betaflight/betaflight #7212: Incorrect Smart Port ID

MikeBland, 2018-12-12:
You are using a SPort application ID of 0x0420 for "distance" (FSSP_DATAID_HOME_DIST = 0x0420).
This is a reserved FrSky ID and should not be used for a "user" sensor.
FrSky have provided a range of "user IDs" for such use and it is one of these that should be used (ID between 0x5100 and 0x51ff).
This is needed to remain compatible with all open source transmitter firmware, and avoid any possible future problems should FrSky make use of "their" ID of 0x0420.

Mike

mikeller, 2018-12-12:
@MikeBland: Can you please elaborate a bit more where you think the problem lies in a flight controller firmware using `FSSP_DATAID_HOME_DIST` to transmit the distance to home - I don't think there is a problem as the typical use case for Betaflight or other flight controller firmware is that it is the only device connected to the SmartPort bus, so that there are no other devices sending sensor values.

Can you please provide a link to the official list of SmartPort sensor values?

I am particularly interested in seeing if there are any 'user IDs' so that will be recognised as 'distance to home' by the TX firmware.

Also, n.b. that Betaflight also sends sensor values for a lot of the other predefined sensor IDs: https://github.com/betaflight/betaflight/blob/9bae508ab8a7f2d7b9f7a7532aa255e986723dfd/src/main/telemetry/smartport.c#L88-L144

kilrah, 2018-12-12:
> Also, n.b. that Betaflight also sends sensor values for a lot of the other predefined sensor IDs:

That's perfectly fine

> I am particularly interested in seeing if there are any 'user IDs' so that will be recognised as 'distance to home' by the TX firmware.

There isn't. But we can decide of one in the 0x5100 - 0x51ff range :)

As Mike said it's OK now, but at some point it's pretty sure FrSky will want to use 0x0420 and whenever that happens we'd start recognising it as what they use it for. Worst case it gets used as a bitfield like for the redundancy bus and hidden/processed separately.

MikeBland, 2018-12-13:
If you are using FrSky SmartPort IDs, without having a NDA with FrSky and so obtaining the required specification, then your entire SmartPort implementation is based on "hearsay".
If you have used, for example, openTx source code to obtain the FrSky IDs, then you should also be able to see the values for "DIY_FIRST" etc. (which are actually slightly incorrect in the frsky.h file in openTx), and so understand you should be using a value in the range 0x5100 to 0x51FF.

Mike

kilrah, 2018-12-13:
Both Mike and myself have access to that documentation and as you probably know by what we work on are in regular contact with FrSky, and are relaying relevant content of it to you. 
I have fixed smart port code in CF/BF in the past when the reverse engineering of it wasn't quite correct.

FrSky couldn't care less if your custom stuff breaks one day because you didn't use the IDs they've reserved for custom stuff. Heck we probably couldn't care less either... But a BF user came to us and talked about that thing, we saw it's incorrect, and as a result we're just making the little effort to give you a friendly heads up about it. If we have to fight for you to believe it then consider it done and do whatever you feel about it.

jflyper, 2018-12-13:
What is FSSP_DATAID_HOME_DIST currently used for? What is the consumer application?

wind0r, 2018-12-13:
I wonder why this just is about FSSP_DATAID_HOME_DIST and not all the other "FrSky" Ids betaflight use to transport telemetry?

I think in the end FrSky just wants Betaflight to remove all FSSP_DATAID_ out of the codebase and move to custom ones.
That sounds realy strange and will annoy nearly every user. Why something like voltage should be send though some custom id when there are allready 9 different FSSP_DATAID_VFAS for that.

If that is the case and frsky realy wants to block every DATAID for them self i would like to get a public announcement. so we can forward that the betaflight users so they are warned. out of the dark we will not break smartport for every betaflight user without any document we can link them to.

I think the sane solution here as every other Manufacture does it. FrSky just should publish which id should be used for what and what data is allowed to be send to. Custom IDs are total understandable for stuff like MSPoverSmartport or other "Product" Specific stuff but common values like voltage, distance and position should be send though common ids so the userinterface will work well and doesnt need a lot of fiddling around. Everything else just will hurt the enduser.

MikeBland, 2018-12-13:
wind0r: The value of 0420 being used as FSSP_DATAID_HOME_DIST is not a defined ID, but may become defined for use in the future. The "other" IDs in use are defined and are being used for the same data types as FrSky. All these "other" IDs are therefore perfectly OK to use, it is just FSSP_DATAID_HOME_DIST that may conflict in the future.

mikeller: I believe there are others (Pixhawk?) using some of the user IDs, but, as you say, they won't be in use at the same time as any you use. They also won't (ever) be processed directly by the Tx firmware as they are, by definition, carrying unknown (user) data that needs special processing, either by passing to a script (LUA or BASIC), or specifically configured to be handled.
Good that you have the 3.4.7 spec. so you may see exactly what is currently used.

Mike

kilrah, 2018-12-13:
> Frankly, you need to adjust your attitude. Nobody will care about you 'giving a friendly heads up' after seeing how you treat users trying to fix bugs in projects that you are involved in

You did something without knowing all the underlying stuff, there was a discussion, and 4 developers agreed that nothing should be done. What do you have against me specifically? 

> Good that you have the 3.4.7 spec. so you may see exactly what is currently used.

Note that 3.4.7 has an error that has been discovered following you mentioning the error with DIY_FIRST, there will be an update to it.

> They also won't (ever) be processed directly by the Tx firmware as they are, by definition, carrying unknown (user) data that needs special processing, either by passing to a script (LUA or BASIC), or specifically configured to be handled.

That's not correct. 5100-52ff are handled by telemetry like now. it's 5000-50ff (missing in 3.4.7) that are routed to scripts.

wind0r, 2018-12-13:
@MikeBland 	thanks for clarification. sounds like a total sane change. i dont have access to the spec but i wonder if there is a other value we can use for that. if that is not the case we should use some custom id for it. in best case the same id that also others are using or even better getting frsky to create one for that kind of telemetry.

If the git history isnt fooling me it looks like we added that value since it is a backport from iNav. Maybe frsky just should use the change and just define HOME_DISTANCE as 0x0420. I also wonder how/why iNav choosed that value.
All this IDs stuff looks like to not to be good documented and with a communication that could be improved.

mikeller, 2018-12-14:
@kilrah: All that I (and everybody that I asked) can get out of your (and other OpenTX core members') attitude presented in opentx/opentx#6037 is that you responded with 'go away, you are not one of us, we are not interested in valid fixes for problems in our project if they are not fixed by one of us'.

If you are trying to use 'without knowing all the underlying stuff' against me, I challenge you to provide examples of places where you showed knowledge of the underlying radio protocol that is used by the FrSky hardware.

So forgive me, this is not personal against you, but this does not put you into a position to allow you to give a 'friendly heads up' for a project that is run as a community effort, compared to OpenTX that seems to be a case of best mates doing inspections of each others' back sides.

As stated before, I will get this resolved with FrSky directly.

kilrah, 2018-12-14:
That was not the case at all. OpenTX is just as much a community project, we do accept 3rd party PRs regularly. 

This one following the discussion was deemed inappropriate for our project and that's the only reason it was rejected. Maybe it would have its place in a specific quad-centric fork of OpenTX that someone could do, but that's not what our team is working on.

Don't tell me you'd accept anything in betaflight that to the best of your evaluation would be detrimental to it or not follow the project's goals/guidelines... There wouldn't be so many forks of CF/BF/iNav/... if you accepted everything, each of these have their own goals.

If you felt we "reject outsiders" it's probably becasue you never accepted our analysis, chellenged it repeatedly and took it personally, but that's not something we can do much about.

kilrah, 2018-12-14:
> If you are trying to use 'without knowing all the underlying stuff' against me, I challenge you to provide examples of places where you showed knowledge of the underlying radio protocol that is used by the FrSky hardware.

We do not have precise knowledge of it, that's the whole point.
If you talk to FrSky, figure out how it works, and can tell us for sure that it's not going to have detrimental effects or unwanted interference that could affect other users then your PR could totally get in.
But it's something you have to provide since we do not have the ressources to dedicate to figuring it on our side at this point.
Again if we came with a complete sensor fusion rewrite for BF you'd want us to validate the heck out of what we're doing before accepting it (well I hope so).

wind0r, 2018-12-14:
`No it's not reserved, would be in the doc otherwise. Looks like a random choice, it's simply then next at the end of a block of other used values.`

Since there is not any Documentation around this still is not established and proven. Also could you please contact the original author to get a proof for that it just was random choice.

Thank you.

`Talk to them, but usually the only IDs they reserve are those they use for their own sensors. They aren't going to reserve an ID for something none of their sensors use, that would then be unavailable for something else on a future product. That's precisely why they allocated a whole bunch of IDs for the community to use.`
That sounds like a realy bad practise and bad design.

kilrah, 2018-12-14:
> Since there is not any Documentation around this

The documentation is Mike's, mine's and @mikeller 's hands...

> That sounds like a realy bad practise and bad design.

A lot of FrSky's stuff is far from ideal and a lot of improvement ideas have been refused by them over time as well, but we make it work. They have their pride in China too.

wind0r, 2018-12-14:
As @mikeller allreay wrote in the first respone to this thread.
> Can you please provide a link to the official list of SmartPort sensor values?

With out such a List it is not possible to check which use of ids is fine and which is not. I can not create a pr and improve it without knowing which ids i am allowed to use and which i am not. We can not build on that some third party needs to check the code and reports back. FSSP_DATAID_HOME_DIST  = 0x042 already shows what happens. It is around for 2 years and no one noticed it  ;)

We are still missing the inital author part. Maybe the inital author did talked with frsky and reserved that id and frsky will do it as soon they have a device for it? It looks like a empty spot in every implementation. I do not get why someone should skip some ids without a reason so maybe it is the case. 

Without any documentation it just not possible for me to judge this topic.

kilrah, 2018-12-14:
As he said in a later post,

> they did send me the 3.4.7 spec for SmartPort.

kilrah, 2018-12-14:
As mentioned earlier in this thread FrSky does not want this distributed freely, obtaining it requires going directly to them and signing an NDA with them.

wind0r, 2018-12-14:
It is not about trust. I just want to read the spec and check of all other values are fine. I did you guys already did that?

Something like this should not be around for over 2 years without anyone noticing it.
I understand that they do not want to full document aviable but atleast those IDs altleast need to be public. 

Also it looks quite shady to sign a NDA and then fix a GPL Licenced Project out of the Informations you got. Isnt that violating the NDA or is there a extra clause for that in it?

MikeBland, 2018-12-14:
I don't use Betaflight, so why would I have noticed this? It was brought to my attention by an ersky9x user asking how to handle it.
Why should those IDs be made public? FrSky produce sensors and have defined (under NDA) the IDs they use for these. Betaflight is emulating FrSky sensors, so if it is doing so without having signed the NDA, how did they obtain the IDs being used?

jflyper, 2018-12-14:
It's sh*t throwing.

Why don't we come back to this when FrSky start to use the ID under question with different data format/representation that is used in the current implementation?

For users, confusion is now or then.

jflyper, 2018-12-14:
Can I close this?

MikeBland, 2018-12-15:
I'm not sure where all this animosity is coming from. I'm simply trying to raise awareness of a potential problem so it is sorted before it actually occurs, and provide information to help avoid any similar problem in the future.
The situation, as I understand it is as follows:
FrSky produce sensors and receivers that use hardware and a protocol called SmartPort. They have said to me that this is patented. They have released, under NDA, to a number of open source developers the specification of the SmartPort, and permit those developers to implement firmware (and hardware), and publish the source code of the firmware, to make use of the SmartPort.
It was brought to my attention that Betaflight is using the SmartPort (no reason why it shouldn't), but that ONE application ID being used was not one defined by FrSky, so is not automatically handled by ersky9x transmitter firmware (or openTx). Also, with the value currently in use (0x0420), it isn't passed (in ersky9x) to scripts (and may not be passed to scripts in openTx).
With the way FrSky make use of the application IDs, they allocate a base value that is a multiple of 0x0100 to "sensor types", so 0x0200 is the base for battery sensing. From this base value, "sub types" are allocated at multiples ox 0x0010, so 0x0200 is battery current, and 0x0210 is battery voltage.
0x0400 is allocated to a sensor type of "Temperature", with, currently, 0x0400 as temperature 1 and 0x0410 as temperature 2. Clearly, 0x0420 may well become temperature 3.

My reason for raising this issue is that using 0x0420 for "Distance" may therefore become a problem.

FrSky have made a range of application IDs available for use by "sensors" they do not produce (quite generous compared to some manufacturers), and it seems to me that a "Distance" sensor should therefore use one of these. Should Betaflight, or any other non-FrSky device, wish to add other sensors that are not currently available from FrSky, then, by using these "User DIY" IDs, no conflict will arise with FrSky in the future.

So, since you said "sounds like a total sane change", I would request the issue remains active.

Mike

wind0r, 2018-12-15:
Just for the record and since the last post mentioned Temperature 1 and 2.
Please have a look at the implementation. You will notice that we do not transfer "temperature" over those ids. 
Which also was the reason to use "Temperature 3" as Home Distance.

Thats the reason I said this is not only about ID 0x0420 and without a public Documentation this will not get any better.