betaflight/betaflight #7214: BUG - IRC Pit Mode

Eganwp, 2018-12-12:
After a lot of troubleshooting and testing I found that when enabling the Pit Mode feature from the modes tab in BFconfig, IRC Tramp protocol for smart audio appears to work when a switch is flipped to take it out of pit mode. However, I’ve confirmed it’s not forcing the VTx out of out of pit mode at all, and even confirmed on the VTx power output status led’s. This leads to not being able to fly a quad further than 15ft away as it’s stuck in pit mode.

- Travis

wind0r, 2018-12-12:
Could you please say which version you are using.

Also it would be great if you follow the issue template the next time. 
It was written for a reason so we dont need to ask about the setup ;)

Eganwp, 2018-12-12:
Sorry was submitting this from my phone in a hurry and didn’t realize the template existed. I’ll submit a dump when I’m back home. 

BF: 3.5.1
BFC: 10.4
FC: Mamba F405 using IRC Tramp smart audio over UART 3
VTx: RunCam TX200U

- Travis

etracer65, 2018-12-12:
As requested in the issue creation instructions, please provide a `diff` of your settings. Also provide details on the actual vtx you are using. There are many vcx's from different brands that implement the tramp protocol - some partially or incorrectly. Also many don't support pit mode at all.

wind0r, 2018-12-12:
Something like IRC Tramp smart audio does not exist.

There was a bug in 3.3 which caused this kind of issue with IRC Tramp but that got fixed with 3.3.3.
So this looks like a hardware issue from the RunCam TX200U since it is working with the IRC Tramp.

Eganwp, 2018-12-12:
Changed some of the OP wording for clarity.

Eganwp, 2018-12-12:
IRC Tramp via OSD does work to change the power levels without issue (ie 25mW to 200mW) and confirmed on the VTx status led’s. 

Also, the RunCam TX200 does support pitmode out of the box and I can confirm it works via button pushes and the status LED’s and signal strength.

wind0r, 2018-12-12:
IRC Tramps works fine for plenty of people.
This is a issue should be raised against RunCam and this TX200 Vtx.
This isnt something Betaflight can or will fix.

Eganwp, 2018-12-12:
Have you confirmed IRC Tramp “smart audio” connection to FC works fine for getting an ImmersionRC branded Tramp VTX into and out of pitmode via the pitmode feature in BFConfigurator in the modes tab? I have a feeling it won’t work on a real ImmersionRC branded Tramp either.

wind0r, 2018-12-12:
Since i do not have access to the quad right in this moment. i am unable to test this currently.

I checked the Code and it just toggles the pitmode flag and never touches the power level also there is an pr around about this topic which fixed pit mod on/off for tramp here:
https://github.com/betaflight/betaflight/pull/5621

it got tested there and the code didnt got touched much after that. so it may is the possibility that it is not working but i would not bet on it. also since there are way more tramp users around then tx200u

but as said above it would be great if someone with a TrampHV can test that.

Still this will not fix other issues like the powerlevel switching is not working with that tx200u