betaflight/betaflight #7216: Not receiving MAVLink HEARTBEAT messages

christianrauch, 2018-12-13:
**Describe the bug**
I activated the MAVLink telemetry feature on the NAZE target (in source and config) on betaflight but cannot connect with a simple example script. Tested with a Naze32 Rev6

**To Reproduce**
1. Apply the following patch to betaflight 3.2.5:
```diff
$ git diff v3.2.5 
diff --git a/src/main/target/NAZE/target.h b/src/main/target/NAZE/target.h
index 28f2d9c45..e838f801a 100644
--- a/src/main/target/NAZE/target.h
+++ b/src/main/target/NAZE/target.h
@@ -116,6 +116,9 @@
 //#define SONAR_TRIGGER_PIN_PWM   PB8
 //#define SONAR_ECHO_PIN_PWM      PB9
 
+#undef LED_STRIP
+#define TELEMETRY_MAVLINK
+
 #define USE_UART1
 #define USE_UART2
 /* only 2 uarts available on the NAZE, add ifdef here if present on other boards */
```

2. Enable telemetry and port sharing (MSP+MAVLink at 115200 kbps):
```
# diff
# Betaflight / NAZE (AFNA) 3.2.5 Dec 13 2018 / 02:26:37 (67bf8c247) MSP API: 1.36

feature TELEMETRY
serial 0 513 115200 57600 115200 115200
profile 0

rateprofile 0
```

3. Connect with example script to UART1: [mavlink_listener.py.txt](https://github.com/betaflight/betaflight/files/2678121/mavlink_listener.py.txt):
`./mavlink_listener.py --device=/dev/ttyUSB0 --baudrate=115200`

**Expected behavior**
The script should receive HEARTBEAT messages, similar to:
```
got heartbeat
Raw message: HEARTBEAT {type : 2, autopilot : 0, base_mode : 0, custom_mode : 0, system_status : 3, mavlink_version : 3}
Raw message: HEARTBEAT {type : 2, autopilot : 0, base_mode : 0, custom_mode : 0, system_status : 3, mavlink_version : 3}
```

**Setup / Versions**
 - Flight controller: Naze32 rev6 full
 - Other components: betaflight 3.2.5
 - how are the different components wired up: serial connection is directly established over the USB-UART

**Additional context**
I also tried to enable MAVLink alone on the UART1 port (hence, setting MSP to UART2), but also without a connection.

etracer65, 2018-12-14:
You're deeply into unsupported territory. The NAZE32 target is no longer supported (3.2 was the last version) and you're trying to use a version that is 3 major releases old. You really should consider upgrading to a supported  flight controller - F4's can be found in the $15 range. Then use the current release and it's more likely that somebody will be able to help you.

christianrauch, 2018-12-14:
I understand that the Naze is not supported by newer firmware versions. But how do I know if this issue is hardware or software related?
- Can someone reproduce the issue with this set of features and the configuration on a different flight controller?
- Is there an example MAVLink client program that is known to work in combination with MAVLink telemetry?

etracer65, 2018-12-14:
The point of unsupported is that it’s unlikely that anybody will invest the effort to support. The only information I can supply is that there are no known issues with MAVlink in the current release.

jflyper, 2018-12-14:
@christianrauch Did you try it on non-shared port (for testing purpose)?

christianrauch, 2018-12-14:
@jflyper Yes, I moved MSP to UART2 and set MAVLink exclusively on UART1, but with the same outcome.

christianrauch, 2018-12-14:
I tried the same on betaflight 3.5.3 with the following source diff:
```diff
$ git diff 3.5.3
diff --git a/src/main/target/NAZE/target.h b/src/main/target/NAZE/target.h
index d583e1d16..bdbd49591 100644
--- a/src/main/target/NAZE/target.h
+++ b/src/main/target/NAZE/target.h
@@ -121,6 +121,9 @@
 //#define RANGEFINDER_HCSR04_TRIGGER_PIN_PWM   PB8
 //#define RANGEFINDER_HCSR04_ECHO_PIN_PWM      PB9

+#undef USE_LED_STRIP
+#define USE_TELEMETRY_MAVLINK
+
 #define USE_UART1
 #define USE_UART2
 /* only 2 uarts available on the NAZE, add ifdef here if present on other boards */
```

And the same configuration (MAVLink shared with MSP):
```
# diff
# Betaflight / NAZE (AFNA) 3.5.3 Dec 14 2018 / 22:52:06 (c00e98756) MSP API: 1.40

feature -DYNAMIC_FILTER
feature TELEMETRY
serial 0 513 115200 57600 115200 115200
profile 0

rateprofile 0
```

With the same result.

Can someone confirm that this configuration is working on an alternative FC with the script that I provided?

jflyper, 2018-12-15:
@christianrauch 
Well, I can see it's transmitting something in my terminal window, but I can't decode it as the `mavlink_listner.py` requires `pymavlink` which I don't know how to install.

```
feature TELEMETRY
serial 0 512 115200 57600 115200 115200
```

I think you are on your own from here.

christianrauch, 2018-12-15:
@jflyper 
Which terminal application do you use? Is it the CLI in the betaflight configurator?

You can install `pymavlink` via `pip`. E.g. on Ubuntu via `sudo -H pip install pymavlink`.

christianrauch, 2018-12-15:
Oh sorry, I just saw now that you have solely MAVLink activated in your configuration.
I got it working now on betaflight 3.5.3, without port sharing (i.e. with deactivated MSP).

Port sharing with MSP does not work, though. But I can live with this.

Thanks @jflyper

jflyper, 2018-12-15:
@christianrauch 
Oh, okay, now you may try the port sharing and see if something new happens when you arm.

mikeller, 2018-12-15:
@christianrauch: The way port sharing between MSP and telemetry works is that the port is used for MSP while unarmed and for telemetry while armed. So unless you arm before testing there is no expectation for this to work.

SiHCl3, 2023-12-28:
> I tried the same on betaflight 3.5.3 with the following source diff:
>
> ```diff
> $ git diff 3.5.3
> diff --git a/src/main/target/NAZE/target.h b/src/main/target/NAZE/target.h
> index d583e1d16..bdbd49591 [...]

> @christianrauch: The way port sharing between MSP and telemetry works is that the port is used for MSP while unarmed and for telemetry while armed. So unless you arm before testing there is no expectation for this to work.

but as everything remain the same only to switch mavlink to LTM,S.PORT, or frsky and i can see something through the uart
while nothing goes out from the uart when i choose mavlink or hott