betaflight/betaflight #7221: Add Stick Overlays to the OSD active elements menu

pkruger, 2018-12-16:
**Describe the solution you'd like**
Add the Stick Overlay OSD elements to the OSD active elements menu so these can be turned on/off from within the OSD.

**Describe alternatives you've considered**
Only configuring these via the CLI/Configurator or having them on a separate OSD profile.

**Additional context**
This may be handy in the field.  Also, what is the criteria for an item to be included in the OSD active elements menu?  (i.e. GPS elements are not listed.)

mikeller, 2018-12-17:
I think unlisted elements are merely an oversight, and not by design.

pkruger, 2018-12-18:
Cool, I'll add it, maybe we should add all the missing ones as part of this?

mikeller, 2018-12-18:
@pkruger: Yeah, go for it!

closed 2018-12-24 by mikeller