betaflight/betaflight #7223: I-relay at min throttle with BF 4.0.0 build >1280

Ralfde, 2018-12-16:
**Important: This is not the right place to get help with configuration issues. Your issue will be closed without further comment.**

Please get help from other user support options such as asking the manufacturer of the hardware you are using, RCGroups: https://rcgroups.com/forums/showthread.php?t=2464844, or Slack (registration at https://slack.betaflight.com/) or other user support forums & groups (e.g. Facebook).

etracer65, 2018-12-16:
It's not clear what you're asking. Why would you want to switch Iterm Relax based on throttle? Also what do you mean "starting function"?

Ralfde, 2018-12-16:
I don't want to change iterm relax on throttle, but BF4.0.0 Build does.
# Betaflight / OMNIBUSF4 (OBF4) 4.0.0 Dec 9 2018 / 18:50:38 (2e81109be) MSP API: 1.41
When I start the motors, I have big I relax, but I don't want it, so the start function does not work correct.

set airmode_start_throttle_percent = 0 does only activate the airmode at 0%, but does not switch of the I relax.
I have no idea why this firmware does make I relax, if I do not have it activated.
I use PID Profil 1.

# version
# Betaflight / OMNIBUSF4 (OBF4) 4.0.0 Dec 9 2018 / 18:50:38 (2e81109be) MSP API: 1.41

board_name OMNIBUSF4
manufacturer_id AIRB
mcu_id 004000203436510133323835
signature 

# reset configuration to default settings
defaults nosave

# name

# resources

# mixer

# servo

# servo mix

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature TELEMETRY
feature LED_STRIP

# beeper

# beacon

# map
map TAER1234

# serial
serial 0 64 115200 57600 0 115200
serial 2 8 115200 57600 19200 115200

# led
led 0 0,0::C:2
led 1 1,0::C:2

# color

# mode_color

# aux
aux 0 0 1 1800 2000 0 0
aux 1 1 0 1400 1600 0 0

# adjrange

# rxrange

# vtx

# rxfail

# master
set gyro_lowpass_type = PT1
set gyro_lowpass_hz = 100
set acc_calibration = -146,0,151
set baro_hardware = NONE
set rssi_channel = 16
set rssi_scale = 125
set rssi_offset = -12
set rc_interp = OFF
set fpv_mix_degrees = 20
set serialrx_provider = SUMD
set airmode_start_throttle_percent = 0
set motor_pwm_protocol = DSHOT600
set vbat_max_cell_voltage = 45
set current_meter = VIRTUAL
set yaw_motors_reversed = ON
set small_angle = 180
set deadband = 10
set yaw_deadband = 10
set pid_process_denom = 1
set osd_vbat_pos = 2103
set osd_rssi_pos = 2137
set osd_tim_1_pos = 54
set osd_tim_2_pos = 33
set osd_flymode_pos = 34
set osd_throttle_pos = 225
set osd_vtx_channel_pos = 377
set osd_crosshairs_pos = 2253
set osd_ah_sbar_pos = 2254
set osd_ah_pos = 2126
set osd_current_pos = 385
set osd_mah_drawn_pos = 353
set osd_craft_name_pos = 362
set osd_gps_speed_pos = 218
set osd_gps_lon_pos = 82
set osd_gps_lat_pos = 65
set osd_gps_sats_pos = 51
set osd_home_dir_pos = 302
set osd_home_dist_pos = 303
set osd_compass_bar_pos = 266
set osd_altitude_pos = 247
set osd_pid_roll_pos = 423
set osd_pid_pitch_pos = 455
set osd_pid_yaw_pos = 487
set osd_debug_pos = 1
set osd_power_pos = 321
set osd_pidrate_profile_pos = 345
set osd_warnings_pos = 2409
set osd_avg_cell_voltage_pos = 76
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_battery_usage_pos = 392
set osd_disarmed_pos = 138
set osd_nheading_pos = 311
set osd_nvario_pos = 279
set osd_esc_tmp_pos = 82
set osd_esc_rpm_pos = 83
set osd_stat_max_spd = OFF
set osd_stat_min_rssi = OFF
set osd_stat_max_curr = OFF
set osd_stat_used_mah = OFF
set osd_stat_bbox = OFF

# profile
profile 0

set dterm_lowpass_type = PT1
set dterm_lowpass_hz = 100
set dterm_lowpass2_hz = 100
set vbat_pid_gain = ON
set iterm_rotation = OFF
set iterm_relax = OFF
set iterm_relax_type = GYRO
set yaw_lowpass_hz = 100
set throttle_boost = 0
set acro_trainer_angle_limit = 40
set f_pitch = 100
set f_roll = 100

# profile
profile 1

set dterm_lowpass_type = PT1
set dterm_lowpass_hz = 100
set dterm_lowpass2_hz = 100
set vbat_pid_gain = ON
set iterm_rotation = OFF
set yaw_lowpass_hz = 100
set acro_trainer_angle_limit = 40

# profile
profile 2

set dterm_lowpass_type = PT1
set dterm_lowpass_hz = 100
set dterm_lowpass2_hz = 100
set vbat_pid_gain = ON
set iterm_rotation = OFF
set yaw_lowpass_hz = 100
set throttle_boost = 0
set acro_trainer_angle_limit = 40

# restore original profile selection
profile 0

# rateprofile
rateprofile 0

set tpa_rate = 10
set tpa_breakpoint = 1250

# rateprofile
rateprofile 1

# rateprofile
rateprofile 2

# rateprofile
rateprofile 3

# rateprofile
rateprofile 4

# rateprofile
rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save

Ralfde, 2018-12-16:
Ups. Never forgot this before. Thanks for the help.

Ralfde, 2018-12-16:
But there is still an issue.
With Build 1262 it works correct.
With the build above, if I activate the starting option by little throttle and go back to min. throttle again, there is a I relax or I discharge, which should not be and is not with 1262.

Ralfde, 2018-12-16:
airmode_start_throttle_percent can also not be changed. It will be shown with 32, even after changing it.

etracer65, 2018-12-16:
Using the latest dev build I just tested that:
1. `airmode_start_throttle_percent` can be changed.
2. Airmode remains inactive (iterm will reset) until the throttle reaches the percentage, then airmode activates and iterm stops resetting.
3. Using `airmode_start_throttle_percent = 0` works as expected. Airmode is activated immediately on arming.