betaflight/betaflight #7225: DALRC F722: ACC init fail from #1285

RipperDrone, 2018-12-16:
up to BF4.0.0 #1284, ACC init button is shown in BFC, greyed out / not found from #1285

FC = DALRCF722, other hardware doesn't matter I guess...

Think this desciption is complete enough to reproduce and identify the likely acc initialization/detection bug.

jflyper, 2018-12-16:
How many times do I have to tell you to attach a config diff at least? 😠

jflyper, 2018-12-16:
CLI `status` will also help in this case.

RipperDrone, 2018-12-16:
ok - diff is null (I tried with defaults nosae, and same issue)
Therefore, resources are default.

jflyper, 2018-12-16:
`status`?

jflyper, 2018-12-16:
CLI `resource list` isn't null?

RipperDrone, 2018-12-16:
just saying I left everything at defaults, so are the resources. Will flash back and post them, anyway... 

Still convinced it's avery basic mal-detection of gyro after the per pin init changes...

dadoribeiro, 2018-12-16:
I can corroborate. Same issue here. No gyro 1 on builds after December 13th. Only Gyro 2 (ICM), but I have an issue with blackbox and 32k gyro so I can´t log it. Complete defaults.

jflyper, 2018-12-16:
@dadoribeiro Thanks, it's a device detection problem, so logging is irrelevant.

@RipperDrone Let me see `set gyro_1` outputs.

RipperDrone, 2018-12-16:
Here ya go:

# set gyro_1
gyro_1_bustype = SPI
Allowed values: NONE, I2C, SPI, SLAVE

gyro_1_spibus = 1
Allowed range: 0 - 4

gyro_1_i2cBus = 0
Allowed range: 0 - 4

gyro_1_i2c_address = 0
Allowed range: 0 - 119

gyro_1_sensor_align = CW180
Allowed values: DEFAULT, CW0, CW90, CW180, CW270, CW0FLIP, CW90FLIP, CW180FLIP, CW270FLIP

# resource
resource BEEPER 1 C13
resource MOTOR 1 C06
resource MOTOR 2 C07
resource MOTOR 3 C08
resource MOTOR 4 C09
resource MOTOR 5 A08
resource MOTOR 6 A09
resource PPM 1 B07
resource LED_STRIP 1 A15
resource SERIAL_TX 1 B06
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 C10
resource SERIAL_TX 4 A00
resource SERIAL_TX 5 C12
resource SERIAL_RX 1 B07
resource SERIAL_RX 2 A03
resource SERIAL_RX 3 C11
resource SERIAL_RX 4 A01
resource SERIAL_RX 5 D02
resource I2C_SCL 1 B08
resource I2C_SDA 1 B09
resource LED 1 C14
resource SPI_SCK 1 A05
resource SPI_SCK 2 B13
resource SPI_SCK 3 B03
resource SPI_MISO 1 A06
resource SPI_MISO 2 B14
resource SPI_MISO 3 B04
resource SPI_MOSI 1 A07
resource SPI_MOSI 2 B15
resource SPI_MOSI 3 B05
resource CAMERA_CONTROL 1 B01
resource ADC_BATT 1 C01
resource ADC_RSSI 1 A00
resource ADC_CURR 1 C00
resource FLASH_CS 1 B02
resource OSD_CS 1 B12
resource GYRO_EXTI 1 B10
resource GYRO_EXTI 2 C04
resource GYRO_CS 1 B00
resource GYRO_CS 2 A04

RipperDrone, 2018-12-16:
gyros 1 and 2 comm over same SPI bus (both =1) is correct?

RipperDrone, 2018-12-16:
[image]

jflyper, 2018-12-16:
> gyros 1 and 2 comm over same SPI bus (both =1) is correct?

Yes.

jflyper, 2018-12-16:
I think I know what's happening... hang on. Possible fix is coming.

RipperDrone, 2018-12-16:
trying to read into PR #7190 but this valley too deep for me :-) Seems related to timings and trigger signals to pre-init and then finally init devices... so I feel useless here.

jflyper, 2018-12-16:
Try this.
[betaflight_4.0.0_DALRCF722DUAL_gyro1-detection-0.hex.zip](https://github.com/betaflight/betaflight/files/2683756/betaflight_4.0.0_DALRCF722DUAL_gyro1-detection-0.hex.zip)

RipperDrone, 2018-12-16:
Part success, I'd say:

- still hanging for about 10-15secs on reboot after flash
- have to unplug/replug USB, wait long, then BFC activates 'connect' button
- I can now calibrate after connect

RipperDrone, 2018-12-16:
# status
MCU F722 Clock=216MHz, Vref=3.31V, Core temp=50degC
Stack size: 2048, Stack address: 0x20010000
Config size: 2819, Max available config: 16384
GYRO=MPU6000, ACC=MPU6000
System Uptime: 76 seconds, Current Time: 2018-12-16T16:55:12.273+00:00
CPU:3%, cycle time: 125, GYRO rate: 8000, RX rate: 33, System rate: 9
Voltage: 0 * 0.1V (0S battery - NOT PRESENT)
I2C Errors: 3
Arming disable flags: RXLOSS CLI MSP

RipperDrone, 2018-12-16:
Hm, after this one-time hanging, everything seems ok (despite a glitch in BFC Nightly that I have to press disconnect twice, and can only then reconnect...)

jflyper, 2018-12-16:
Can I see another (new) `status` again?

RipperDrone, 2018-12-16:
# status
MCU F722 Clock=216MHz, Vref=3.30V, Core temp=54degC
Stack size: 2048, Stack address: 0x20010000
Config size: 2819, Max available config: 16384
GYRO=MPU6000, ACC=MPU6000
System Uptime: 98 seconds, Current Time: 2018-12-16T16:59:18.952+00:00
CPU:3%, cycle time: 126, GYRO rate: 7936, RX rate: 33, System rate: 9
Voltage: 0 * 0.1V (0S battery - NOT PRESENT)
I2C Errors: 1
Arming disable flags: RXLOSS CLI MSP

RipperDrone, 2018-12-16:
not exactly 8k gyro sampling... :-)

jflyper, 2018-12-16:
The one-time hanging seems to have occurred due to two extra I2C errors... Can't tell what the board was trying to detect.

RipperDrone, 2018-12-16:
I see... which devices are I2C comm related?

RipperDrone, 2018-12-16:
I tried to disconnect/reconnect, restart BFC several times. Seems to stay ok now!

RipperDrone, 2018-12-16:
Shall we consider it resolved? Thanks anyway - have to head for a bite now. Be back in some 45mins :-)

jflyper, 2018-12-16:
The two extra I2C errors are probably not related with the gyro detection issue.

I will submit a new PR for the fix. Thanks for your cooperation :)

(And be sure to attach least info right from the beginning next time ;))

jflyper, 2018-12-16:
@RipperDrone This issue ticket will automatically be closed via #7227.

PR #7227 (2018-12-17): Do call gyroPreInitsensor for gyro 2 even if it is not activated
Fixes #7225 

`gyroPreInitSensor` (and consequently `spiPreinitRegister`) must be called for second SPI gyro even if it is not activated.

DALRCF722DUAL revealed this bug as it uses two gyros on the same SPI bus 😭
closed 2018-12-17 by mikeller