betaflight/betaflight #7226: Motor Configuration error unknown

nwb3, 2018-12-16:
I recently got a drone, using a crazybee f3 flight controler. The quad is armed and bound, but the motors will not spin. I could not find and error code or anything. Could I have some help?

RipperDrone, 2018-12-16:
Motor protocol correct? USB connection removed (safety feature not to arm while connected)? No pre-arm condition defined keeping you from being able to single-switch arm?

nwb3, 2018-12-16:
Its set to dshot 600. What are some of the pre-arm conditions. The motors will also not spin in betaflight, although, I know the motors worked, since I have tested them.

jflyper, 2018-12-16:
I'm closing this for the following reasons.

1. The author has not followed the basic instruction.
2. The issue is most likely a configuration problem.

closed 2018-12-16 by jflyper

nwb3, 2018-12-16:
I'm tryig to find the issue!

jflyper, 2018-12-16:
You didn't even give which variant of the CrazyBee controller you are having trouble with, no firmware version is given, no config diff dump. These are requested in the instruction given when you opened the new issue for the first time.