betaflight/betaflight #7229: SmartAudio doesn't work (Kakute F7 AIO and TBS Unify Pro 5G8 V3 on UART4)

andrejpodzimek, 2018-12-17:
Using Betaflight 3.5.3, Holybro Kakute F7 AIO, TBS Unify Pro 5G8 V3 (with a 30.5x30.5 mounting board from TBS) and RunCam Split V2.

There's 4.7 kΩ between UART4 TX and the audio pad on the VTX; more precisely, I'm using [the "Ideal connection"](https://github.com/betaflight/betaflight/wiki/Unify-Smartaudio#using-both-smartaudio-and-microphone-at-the-same-time) described in the wiki. The VTX is powered from the FC's 5V branch that does **not** get power from USB, only from the battery.

In the OSD, the `VTX SA` menu has just dashes and nothing meaningful there. Setting it to "online V2" manually doesn't help. (It shows offline by default.) Also tried to set `vtx_halfudplex` off and back on, but that didn't help either. Here's my configuration diff:

```
# diff

# version
# Betaflight / KAKUTEF7 (KTF7) 3.5.3 Nov 14 2018 / 23:49:50 (61d5abf00) MSP API: 1.40

board_name KAKUTEF7
manufacturer_id 

# name

# resources

# mixer
mixer HEX6X

# servo

# servo mix

# feature
feature GPS
feature TELEMETRY
feature LED_STRIP
feature ESC_SENSOR

# beeper

# beacon

# map

# serial
serial 0 2 115200 57600 0 115200
serial 1 64 115200 57600 0 115200
serial 2 16384 115200 57600 0 115200
serial 3 2048 115200 57600 0 115200
serial 5 0 115200 57600 0 115200
serial 6 1024 115200 57600 0 115200

# led
led 0 10,15:ESD:CIW:10
led 1 15,7:ED:CIW:6
led 2 10,0:NED:CIW:4
led 3 5,0:NWD:CIW:4
led 4 0,7:WD:CIW:2
led 5 5,15:SWD:CIW:10

# color

# mode_color

# aux
aux 0 0 0 1700 2100 0 0
aux 1 2 1 1700 2100 0 0

# adjrange
adjrange 0 0 2 900 2100 12 2 0 0

# rxrange

# vtx

# rxfail

# master
set gyro_sync_denom = 2
set gyro_use_32khz = ON
set acc_trim_pitch = 14
set mag_declination = 2180
set serialrx_provider = FPORT
set serialrx_inverted = ON
set serialrx_halfduplex = ON
set blackbox_p_ratio = 0
set blackbox_device = SERIAL
set min_throttle = 1070
set dshot_idle_value = 200
set use_unsynced_pwm = OFF
set motor_pwm_protocol = DSHOT1200
set motor_pwm_rate = 480
set battery_meter = ESC
set gps_provider = UBLOX
set gps_sbas_mode = EGNOS
set gps_auto_baud = ON
set pid_process_denom = 1
set frsky_unit = METRIC
set osd_warn_core_temp = OFF
set osd_vbat_pos = 2436
set osd_tim_2_pos = 2453
set osd_flymode_pos = 2380
set osd_current_pos = 2443
set osd_warnings_pos = 329
set vtx_band = 0
set vtx_channel = 0
set vtx_power = 0
set vtx_freq = 0
set vcd_video_system = PAL

# profile
profile 0

set dterm_notch_cutoff = 0
set vbat_pid_gain = ON
set anti_gravity_gain = 3000
set d_level = 80

# rateprofile
rateprofile 0```
```

I can see UART4 TX in both `resource` and `resource list`, so this is (hopefully) not a resource conflict. The other UART devices I have configured (GPS, RX, ESC telemetry, RunCam Split) work, AFAICT (although for RunCam this has been hard to determine throughout the past ~year). All 6 motors work too. Video works; the VTX appears to be set to A1 at 25 mW, i.e., something default. (It's not in pit mode.)

In the manual for Kakute F7 AIO I've spotted a recommendation to put SmartAudio on UART**1** TX. (Sadly I've only spotted it after having soldered my "usual" UART layout on.) Can UART1 be somehow special on na F7? UART1 currently has GPS on it, which works just fine. Is this worth the hassle of re-soldering and retries (perhaps first on UART6, which is free at the moment) or are there other things to check?

andrejpodzimek, 2018-12-17:
I've just found a stale issue #6676 that might be related. But there are differences:

* I haven't seen a "softlock". Would that look like a freeze? Haven't seen that; not even once.
* This board (Kakute F7 AIO) doesn't have UART5 accessible and/or listed in the configurator, so there's no way to test UART4 and UART5 at once.

jflyper, 2018-12-17:
@andrejpodzimek Did you try without the "ideal" thing; i.e., straight connection from TX4.

andrejpodzimek, 2018-12-17:
@jflyper No, thus far I've just tried it with UART6 and it didn't work either. (So it's not like #6676, I suppose.) I'll try to short the "ideal" thing from the VTX's audio to UART6 TX to see what happens.

andrejpodzimek, 2018-12-17:
The good news is that the data connection works. The bad news is that analog audio probably just won't work, because it always breaks the data connection.

I looked at "the ideal thing" with my oscilloscope. The signal had peaks around 3.2 V on the UART side. On the SmartAudio side, behind the 4.7 kΩ, the signal was reasonably clear, but only reached ~130 mV, which is completely out of specs.

Then I tried to remove the "ideal thing" completely and connected the camera's audio output, the UART and the SmartAudio. It just didn't work and the UART signal had around 320 mV this time; still not enough.

Then I added just a 100 nF capacitor between the camera's audio pad and the rest. That didn't work either; the signal looked somewhat rounded and distorted. I don't recall what its peak values were, but the VTX just didn't respond to it at all.

Ultimately I disconnected the audio altogether :angry: and just connected UART6 TX with SmartAudio, with no parts in between. That way SmartAudio works, I could set channels using the OSD menu.

OK, this doesn't look like a bug. Maybe the "ideal thing" is ideal for original TBS components (FC + camera + VTX), but in my case only the VTX is from TBS.

closed 2018-12-17 by andrejpodzimek

jflyper, 2018-12-17:
@andrejpodzimek 
130mV peak suggest that the effective voltage divider ratio is 0.13/3.3 = 1:25, which means lower side of the divider is around R200, which further suggest there is an internal DC resistance of around R225 inside the Unify.

Now, R225 is abnormally low for input side of any descent circuit. It should be much much higher.

I suggest that you make sure the 1K8 is actually 1K8 or similar.

andrejpodzimek, 2018-12-18:
It was exactly that value, both according to the manufacturer and according to my multimeter.

There were more surprising problems though. The 100 nF capacitor would get “downgraded” do ~11 nF when connected between the camera’s audio output and the SmartAudio. The cause for this was some continuity in parallel with the capacitor, albeit at 80 kΩ. (Then the measured value was something undefined rather than capacitance.) In other words, the camera’s audio pad is (without any wires) connected to T6<—>SmartAudio over ~80 kΩ of resistance. That quite likely invalidates some of the resistance and capacitance considerations.

Jetrell, 2019-09-24:
@andrejpodzimek  Just out of interest. Do you find a fix to get the Split 2 audio and SA working together?
The suggested TBS components mod works with the Split 1.  But I have had no success either, with the Split 2 and later models.

andrejpodzimek, 2019-09-24:
I just don’t use audio. VTXs with a built-in microphone don‘t need it anyway and many VTX monitors don’t support it either. Finding a combination of resistors and capacitors that could possibly work is way beyond my capabilities and not worth the effort anyway. (Needless to say that I don’t fly FPV, ever, and only have video installed to record OSD numbers. So I don’t miss audio all that much.)

andrejpodzimek, 2019-10-06:
Just for the record: There might be [a glimmer of hope](https://banggood.app.link/zS9o7CrBz0). That VTX has, IIUC, separate inputs for SA and audio, i.e., it should also work with RunCam Split >1. Unless it has “the ideal connection” installed internally, which would just fail the same way again. I have ordered one; looking forward to trying it out.