betaflight/betaflight #7236: Improve Gps documentation

bluehallu, 2018-12-18:
I've just gone through installing GPS and found a lot of things that had no documentation and no information online. Sorry for the braindump 😅

- What is frsky_gps_format? Code seems to use this to change how latitude/longitude are reported in telemetry, but I see no difference.
- GPS sensors won't appear until GPS acquires a lock, and the Taranis will complain of sensors lost whenever lock is lost. Is this intentional?
- What is sensor 420? Looking at the code it appears it's distance to home?
- What units are used for the sensors? It seems that a lot of them use strange units: Gps altitude is in cm even though Taranis reports it in meters, while gps speed is in... knots? 
- Is gps_ublox_use_galileo desirable? Where and where not?

McGiverGim, 2018-12-18:
I can only answer to the galileo part: when activated, this removes the QZSS (Japanese) and replaces it for the Galileo (European) system. This only works if auto config for GPS enabled, and the GPS configured as UBLOX

So if you are not Japanese, usually yes, specially if you live in Europe.

You can activate it and see if you see more satellites than before or not.

mikeller, 2018-12-18:
@bluehallu:
- `frsky_gps_format` is indeed the format of the GPS data sent over FrSky Hub telemetry (**not** SmartPort);
- sensor 0x420 is 'distance to home';
- (assuming you are using SmartPort), FrSky specifies altitude / ascent / sink rate in cm and cm/s. If your TX firmware reports them in m without conversion this is a bug in your TX firmware. GPS speed is specified to be knots.

Feel free to collate this information and put it into a wiki page or open a pull request for a page under `docs/`, we encourage community contributions of all types, including documentation.