betaflight/betaflight #7237: Fuel range OSD Warning

bluehallu, 2018-12-18:
When GPS is enabled, use remaining mah, distance to home, current consumption and gps speed to determine if there's enough battery left to arrive home. If not, display a warning (ideally before it's too late)

jflyper, 2018-12-18:
That's probably unusable 🤣

mikeller, 2018-12-18:
Even with an airspeed sensor, 'enough battery left to arrive home' still has so many variables (ability to fly back home in a direct line vs. having to retrace the flight path, changing winds, flight speed).

Have a look at the battery capacity / battery current based `Remaining Time Estimate`. This actually does a decent job of predicting remaining flight time. If you turn around whenever the output of this is roughly equal to the total armed time you should have enough battery left to get back safely.

etracer65, 2018-12-18:
The only thing that's probably reasonable is some kind of "bingo" fuel warning. Triggered when the distance to home is >= distance flown (available in 4.0) and battery remaining is 50%. Even this is not terribly accurate as the distance flown is total traveled distance and distance to home is "as the crow flies" so it would be cutting it really close.

bluehallu, 2018-12-18:
I of course don't expect this to be perfect, but if I can get an estimated flight time, I surely can get an estimated range?

etracer65, 2018-12-18:
We have no idea how the remaining mah relate to distance because we don't know how fast you'll fly and what that effect will have on current consumption. Estimating based on how far flown already vs. actual consumption might work but will be wildly inaccurate (see the "bingo" fuel suggestion above).

butterSnake, 2018-12-19:
i set my mah warning to 40% leaving me 60% to get home.. basically you want to turn around at that point or your gona run the risk of not making it back

jflyper, 2018-12-19:
> i set my mah warning to 40% leaving me 60% to get home.. basically you want to turn around at that point or your gona run the risk of not making it back

Your proposed method of GPS based estimation is probably no better than this, may be worse in consequences.

closed 2019-01-25 by stale[bot]