betaflight/betaflight #7238: Naming and renaming resource and related command options

jflyper, 2018-12-18:
3.x is using

> `resource` : Show current configuration.
> `resource list` : Show current active allocation (assignment, usage, ...).

I think this is bit misleading, and want to change these to

> `resource config` : Show current configuration.
> `resource active` : Show current active allocation.

A new `dmaopt` command will then use the following options.

> `dmaopt config` : Show all current configuration.
> `dmaopt <periph> <index> config` : Show current configuration for `<periph> <index>`
> `dmaopt <periph> <index> list` : Show possible options for `<periph> <index>`
> (`dma` will show active allocation/usage.)

Similar for `timer` command.

Thoughts?

mikeller, 2018-12-18:
Not sure about all of this.

The pattern that is used throughout pretty much all CLI commands (including `set`) is that an invocation without parameters returns the current configuration. While this is not super intuitive, at least it is consistent. I would be in favour of staying with this pattern for resource management commands.

For commands with list based parameters, `<command> list` is usually used to enumerate the list of possible parameter values (see `feature`, `beeper`, `beacon`). The way `list` is used with the `resource` command is a different case, as we are not enumerating a parameter value list, but showing the currently active assignments (based on configuration and precedence). But I would prefer to use a verb as the subcommand for this, how about `showactive`?

So to summarise:

- `resource`: Show current configuration (no change);
- `resource showactive`: Show current active allocation;
- `resource <resource> <index> <pin>`: Assignment (No change);
- `dmaopt`: Show all current configuration;
- `dmaopt <periph> <index>`: Show current configuration for <periph> <index>;
- `dmaopt <periph> <index> list`: Show possible options for <periph> <index>;
- `dmaopt <periph> <index> <opt>`: Assignment.

jflyper, 2018-12-19:
Okay! Not bad ;)

mikeller, 2019-01-18:
@jflyper: Is this still needed?

closed 2019-01-25 by stale[bot]