betaflight/betaflight #7243: GPS RTH issue: Max altitude doesn't reset with disarm

aowi7280, 2018-12-18:
**Describe the bug**
After landing and disarming, the new location is set as the GPS home position (correct behavior), but the max altitude is kept from the previous flight. It should reset the altitude to zero.

**To Reproduce**
Arm, fly to new location and note max altitude reached, land, disarm. Rearm without disconnecting battery, fly away and activate GPS-Rescue. The craft will climb to the previous flight's altitude+the GPS Altitude setting, but return to the new home location.

**Expected behavior**
After landing and disarming, both the Home position and Max altitude reached should reset to the new position and zero altitude.

**Flight controller configuration**Target: OMNIBUSF4SD
Name/Version: 4.0.0 #1283
Binary: betaflight_4.0.0_OMNIBUSF4SD-1283-3b93361ac.hex
Date: 12-12-2018 05:43

**Setup / Versions**
 - Flight controller Omnibus F4 Pro from RTFQ
 - Other components [RX, VTX, brand / model for all of them, firmware version where applicable];
 - how are the different components wired up.

**Additional context**
Add any other context about the problem here.

etracer65, 2018-12-18:
Yes, this looks like a bug. I'll look into it.

etracer65, 2018-12-18:
@aowi7280 Can you please test this version to see if it fixes the problem?

[betaflight_4.0.0_OMNIBUSF4SD.hex.zip](https://github.com/betaflight/betaflight/files/2692576/betaflight_4.0.0_OMNIBUSF4SD.hex.zip)

aowi7280, 2018-12-18:
OK, I will flash it in tonight.

aowi7280, 2018-12-19:
Flashed in OK, but wind speeds of 35 to 50 MPH today. I will test it as soon as I can.

etracer65, 2018-12-20:
This is unrelated.  So you have the sanity checks enabled? What's your value for `gps_rescue_sanity_checks` ?

aowi7280, 2018-12-20:
Thanks for fixing that first problem. No, Sanity_checks = off
I had it drop out of the sky one time and now make sure it is off.

etracer65, 2018-12-20:
That's the reason. The minimum distance to home when manually activating is a sanity check.

aowi7280, 2018-12-20:
Not a bug. Good to know. Thanks.

PR #7244 (2018-12-21): Reset GPS Rescue max altitude and distance on arming
Fixes #7243
closed 2018-12-20 by aowi7280