betaflight/betaflight #7252: GPS Rescue: Sanity Check while climbing?

McGiverGim, 2018-12-19:
**Describe the bug**
When activated the GPS Rescue, the quad started to get the desired height. But after 10 seconds, when just arrived to the desired height, the Sanity Check was fired disarming the quad (this is a supposition, no event in the blackbox log when disarmed).

**To Reproduce**
I'm not too sure. Activate the GPS Rescue. The desired GPS Rescue height must be reached in 10 seconds (the desired height must be very high or the maximum throttle low to let reproduce this easily)

**Expected behavior**
The sanity check while climbing must only check if the quad is higher each time, not if the direction or other validations are ok.

**Flight controller configuration**
```
# diff

# version
# Betaflight / MATEKF722 (MKF7) 4.0.0 Dec 15 2018 / 09:12:20 (497ede203) MSP API: 1.41

board_name MATEKF722
manufacturer_id 

# name
name MII-MIGMUL

# resources
resource MOTOR 1 C08
resource MOTOR 2 C06
resource MOTOR 3 C09
resource MOTOR 4 C07

# mixer

# servo

# servo mix

# feature
feature GPS

# beeper

# beacon
beacon RX_SET

# map

# serial
serial 0 2 115200 57600 0 115200
serial 2 2048 115200 57600 0 115200
serial 4 16384 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1400 1850 0 0
aux 1 1 1 1275 1425 0 0
aux 2 1 255 1900 2100 0 0
aux 3 2 1 1425 1600 0 0
aux 4 46 1 1900 2100 0 0
aux 5 13 1 1900 2100 0 0
aux 6 26 0 1400 1850 0 0
aux 7 28 1 900 1100 0 0
aux 8 35 1 1600 1750 0 0
aux 9 36 0 1675 1850 0 0

# adjrange

# rxrange

# vtx

# rxfail
rxfail 5 s 1500

# master
set gyro_sync_denom = 2
set gyro_lowpass2_hz = 100
set gyro_use_32khz = ON
set align_mag = CW270FLIP
set mag_declination = -110
set mag_calibration = -66,-308,-267
set max_check = 1950
set serialrx_provider = FPORT
set serialrx_inverted = ON
set serialrx_halfduplex = ON
set blackbox_p_ratio = 64
set motor_pwm_protocol = DSHOT600
set failsafe_delay = 3
set failsafe_off_delay = 20
set failsafe_throttle = 1300
set failsafe_procedure = GPS-RESCUE
set bat_capacity = 1800
set vbat_detect_cell_voltage = 55
set beeper_dshot_beacon_tone = 2
set small_angle = 180
set gps_provider = UBLOX
set gps_sbas_mode = EGNOS
set gps_ublox_use_galileo = ON
set gps_rescue_throttle_hover = 1400
set gps_rescue_sanity_checks = RESCUE_SANITY_FS_ONLY
set gps_rescue_min_sats = 0
set gps_rescue_min_dth = 50
set deadband = 3
set yaw_deadband = 5
set pid_process_denom = 2
set osd_warn_core_temp = OFF
set osd_rssi_alarm = 45
set osd_cap_alarm = 1400
set osd_tim1 = 0
set osd_tim2 = 1793
set osd_vbat_pos = 2433
set osd_rssi_pos = 2081
set osd_tim_1_pos = 54
set osd_tim_2_pos = 2455
set osd_flymode_pos = 2145
set osd_throttle_pos = 2424
set osd_vtx_channel_pos = 2049
set osd_crosshairs_pos = 2253
set osd_ah_sbar_pos = 2254
set osd_ah_pos = 2126
set osd_current_pos = 2400
set osd_mah_drawn_pos = 2370
set osd_craft_name_pos = 2442
set osd_gps_speed_pos = 2198
set osd_gps_lon_pos = 2097
set osd_gps_lat_pos = 2065
set osd_gps_sats_pos = 2060
set osd_home_dir_pos = 2126
set osd_home_dist_pos = 2127
set osd_compass_bar_pos = 2154
set osd_altitude_pos = 2166
set osd_pid_roll_pos = 359
set osd_pid_pitch_pos = 359
set osd_pid_yaw_pos = 359
set osd_debug_pos = 391
set osd_power_pos = 321
set osd_pidrate_profile_pos = 344
set osd_warnings_pos = 2345
set osd_avg_cell_voltage_pos = 76
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_battery_usage_pos = 302
set osd_disarmed_pos = 2379
set osd_nheading_pos = 106
set osd_nvario_pos = 1
set osd_esc_tmp_pos = 82
set osd_esc_rpm_pos = 83
set osd_stat_max_dist = ON
set osd_stat_min_batt = OFF
set osd_stat_endbatt = ON
set osd_stat_max_alt = ON
set osd_stat_bbox = OFF
set vtx_band = 5
set vtx_channel = 2
set vtx_power = 2
set vtx_freq = 5695
set vcd_video_system = NTSC

# profile
profile 0

set dyn_lpf_dterm_min_hz = 120
set dyn_lpf_dterm_max_hz = 180
set dterm_lowpass_hz = 120
set anti_gravity_gain = 8000
set feedforward_transition = 15
set iterm_rotation = OFF
set iterm_relax = RPY_INC
set p_pitch = 53
set d_pitch = 37
set f_pitch = 80
set d_roll = 35
set f_roll = 80

# rateprofile
rateprofile 0

set roll_expo = 20
set pitch_expo = 20
set roll_srate = 75
set pitch_srate = 71
set yaw_srate = 61
```

**Setup / Versions**
 - Flight controller: Matek F722
- GPS: Radiolink SE100

**Additional context**
I suppose that the GPS Rescue first tries to climb, and until the desired height is reached, it does not start to move in the home direction. I have used the GPS Rescue before without problems. I don't know if something has changed lately.

Casualty or not, the desired height was 50 meters and the quad just disarmed at this height, without trying to point home. I don't know if because it had no time or other thing went wrong.

Here you have the Blackbox log if it helps:
https://drive.google.com/open?id=1S1m0LzSJGX6jhOGInoQhK5ANInxwfrIW

I have a Video too if needed.

McGiverGim, 2018-12-19:
Checking the GPS Rescue documentation, it says:

> When Rescue Mode is activated, your quad will point home and try to climb to a safe altitude relative to your takeoff point.

I have not seen in the Video that the quad point home, only started to climb. So maybe other thing went wrong.

McGiverGim, 2018-12-19:
More info: seeing the video, the quad points home but it goes backward three or four meters while climbing. Maybe the sanity checks detect this and disarms the quad. Is the GPS Rescue going forward while climbing? If yes then I have something bad configured. If not, then it has no sense the check of backwards at this time.

etracer65, 2018-12-19:
Could be the flyaway check. #7040 

@TonyBlit any thoughts?

TonyBlit, 2018-12-19:
Without having checked the code (yet), it indeed looks like the stalled/flyaway checks are getting wrongly triggered.
In my experience, the quad points home since the very beginning, unless the current direction is not accurate, then once the climb altitude is reached, rescue starts to fly straight which allows for the direction to be calibrated properly and the attitude gets corrected over time.
I don't recall seeing my quad going upwards perfectly, it always seems to be tilted so it gets some ground speed and that's probably why I haven't hit this problem before.
But yes, those stalled/flyaway checks should be done ONCE the climb altitude is reached, or even better, during that phase the checks should monitor if the quad is actually gaining altitude (think the battery is dying or something).
I will check the code later and I will confirm.

Yamaford, 2018-12-19:
How are you checking Rescue mode? You have set gps_rescue_sanity_checks =
RESCUE_SANITY_FS_ONLY .
So unless your initiating a FS situation, sanity check shouldn't apply. If
you are initiating FS via a switch, how is your failsafe setup? I guess I
could answer my own question with the diff provided.

etracer65, 2018-12-19:
If the quad was in fast forward flight away from the home point when GPS Rescue was triggered it will yaw towards home but be initially still heading away. The climb to altitude won’t do anything to stop this as it’ll just be in self-level. Probably needs to be a delay after reaching altitude before checking flyaway.

McGiverGim, 2018-12-19:
Yes, that is what I thought. Thanks for the analysis and the fast fix!!!

TonyBlit, 2018-12-20:
Reviewing your config, I've seen this:
set gps_rescue_min_sats = 0 
This is a bit contradictory with enabling sanity checks: this allows you to start flying without a valid home fix, and this combined with the fact that you can stop receiving valid fixes (or get wrong ones) during rescue is a madness for the sanity checks.
I'm a bit disappointed with how easy is to mess things up with a bad configuration+handling and then blame rescue (not saying it's your case @McGiverGim).

TonyBlit, 2018-12-20:
Totally agree, and I've been tempted to do so on some occasions, but it's a risk that only a fraction of pilots are fully aware. For instance and I know this is an unsolvable extreme case, I've seen people complaining for misbehaviour of rescue when they disabled sanity checks... I was simply thinking out loud.

McGiverGim, 2018-12-20:
A good option can be show a warning, 2 or 3 seconds, when taking off if not home position fixed or no minimum gps rescue satellites fixed. Maybe a CLI command to let decide between blocking the take off or show a warning.

TonyBlit, 2018-12-20:
Yeah, any of those ideas would be a good enhancement. I'd be happy to work on them after holidays.

closed 2018-12-21 by mikeller