betaflight/betaflight #7258: Motors twitch when armed or in flight every 6 seconds when using FrSky D16 mode

ifixwell, 2018-12-21:
Using Frsky D16 mode I get twitching motors when armed and 0 throttle or when flying. Happens about every 5 seconds repeatedly. I tried using BF 4.0, 3.5.4 and 3.5.3

Does not happen with Frsky D8 mode. 

Mobula 7 Came with BF 4.0
CrazybeeF3FR

mikeller, 2018-12-21:
@ifixwell: Can you please provide the information that you were asked to provide when opening this issue?

ifixwell, 2018-12-21:
New to github. Which info should I include?

etracer65, 2018-12-21:
> **Describe the bug**
> A clear and concise description of what the bug is.
> 
> **To Reproduce**
> Steps to reproduce the behavior:
> 
> **Expected behavior**
> A clear and concise description of what you expected to happen.
> 
> **Flight controller configuration**
> Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/)
> 
> **Setup / Versions**
>  - Flight controller [what type is it, where was it bought from];
>  - Other components [RX, VTX, brand / model for all of them, firmware version where applicable];
>  - how are the different components wired up.
> 
> **Additional context**
> Add any other context about the problem here.
>

ifixwell, 2018-12-21:
Describe the bug
Using Frsky D16 mode I get twitching motors when armed, 0 throttle or when flying. Happens about every 5 seconds repeatedly. I tried using BF 4.0, 3.5.4 and 3.5.3

To Reproduce
Used Frsky D8 mode and it did not happen. Switched back to D16 and it twirches.

Flight controller configuration
Mobula 7 came with BF 4.0, (I have tried several other versions of BF, All exhibit the twitch every 5 seconds in D16 mode).
CrazybeeF3FR

Here is the Diff I am using.
\\\
# diff all
# Betaflight / CRAZYBEEF3FR (CBFR) 3.5.3 Nov 14 2018 / 23:20:22 (61d5abf00) MSP API: 1.40

board_name CRAZYBEEF3FR
manufacturer_id 
mcu_id 003200285734571120363930
defaults nosave
name Meeebula7

feature -ANTI_GRAVITY
feature -DYNAMIC_FILTER
map TAER1234
serial 2 2048 115200 57600 0 115200
aux 0 0 0 1300 2100 0 0
aux 1 1 1 1200 1550 0 0
aux 2 28 1 1550 2100 0 0
aux 3 35 2 1550 2100 0 0
set rc_smoothing_type = FILTER
set motor_pwm_protocol = DSHOT600
set vbat_min_cell_voltage = 30
set vbat_warning_cell_voltage = 29
set ibata_scale = 1175
set osd_warn_esc_fail = OFF
set osd_vbat_pos = 2146
set osd_rssi_pos = 2114
set osd_tim_1_pos = 54
set osd_tim_2_pos = 2134
set osd_flymode_pos = 2456
set osd_throttle_pos = 2423
set osd_crosshairs_pos = 2253
set osd_ah_pos = 2126
set osd_current_pos = 2401
set osd_mah_drawn_pos = 2434
set osd_craft_name_pos = 2445
set osd_warnings_pos = 329
set osd_disarmed_pos = 2346
set frsky_spi_tx_id = 20,34
set frsky_spi_offset = -16
set frsky_spi_bind_hop_data = 4,38,72,106,140,174,208,7,41,75,109,143,177,211,10,44,78,112,146,180,214,13,47,81,115,149,183,217,16,50,84,118,152,186,221,19,53,87,121,155,189,223,22,56,91,124,158,0,0,0
set frsky_x_rx_num = 5
profile 0

set p_yaw = 45
set i_yaw = 100
profile 1

set p_yaw = 45
set i_yaw = 100
profile 2
\\\

Thanks for the help.

mikeller, 2018-12-21:
@ifixwell: Welcome to GitHub. Next time, please do a search for your board name / symptoms before opening an issue.

This is a duplicate of #7153, with the solution presented here: https://github.com/betaflight/betaflight/issues/7153#issuecomment-443403496

Closing as duplicate.

closed 2018-12-21 by mikeller

ifixwell, 2018-12-22:
Thanks for the reply. In viewing the thread it was mentioned to disable telemetry that it is not supported on the board. I unchecked in BF. Telemetry still shows and quad still twitches every 5 secs. Did I miss something?

This is a duplicate of #7153<https://github.com/betaflight/betaflight/issues/7153>, with the solution presented here: #7153 (comment)<https://github.com/betaflight/betaflight/issues/7153#issuecomment-443403496>

Closing as duplicate.

mikeller, 2018-12-22:
@ifixwell:

> set motor_pwm_protocol = DSHOT600

This is another problem - the hardware of the ESCs on this board is too slow to handle Dshot600. Try Dshot300.

ifixwell, 2018-12-22:
I have tried Dshot300. It doesn't seem to have any effect.