betaflight/betaflight #7270: Table of F# compatibility

laurentopia, 2018-12-22:
Currently, finding which features are available in F3 requires a lot of digging, in fact I couldn't find wether F3 supports GPS again in 4.0
What would solve that is a table of compatibility in README.md

mikeller, 2018-12-22:
@laurentopia: Unfortunately, the flash space shortage on F3 is so dire that we have to resort to removing individual features from a lot of the F3 boards based on what hardware (like BARO, OSD) they have to support. This means for a table to be useful it would have to be on a per-target basis, and would change pretty much weekly.

Feel free to compile such a table, and then open a pull request to add it.

wind0r, 2018-12-22:
Such Information is always in the release notes. If someone wants to use nightly builds they as always should track the changes on Github. Since those will be the future release notes.

laurentopia, 2018-12-25:
Thanks for your answers. I am just familiarizing myself with betaflight and this helps.
So if I get that right, only CRAZYBEEF3FR, CRAZYBEEF3FS, FRSKYF3, FURYF3, FURYF3OSD, OMNIBUS, SPRACINGF3, SPRACINGF3EVO, SPRACINGF3MINI, SPRACINGF3NEO are missing GPS?

wind0r, 2018-12-25:
Depending on which version you are talking about.
For the 3.5.x release it sounds correct. 

and if you have a look at 4.0 you will notice that currently gps got "removed" for every f3 target to make development easier (it get readded for those targets that have space befor the next release) https://github.com/betaflight/betaflight/blob/master/src/main/target/common_pre.h#L218

closed 2018-12-25 by wind0r

laurentopia, 2018-12-27:
I see, it directly depends on the flash size, clever.
I'll just get an f4, make life easier.