betaflight/betaflight #7273: Can "Core Temperature" be available as a Telemetry option

0crap, 2018-12-23:
For logging purposes, on the Taranis radio, it would be nice to have Core Temperature available as a Telemetry option.

The Tmp1 sensor I can get now doesn't make sense.
Maybe replace that for Core Temperature?

[image]
[image]

klutvott123, 2018-12-23:
Tmp1 is used for flight modes and arming status. The number 3 to the left is just a counter. The lua script uses it to detect if there's a telemetry connection, but it doesn't care about the values it contains. If it's used for something other than that i don't know. If it's for logging purposes, maybe it would make more sense to have the blackbox log it(maybe it's already capable of it, i don't know)?

mikeller, 2018-12-23:
We can have both. `debug_mode = ADC_INTERNAL` will log the temperature into debug field 0. Also, we have more than one temperature slots available, we can send the core temperature in a free one.

0crap, 2018-12-23:
Thx for the comments!

Didn't think of blackbox, I'll check that.
However it would be better to have it on Telemetry, that way you can also make nice alarms for high temp on the Taranis. (Using logical switches.)

I tried debug_mode = ADC_INTERNAL however that gives an error. (On MATEKF405 @ BF 3.5.4)
There is something like debug_mode = core_temp but that doesn't do anything with the "Tmp1" sensor.

mikeller, 2018-12-23:
@0crap: Did you try `set debug_mode = ADC_INTERNAL`?

> There is something like debug_mode = core_temp but that doesn't do anything with the "Tmp1" sensor.

`debug_mode = core_temp` is news to me. My suggestion about using a Tmp sensor for this was meant as 'we'd have to implement this first'.

0crap, 2018-12-23:
> 
> 
> @0crap: Did you try `set debug_mode = ADC_INTERNAL`?
> 
> > There is something like debug_mode = core_temp but that doesn't do anything with the "Tmp1" sensor.
> 

[image]

> `debug_mode = core_temp` is news to me. My suggestion about using a Tmp sensor for this was meant as 'we'd have to implement this first'.

Ahhh, well since you and all devs have your hands full on getting 4.0 done, I'm not going to ask anything. :-)

mikeller, 2018-12-23:
Drats. That got renamed between 3.5 and 4.0, and I was looking at 4.0. 
`set debug_mode = core_temp` is what you want, the temperature value will be in debug field 0.

0crap, 2018-12-23:
Perfect! I'm going to try that.
Usually debug_mode = gyro_scaled, for me to play with BB and Plasmatree.

I sure hope it will end up some day as a Telemetry sensor.  👍

mikeller, 2018-12-24:
Merry Christmas!

#7278:

[image]

0crap, 2018-12-24:
Merry Christmas! 

Fantastic looking picture with the Tmp1 sensor. 👍 
For sure one to turn on for logging in the Taranis and make some nice Temp alarms.
Can't wait to test it out when it becomes available as a nightly build.

I'll test it and let you know!
Thx again for the hard work.

klutvott123, 2018-12-24:
Nice! I'll try it out later. Merry Christmas!

mikeller, 2018-12-26:
@0crap: Did you have a look at the comments in #7278? If you're using SmartPort, you're going to have to disable some other sensors to be able to see this one, as the telemetry link seems to be overloaded - or just move to FPort that doesn't have this problem.

0crap, 2018-12-26:
Sorry, no I did not.
Currently I'm on SmartPort and only use/log a few sensors on my Taranis. (Deleted all sensors I don't want.)
Like this:
[image]
I would love to switch to FPort but that is a bit hard because of all the inversion crap on my F405 controller. And the lack of a decent inversion hack on the XSR-E receiver.

So I think from #7278 I need to give a CLI command to use the sensors I need and include Tmp1 for the core temp, right?

mikeller, 2018-12-26:
Yes, you will need a CLI command `set telemetry_disabled_sensors = 524280 should work for what you want. (Deleting sensors on the TX will not actually stop the flight controller from sending values for these sensors, and clogging up the telemetry link.)

0crap, 2018-12-26:
Thx, I understand.
Back on development build # 1312 and typed set telemetry_disabled_sensors = 524280
Deleted all sensors on the Taranis and did a new discovery.
Almost there I guess, a lot of sensors I don't use are not there anymore.
[image]

Is there some kind of table I can lookup to see which CLI command I need for my target?
Using a MATEKF405 (MKF4).

mikeller, 2018-12-26:
Exactly @klutvott123. There is a more elaborate way to configure the telemetry sensors in the works in #7284, but that's not quite ready yet.
Re not getting sensor values, there is something weird going on when using SmartPort, it looks like a lot of the sensors are missed when multiple sensors are enabled.

klutvott123, 2018-12-29:
@0crap Good! 👍 Too bad you have to pick between core temp and VFAS. 😞 

@mikeller I wonder if it's a bug or a "feature". Wouldn't surprise me if it only affects some smartport receivers. I keep thinking that it would be so much easier if they just made the receivers as transparent as possible.

mikeller, 2018-12-29:
@klutvott123: It definitely expresses as a bug. The RX firmware implementations seem to be fairly inconsistent, as we've seen with the '125 bug'.

klutvott123, 2018-12-30:
I see. I just assumed it would break right away, but if it behaves like you say, it will probably be fine. 👍 
Please let us know what frsky has to say about this!

Hydroculture, 2018-12-30:
as suggested i disabled all sensors but VFAS with 

set telemetry_disabled_sensors = 1048574

VFAS is found, but shown as 0.00V as described here https://github.com/betaflight/betaflight/issues/7316

best
Holger

0crap, 2018-12-31:
@Hydroculture Interesting, when I do set telemetry_disabled_sensors = 1048574
I get VFAS with a proper reading, but also A2 and A4.
I believe that is from the FC and should not show with 1048574
Anyway, when VFAS is found I do have a proper reading on a XSR-E with version 170619 EU-LBT firmware.

[image]

0crap, 2018-12-31:
@mikeller On the FrSky website I see there is newer firmware for my XSR-E receiver.

**2018-11-06 | 180322 | Supports telemetry enable/disable function.**
Should I try??

0crap, 2018-12-31:
Well, not the hardest quad to reach for the receiver, so I flashed the latest XSR 180322 LBT firmware.

Unfortunately same behaviour. :-(

klutvott123, 2018-12-31:
@0crap At least now you have the latest and greatest firmware. I wonder what they mean by "enable/disable". A more detailed description wouldn't hurt.

0crap, 2018-12-31:
👍 latest is always better. ;-)
Fuzzy description indeed.

klutvott123, 2019-01-01:
@mikeller Shouldn't there be a ``*clearToSend = false`` below this? https://github.com/betaflight/betaflight/blob/4778ad6c0fa4cf2c145513601b1378a98901e6cd/src/main/telemetry/smartport.c#L801

For smartport, if ``*clearToSend`` isn't cleared, it will just loop back, increment the id and attempt to send the next sensor which will be VFAS in our case. Only then will ``*clearToSend`` be cleared and it breaks out of the loop. How the receiver reacts to getting the data for two sensors at once i don't know, but i bet it doesn't like it. For FPORT this doesn't matter since the loop only runs once per telemetry request.

Asizon, 2019-01-01:
@Hydroculture I can create own target for you,come to Slack.

klutvott123, 2019-01-01:
Update: I changed over to smartport and tested it on my quad. I see the same behaviour that you guys are seeing with missing sensors. Implemented the fix i described above, and now it works like it should. With ``telemetry_disabled_sensors`` set to the default value i get all the sensors i expect to see + the new core temperature sensor. I have made a pull request for it.

0crap, 2019-01-01:
@klutvott123 🥇 💯 
Can't wait to test it! Thx!

0crap, 2019-01-01:
Could not wait so I build my own target on my local PC with your correction.
Fix confirmed!
On BF defaults it does find twice the Tmp1 sensor. But that's it, not again and again new sensors as seen before.
Simple delete sensor on the taranis and all OK.
[image]

I also tried 'set telemetry_disabled_sensors = 524280'.
That gave **exactly** as expected, only one Tmp1 sensor with the core temp. 👍 
Thumbs up. :-)

[image]

klutvott123, 2019-01-01:
@0crap Good to hear it's working for you. Thanks for testing it! You could also just rename the core temperature Tmp1 to something more suitable. That they show up with the same name is just an opentx thing i think. If you go into the edit menu you'll see that they have different ids.

@Hydroculture If you still need it 🙂 
[betaflight_4.0.0_OMNIBUSF4.zip](https://github.com/betaflight/betaflight/files/2719472/betaflight_4.0.0_OMNIBUSF4.zip)

Hydroculture, 2019-01-02:
@klutvott123 da danke ich aber schön! (= thank you very much)
Im at work atm. Looking forward to test out tonight.

mikeller, 2019-01-08:
Implemented in #7278.

closed 2019-01-08 by mikeller

0crap, 2019-01-08:
Thx all! 👍

Hydroculture, 2019-01-09:
Thank you very much!