betaflight/betaflight #7275: Channel AUX endpoints/range RSSI Spektrum IX12 SPM4649T

rolial, 2018-12-23:
Hi,
I set the AUX channels to be between 1000 to 2000 so when I select the AUX channel to broadcast RSSI the channel is 2012.
This happens on all auxiliary channels.
I'm using TX IX12 with the RX 4649T (Firmware 1.1.10) and FC Matek F405-CTR.
After the selected channel I can not set the same for 2000.
The auxiliary channel marking 2012 RSSI is not real. Right?

(video) https://drive.google.com/open?id=1oz50oOhkPxVfx24QHPqdrCCo1OxKAsMn

diff

version

Betaflight / MATEKF405 (MKF4) 3.5.4 Dec 17 2018 / 12:06:27 (66b8ea1) MSP API: 1.40

board_name MATEKF405
manufacturer_id

name

resources

resource MOTOR 5 NONE
resource LED_STRIP 1 A15

mixer

servo

servo mix

feature

feature LED_STRIP
feature AIRMODE

beeper

beacon

map

map TAER1234

serial

led

led 0 6,2::A:0
led 1 6,3::A:0
led 2 6,4::A:0
led 3 6,5::A:0
led 4 7,2::A:0
led 5 7,3::A:0
led 6 7,4::A:0
led 7 7,5::A:0

color

mode_color

mode_color 6 1 2

aux

aux 0 0 0 1500 2100 0 0
aux 1 1 1 1700 2100 0 0
aux 2 2 1 1300 1700 0 0
aux 3 13 4 900 1700 0 0

adjrange

rxrange

rxrange 0 1159 1840
rxrange 1 1159 1840
rxrange 2 1159 1840
rxrange 3 1159 1841

vtx

rxfail

rxfail 9 s 1000

master

set acc_calibration = 0,68,-183
set mag_hardware = NONE
set rssi_channel = 10
set serialrx_provider = SRXL
set min_throttle = 1065
set motor_pwm_protocol = MULTISHOT
set motor_pwm_rate = 2000
set bat_capacity = 1300
set osd_vbat_pos = 2529
set osd_rssi_pos = 2497
set osd_ah_pos = 2221
set osd_current_pos = 2519
set osd_mah_drawn_pos = 2551
set osd_warnings_pos = 2409

profile

profile 0

rateprofile

rateprofile 0

AndersHoglund, 2018-12-23:
Some assumptions and misunderstandings here how RSSI works with a Spektrum RC system. RSSI is directly inserted into the selected AUX channel by the Rx. RSSI is not broadcast by the TX on any AUX channel at all, pointless to try and adjust endpoints on the Tx. Other RC systems send RSSI as a Rx -> Tx telemetry value, the Tx will then encode this value on a selected Aux channel and the BF will pick it up from there. This de-tour over the Tx via telemetry and  is  NOT done with a Spektrum system. Instead, as said, directly from Rx -> BF on the selected AUX ch.
A value of 2012 is perfectly normal and indicates full signal strength. As this is a non-issue and not a problem, I will close this.

closed 2018-12-23 by AndersHoglund

rolial, 2018-12-23:
OK understood!
Thank you @AndersHoglund