betaflight/betaflight #7277: A better option for for GPS rescue feature initial altitude.

rrk1, 2018-12-23:
In its current implementation, GPS rescue feature tries to raise copter to a minimum configurable altitude or to max achieved altitude + some meters.

I think this may be not optimal and can lead craft to crash. 

Scenario: We are flying near a mountain/hill. We got to the summit and completed a dive, yeah great footage. :) Now on the way to home pilot lost orientation and turns on rescue manually. Or we are cruising near home killing remaining battery capacity and just suddenly RX malfunctions. With current rescue behavior, copter will try to climb aggressively to the max altitude again, but possibly with not have enough mAH left in the battery. Result - it drops from high altitude in a difficult terrain with flat battery...

What is suggested - please add option for return altitude to be current altitude, or better to current + configurable X meters. In my opinion, this is optimal for most terrains around, at least until pilot does sane things :)

etracer65, 2018-12-24:
How is GPS Rescue supposed to know that you’re flying in mountains vs. you flew over some trees that are in your way back home?

rrk1, 2018-12-24:
Okay, minimum altitude set to fixed 50-100 m really takes care of trees. 

Let's look to another typical scenario. At start of flight the pilot does a punchout  to some hundreds of meters for some sightseeing. For all remaining flight max altitude will be recorded and used even with flat battery. Which is not optimal and dangerous.

rrk1, 2018-12-24:
Letting this option to be configurable like in INAV is the best IMHO. Last + X fits better for nature, maximum altitude might be better for abandoned buildings etc. Not much code complexity, just a few extra lines.

mikeller, 2018-12-24:
@rrk1: iNav does 'return to home'. Betaflight does 'GPS rescue'. See https://github.com/betaflight/betaflight/issues/7277#issuecomment-449667030.

etracer65, 2018-12-24:
> Let's look to another typical scenario. At start of flight the pilot does a punchout to some hundreds of meters for some sightseeing. For all remaining flight max altitude will be recorded and used even with flat battery. Which is not optimal and dangerous.

Again, how does GPS Rescue know that the pilot was sightseeing in an open area instead of flying over some high terrain between it's current position and the recorded home position?

rrk1, 2018-12-24:
No way to know for autopilot, until it records detailed trackback, for sure. But the pilot most likely will know and can choose most appropriate setting for a current terrain :)

INAV for example has many options for RTH altitude. In my humble opinion, also confirmed by a lot of INAV flights in nature, using default (current altitude or as least ~50m) for RTH  is safest.   
 
IMHO, current BF GPS rescue approach (max altitude+X) may cause troubles by blindly forcing the craft to reach altitude it no longer can do.

etracer65, 2018-12-24:
You're still missing the point about what GPS Rescue is. It's not a RTH functionality - never intended to be and never will. It's a "rescue" function with the goal to get the quad back into range so the pilot can regain control. Popping up to the altitude recorded to clear obstacles aids in this as the radio link is easier to establish with a clear line of site. The pilot is not supposed to let the quad "fly home" on it's own. They should retake control at the first opportunity.

And I find this concern about climbing to an altitude risky because the battery might go flat unconvincing. If your battery state is that critical then you shouldn't have been that far from home anyway.

rrk1, 2018-12-24:
Taking into account rather unprecise  GPS altitude which rescue feature uses, i'd vote for (100m or current altitude+50m) default.

rrk1, 2018-12-24:
But in fact whatever it called, rescue is nothing more or less than a bit dumbed down RTH.  Just minus controlled landing.

As for battery. please read my example #2 again. Did some high punchouts/dives, max alt recorded. Now gently cruising at 100m alt close home to kill remaining 20% of charge left. Nothing criminal/irresponsible. But with default algorithm, I can no longer attempt to use rescue (video lost, RX failure, etc) as the craft can not reliably reach 500+m with 20% charge any more.

rrk1, 2018-12-27:
Okay. Got a point, but still can not understand why this rescue feature is mostly treated like "orphan child" here in BF and developers are so reluctant to discuss some glaring flaws of its implementation.

I do not need a "true RTH" most of the time, need just a little insurance for long-range-ish flights to fly back a bit to regain control/video. What exactly rescue initially promised.

BUT! Current implementation is *unusable*, enough tears poured already. We had a frame broken just on the first test and I heard of a quad drowned in a river :)

So, better to close this issue?

ijustwannafly, 2018-12-27:
I have gps rescue configured, enabled and working on two of my quads.

It works by its design within the confines of how it is programmed.

It cannot take into account all scenarios nor will it make assumptions about your flying environment (battery remaining, are you in mountains, etc.).

It is programmed as simply as possible to achieve the most direct result...turn your quad around and bring it closer to you to regain reliable control.

How everything else is setup is on the user.  The wiki states the initial altitude is set to default of 50.  It also states this altitude is relative to the takeoff point OR the highest altitude recorded + 15m whichever is higher.

If you feel the quad is climbing too high on rescue then fly at a lower altitude and set your initial alt value low.  Obviously flying too low is not an option so you have to take into consideration your flight plan.

Betaflight can't do everything and it especially cannot prevent pilot error.

rrk1, 2018-12-27:
Unfortunately, being able to configure and turn on a feature does not always mean it's usefulness in the field :)