betaflight/betaflight #7286: Help from TTTRC

TTTRCF7, 2018-12-25:
I'm CEO of TTTRC: Elephant. My agent told me that the nomination for TTTRCF7 failed. I think we should be able to contact you here. We are a product quality-oriented brand. We have been helping to use BF Flight Control conveniently. At present, our FC has integrated the 3IN seats of four-way LED lights. In the next generation of products, we will integrate Bluetooth in flight control. Make users more convenient. However, the failure of adding the name made it difficult for us to develop the product later. I wonder if BF can help me solve this problem.

mikeller, 2018-12-25:
Hi CEO.

Please see #7283 for an explanation of our policy with regards to adding targets for cloned boards.

Can you provide some more details on how you have been helping 'to use BF Flight Control conveniently'?

> At present, our FC has integrated the 3IN seats of four-way LED lights. In the next generation of products, we will integrate Bluetooth in flight control. Make users more convenient. 

All these things do not require a different target - Betaflight is very flexible, and allows users to install the same target onto boards with LEDs, and even with Bluetooth! Just tell your users to select the EXF722DUAL target and install that, and everything will work fine.

> However, the failure of adding the name made it difficult for us to develop the product later.

Can you explain a bit more how the naming of the target to install is affecting your development efforts? To me, these things seem to be completely unrelated.
The way this works is that you first develop a target that includes **innovative** new technology, and then we create a new target for it.

Also, take note that we are working on restructuring the way that manufacturer specific target configurations are done, and from 4.0 on we will be offering manufacturers a way to publish their own board configurations (with custom naming), based on officially released Betaflight firmware. Expect this to be available from 19Q2.

TTTRCF7, 2018-12-25:
But SPEEDYBEE. that is, runcam authorized me to only TTTRCF7 target firmware, other third-party firmware, will lock.

TTTRCF7, 2018-12-25:
I hope you can use BF more conveniently and have a pleasant flight. I've been working hard for that.

mikeller, 2018-12-25:
That means that the SpeedyBee software is broken, and not Betaflight. Tell them to fix their software.

TTTRCF7, 2018-12-25:
So why do FOXEER and DALRC have the same goals, but they have already nominated?

mikeller, 2018-12-25:
Please read #7283 for an explanation of our policy with regards to clones, and the reasons behind it.

There is nothing more to discuss.

TTTRCF7, 2018-12-25:
My target firmware is to increase the convenience of Bluetooth on the basis of the original, so the target object is different.

TTTRCF7, 2018-12-25:
能否告诉我你的facebook ID，我想和你讨论

TTTRCF7, 2018-12-25:
Can you tell me your Facebook ID? I want to discuss it with you.

mikeller, 2018-12-25:
Betaflight works perfectly well with bluetooth with the EXF722DUAL target. If the SpeedyBee app does not work with this firmware then it is broken and needs to be fixed, and you need to talk to SpeedyBee about this.

TTTRCF7, 2018-12-25:
SpeedyBee's Bluetooth integration scheme is a joint project between me and their engineers. I also want to realize the wireless connection between FC and PC on the computer side of BF without wired connection.

RipperDrone, 2018-12-25:
Let me try to understand: @TTTRCF7 wants to help introducing a flight controller which is cloned from DAL or Foxeer F7 Dual FCs, the difference being that the TTTRCF7 flight controller will have a new feature added which is wireless Bluetooth bridge from FC to PC, over and above FC to Speedybee app which is common for all 3 targets. This new feature has been authorized by Runcam, but only for the new TTTRCF7 target, not for other (e.g. DAL or Foxeer) targets. So using a common target EXF722DUAL for the TTTRCF7 might be a problem regarding the aforementioned Runcam license. What should TTTRCF7 do now to make it work? I agree that many people don't appreciate cloning in general (me either), and BF has to limit the maintenance efforts since people are putting in their private time for the project. Plus BF is not there to advertise / promote commercial projects. On the other hand, if there is a new piece of innovative technology (I love the SpeedyBee approach for my part - live configuration in the field from my mobile phone, without even a cable needed), BF should somehow offer an inroad how to best implement those features. @TTTRCF7 : Maybe Runcam can be convinced to open their license to any F7 FC, thereby broadening the user base... so you can eventually use the common EXF722DUAL firmware?...

TTTRCF7, 2018-12-25:
@RipperDrone YES.Yes. This Bluetooth can even connect directly to the PC. Get rid of the use of the line. This is very important. RUNCAM cannot be changed. So can we change it? Maybe I think so. BF will change faster. In fact, I also implemented it with other Bluetooth modules. However, it is limited to firmware of OMNIBUS. RUNCAM defines the target source

wind0r, 2018-12-25:
All questions got answerd above there is no point in repeating it all the time
What is even worse that we need to talk with people with out any hardware and software  technically knowledge. It is quite hard and stupid for speedybee to detect the fc and if the realy want to do so it is thier option. We will it change anything just to make a third party happy. 

What is even more stupid is that all this lock in doesn't conform with the licence and als will not be able to do with the 4.0 release and generic target

It also would be great if @tttrcf7 would answer the questions @mikeller asked instead of spamming here. There is no point in writing us over different channels.

wind0r, 2018-12-25:
Also since a similar example got added to every other target name discussion is that someone needs to pay for the travis ci premium plan (https://travis-ci.com/plans) around $489 and the same amount for the Jenkins build server. We do not have the resources just to create targets with just a changed name. that shows how much the author knows about the firmware and the development process.

soo if @TTTRCF7 is willing to donate around $1000 per month we could discuss further. every other response just wastes more time and also is quite senseless with the generic targets for the future.

closed 2019-01-31 by stale[bot]