betaflight/betaflight #7287: #1280 Clip gate, spin, disarm, starts spinning uncontrolled when re-arm

oyvinla, 2018-12-25:
**Describe the bug**
When a gate is clipped, and the quad starts to spin, I disarm, but when I arm again, the quad starts spinning and takes off uncontrollably. I can disarm, wait for a few seconds, then all is fine. See video link at the bottom

**To Reproduce**
Clip a gate, disarm, wait untill stand-still, arm imediately.

**Expected behavior**
Normal flight behaviour after re-arm

**Flight controller configuration**
```
# diff

# version
# Betaflight / MATEKF405 (MKF4) 4.0.0 Dec 10 2018 / 14:26:57 (6972f3eb0) MSP API: 1.41

board_name MATEKF405
manufacturer_id 

# name

# resources
resource SERIAL_TX 4 NONE
resource CAMERA_CONTROL 1 A00

# mixer

# servo

# servo mix

# feature
feature -SOFTSERIAL
feature LED_STRIP
feature AIRMODE

# beeper
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING_GPS_FIX
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CAM_CONNECTION_OPEN

# beacon

# map
map TAER1234

# serial
serial 0 128 115200 57600 0 250000
serial 3 16384 115200 57600 0 115200
serial 4 2048 115200 57600 0 115200

# led
led 0 0,0::A:2
led 1 1,0::A:2

# color

# mode_color
mode_color 6 1 2

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 2 1700 2100 0 0
aux 2 13 2 1250 1650 0 0
aux 3 15 4 1300 1700 0 0

# adjrange

# rxrange

# vtx

# rxfail

# master
set min_check = 1020
set max_check = 2000
set rssi_channel = 9
set dshot_idle_value = 700
set motor_pwm_protocol = DSHOT600
set align_board_yaw = 270
set vbat_min_cell_voltage = 31
set vbat_warning_cell_voltage = 33
set ibata_scale = 400
set yaw_motors_reversed = ON
set small_angle = 180
set pid_process_denom = 1
set osd_vbat_pos = 2433
set osd_rssi_pos = 2487
set osd_tim_2_pos = 2455
set osd_current_pos = 2464
set osd_mah_drawn_pos = 2448
set osd_craft_name_pos = 40
set osd_warnings_pos = 329
set osd_avg_cell_voltage_pos = 2440
set osd_stat_tim_1 = ON
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set vtx_band = 5
set vtx_channel = 3
set vtx_freq = 5732
set camera_control_ref_voltage = 325
set camera_control_key_delay = 125

# profile
profile 1

# rateprofile
rateprofile 0

```
**Setup / Versions**
Emax Babyhawk R

**Additional context**
https://www.youtube.com/watch?v=3VNU1yO8GQc&feature=youtu.be

etracer65, 2018-12-25:
You just need to give it a moment to settle and let the attitude estimate reset. After a big high rate spin and crash like that the attitude estimate will be way off (quad may think it's upside down or on it's side for example). When you disarm and the quad is sitting still it will reset after about 0.5 seconds. The problem was than on your first disarm the quad was still moving (rocking back and forth) and then you rearmed. This never let the attitude reset as it looks for no motion so that it can trust the accelerometer. On the second time you let it sit still (longer than needed) and everything was fine.

I don't think there are any problems here.

jflyper, 2018-12-25:
> I don't think there are any problems here.

Although the phenomenon does perfectly align with the specification, I think it _*IS*_ a problem. We need something that ensures attitude estimation is correct and stable before re-arming, as it is not a robust solution to ask for a race pilot to "wait for _a while_" before re-arming. Thoughts?

etracer65, 2018-12-25:
There is already logic in place to accelerate the attitude estimate convergence very quickly on disarm. As mentioned it only needs a fraction of a second - but the quad must be still so we can trust the accelerometer. That was the problem in this case.

The attitude estimate becomes unreliable in instances like this because the gyro rate exceeds 2000dps and/or the acceleration exceeds 16g and the sensor maxes so there's no way for the IMU to keep an accurate estimate.

jflyper, 2018-12-25:
The conclusion is that there is no technical way to prevent the phenomenon?

etracer65, 2018-12-25:
Nope. When the sensors max out there's no way the IMU can maintain an accurate attitude estimate. So we mitigate this by accelerating convergence when disarmed and not moving (when ACC is trustworthy). This takes about 0.5 seconds so it's really not much of a burden.

jflyper, 2018-12-25:
Thanks @etracer65

closed 2019-01-31 by stale[bot]