betaflight/betaflight #7299: VTX frequency and power table migration

jflyper, 2018-12-27:
So, we have two outstanding PRs atm.

1. [VTX] CLI parser for vtxtable #7251 
A large, +3558 bytes PR, that adds `vtxtable` CLI command processing.

2. [VTX] Convert direct reference to string and freq table to vtxCommon services #7285 
A small, +176 bytes PR, that provides table _source_ independence to vtx related codes, and also provides seamless merging of above PR 1.

- Above 1 and 2 can be merged right now (flash space permitting for F3s), without changing vtx CMS and drivers.
- Actual transition still requires

a. `vtxStringXXX` that supplies tables from above PR 1 instead of current `vtx58_XXX` and driver embedded tables. At this point, static tables can be removed.

b. Changes to vtx CMS and drivers. Tramp should be easy, as it is already frequency driven. Same can probably be said for RTC6705 also. SmartAudio needs large piece of rework.

C. A backward compatible `vtxTableXXX` on-the-fly PG modifier if we want to delay the exposure of user configured table mechanism to users. (Rather stupid thought?)

- At some point in the migration, we should recruit testers with different VTXs.
Those who asked for LoRace band or similar are candidates (See #4959 for example).

jflyper, 2018-12-27:
For now, I would like to go ahead with #7251 and #7285 to reserve a space needed.

fujin, 2019-01-10:
hey @jflyper, do you think any of this will impact SA2.1 work (#7336)? we should synchronize to avoid conflicts (specifically part 'b'); or if you had ideas on how you think the SA2.1 should look, happy to take 'em into account as we work through it.

closed 2019-02-16 by stale[bot]