betaflight/betaflight #7301: no PPM input for fpvmodel betaflight f7

fantasycenturay, 2018-12-28:
I'm using a frsky tfr4 receiver with the fpvm f7 fc. With ch3 and ch4 jumped, this receiver outputs ppm from ch1 and rssi from ch2. 
ch1 is connected to rx6 of the fc, ch2 is connected to rssi pad. The receiver is properly powered using the 5v pad and the gnd pad.
In the configurator, ppm rx input was selected. Uart6 serial rx was disabled.
Under the receiver tab, there's no input. Only the first four channels were shown, but not responding to any stick movement.
The receiver works, and I can also read the rssi value. There's simply no ppm input.
I'm running betaflight v3.5.4, also the latest configurator.

jflyper, 2018-12-28:
Let us see your CLI `resource list` output.

mikeller, 2018-12-28:
@fantasycenturay: Please provide the information that you were asked to provide when opening this issue.

fantasycenturay, 2019-01-11:
[image]
@jflyper Here's the resource list. I saw the ppm there. Thanks.

fantasycenturay, 2019-01-11:
[image]
[image]
[image]

jflyper, 2019-01-11:
@fantasycenturay 

Try this
[code]
resource serial_rx 6 none
resource ppm c7
[/code]

fantasycenturay, 2019-01-11:
@jflyper I fixed it by soldering the receiver to the a03 pad instead. Your method is easier actually. Now we have two methods to fixed this problem!
Thanks for getting me to start using the cli function! It makes understanding all the ports easier.

closed 2019-01-11 by fantasycenturay