betaflight/betaflight #7307: analogue rssi via CAMERA_CONTROL/LED_STRIP or any UARTs

simtcrom, 2018-12-29:
I am running Betaflight / OMNIBUSF4FW (OBFW) 3.5.1 on a Omnibus Nano F4 v6.
The FC does not have an analogue RSSI pad. 
FC does have CAMERA_CONTROL, LED_STRIP and free UART3.
Can I resource map any of the above to become an analogue RSSI pad?

resource
```
resource ADC_RSSI 1 A00
resource LED_STRIP 1 B06
resource CAMERA_CONTROL 1 B09
resource SERIAL_TX 3 B10
resource SERIAL_RX 3 B11
```

resource list
```
A00: FREE
B06: LED_STRIP
B09: CAMERA_CONTROL
B10: FREE
B11: FREE
```

simtcrom, 2018-12-29:
> What RX / protocol are you using by the way?

I have UART3 and 6 pads unused .
Also have smart audio, buzzer, sbus, ESC telemetry or RX4
Here is the layout, [https://cdn8.bigcommerce.com/s-xkoep7/product_images/uploaded_images/omnibus-f4-nano-v6-pinout.jpg](url)

I am just using Rx, video in, video out,

I am using [FS-RX2A Pro V1](https://m.banggood.in/FS-RX2A-Pro-V1-2_4G-Compatible-Receiver-for-FS-I6-FS-I6X-FS-I6S-FS-TM8-FS-TM10-FS-I10-Transmitter-p-1257344.html?rmmds=myorder) connected to UART1 using IBUS.

mikeller, 2018-12-29:
Doesn't look good - I don't think any of these pins is ADC capable. 

It's a shame IBus does not support transmission of RSSI as part of the protocol.

simtcrom, 2018-12-29:
> Doesn't look good - I don't think any of these pins is ADC capable.
> 
> It's a shame IBus does not support transmission of RSSI as part of the protocol.

too bad. I was hoping to get RSSI on my micro brushless quad.
I guess I will have to look for another RSSI compatible Rx.

McGiverGim, 2018-12-29:
I think the only solution is to change your receiver by a IRangeX Fli14. This has the rssi signal in one of the aux channels.

McGiverGim, 2018-12-29:
No, sorry. I have not tested any of them.