betaflight/betaflight #7316: VFAS telemetrie not discovered

Hydroculture, 2018-12-30:
**Describe the bug**
VFAS is not displayed in the Taranis QX7s telemetry screen. 
When trying to discover new sensors, 
RSSI, RxBt, A4, Hdg, AccX, AccY, AccZ, Temp1
are correctly discovered, but VFAS only shows "---"

**To Reproduce**

Flash a development build > #1311

**Expected behavior**
show VFAS again

**Flight controller configuration**
```# version
# Betaflight / OMNIBUSF4 (OBF4) 4.0.0 Dec 25 2018 / 03:25:47 (69bce7bff) MSP API: 1.41

board_name OMNIBUSF4
manufacturer_id AIRB
mcu_id 005300333436510937333338
signature 

# reset configuration to default settings
defaults nosave

# name
name ZWEIER

# resources
resource MOTOR 5 NONE
resource MOTOR 6 NONE
resource PPM 1 NONE
resource PWM 1 NONE
resource SONAR_TRIGGER 1 NONE
resource SONAR_ECHO 1 NONE
resource LED_STRIP 1 NONE
resource SERIAL_TX 11 B14
resource SERIAL_TX 12 A01
resource SERIAL_RX 12 A01
resource LED 1 NONE
resource ESCSERIAL 1 NONE
resource ADC_RSSI 1 NONE
resource ADC_CURR 1 NONE

# mixer

# servo

# servo mix

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature SOFTSERIAL
feature TELEMETRY

# beeper
beeper -ON_USB

# beacon
beacon RX_LOST
beacon RX_SET

# map
map EATR1234

# serial
serial 0 64 115200 57600 0 115200
serial 5 16384 115200 57600 0 115200
serial 30 32 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1300 2100 0 0
aux 1 2 0 1300 1700 0 0
aux 2 13 1 1300 1700 0 0
aux 3 26 2 1450 1700 0 0
aux 4 26 2 900 1300 0 0
aux 5 28 0 1300 2100 0 0
aux 6 30 2 1450 2100 0 0
aux 7 33 1 1700 2100 0 0

# adjrange

# rxrange

# vtx

# rxfail
rxfail 4 s 1000
rxfail 5 s 1500
rxfail 6 s 1350
rxfail 7 s 1000

# master
set dyn_lpf_gyro_max_hz = 800
set mag_hardware = NONE
set baro_hardware = NONE
set rssi_channel = 8
set max_aux_channels = 4
set serialrx_provider = SBUS
set serialrx_inverted = ON
set dshot_idle_value = 600
set motor_pwm_protocol = DSHOT1200
set failsafe_delay = 10
set current_meter = NONE
set force_battery_cell_count = 4
set vbat_scale = 112
set beeper_dshot_beacon_tone = 2
set yaw_motors_reversed = ON
set deadband = 10
set yaw_deadband = 10
set frsky_vfas_precision = 1
set osd_warn_core_temp = OFF
set osd_warn_fail_safe = OFF
set osd_warn_launch_control = OFF
set osd_rssi_alarm = 40
set osd_cap_alarm = 850
set osd_tim2 = 2817
set osd_vbat_pos = 2496
set osd_rssi_pos = 2273
set osd_tim_1_pos = 54
set osd_tim_2_pos = 23
set osd_flymode_pos = 333
set osd_throttle_pos = 225
set osd_vtx_channel_pos = 377
set osd_crosshairs_pos = 234
set osd_ah_sbar_pos = 234
set osd_ah_pos = 200
set osd_current_pos = 470
set osd_mah_drawn_pos = 472
set osd_craft_name_pos = 362
set osd_gps_speed_pos = 218
set osd_gps_lon_pos = 82
set osd_gps_lat_pos = 65
set osd_gps_sats_pos = 51
set osd_home_dir_pos = 302
set osd_home_dist_pos = 303
set osd_compass_bar_pos = 266
set osd_altitude_pos = 247
set osd_pid_roll_pos = 423
set osd_pid_pitch_pos = 455
set osd_pid_yaw_pos = 487
set osd_debug_pos = 1
set osd_power_pos = 321
set osd_pidrate_profile_pos = 345
set osd_warnings_pos = 2505
set osd_avg_cell_voltage_pos = 76
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_battery_usage_pos = 392
set osd_disarmed_pos = 2283
set osd_nheading_pos = 311
set osd_nvario_pos = 279
set osd_esc_tmp_pos = 473
set osd_esc_rpm_pos = 440
set osd_stat_rtc_date_time = ON
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set osd_stat_used_mah = OFF
set debug_mode = GYRO_SCALED
set vtx_band = 5
set vtx_channel = 8
set vtx_power = 3
set vtx_freq = 5917

# profile
profile 0

set dyn_lpf_dterm_min_hz = 120
set dyn_lpf_dterm_max_hz = 180
set dterm_lowpass_hz = 120
set i_pitch = 85
set d_pitch = 16
set p_roll = 50
set i_roll = 78
set d_roll = 16
set p_yaw = 10
set i_yaw = 130

# profile
profile 1

# profile
profile 2

# restore original profile selection
profile 0

# rateprofile
rateprofile 0

# rateprofile
rateprofile 1

# rateprofile
rateprofile 2

# rateprofile
rateprofile 3

# rateprofile
rateprofile 4

# rateprofile
rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
```

**Setup / Versions**
 - HGLRC XJBF428 (OBF4,FW's above FW#1311 reproduce the error)
 - FrSky RXSR, sbus, sport telemetry
 - VTX003

**Additional context**
 - Firmware versions above #1311 reproduce the error.

 - Thank you for having a look at my problem
   
best
 Holger

klutvott123, 2018-12-30:
Can you try ``set telemetry_disabled_sensors = 1015808`` and discover new sensors again?

Hydroculture, 2018-12-30:
I will try this right now

Hydroculture, 2018-12-30:
At #1312 i got this:
# get telem
telemetry_disabled_sensors = 491520
Allowed range: -1 - 15

i will flash latest fw and try again

klutvott123, 2018-12-30:
Does it show 0.00V even with the battery plugged in? Strange. 
There was a new sensor added in 1312 which results in some strange behaviour with smartport. 
Read more about it here:  #7273

Hydroculture, 2018-12-30:
yes, it shows 0.00V with the lipo connected :(

Hydroculture, 2018-12-30:
I calculated the telemetry flags to  524286 (only vfas enabled)  = 1111111111111111110

Will give it a try

Hydroculture, 2018-12-30:
As my problem  is described in  #7273 i close this issue. Sorry for the double post

Hydroculture, 2018-12-30:
umpf, only half success. VFAS is found, but shown as 0.00V

klutvott123, 2018-12-30:
Have you tried deleting all sensors in opentx and discover them again?