betaflight/betaflight #7317: RunCam Split v2 doesn't respond to Betaflight's power button mode

andrejpodzimek, 2018-12-31:
On two different builds, `KAKUTEF7` and `ANYFCF7`, RunCam Split v2 doesn't respond to the UART-based power button channel / mode / command.

I've checked that the transmitter channel makes it to Betaflight and is shown in the modes tab correctly. It just doesn't get to the RunCam Split v2 from there. Recording cannot be started or stopped using the transmitter.

Both of my Split v2 instances have the November 2018 firmware, camera control is set to *Flight Control* in RunCam app settings and both record video just fine when their button on the PCB is pressed.

## Related bugs and context

* Unlike #4840 (a problem some users and I have with Split v1), flipping the channel up and down multiple times doesn't help at all with Split v2.
* I haven't had any FC crashes / reboots (unlike Split v1 in #5636) with the Split2; just no response at all. (I should have responded to #5636, but haven't had much hobby time recently; deeply sorry about that.) 
* Another distantly related bug may be #5832. Not sure what the status of that one is.
* #6140 might be related. It worked for some people when they used a different UART while it didn't work at all for others. `KAKUTEF7` is mentioned there. (I tried ~3 different UARTS, to no avail.)

## `KAKUTEF7` diff

#### Config diff (RunCam on UART3)

```shell
# version
# Betaflight / KAKUTEF7 (KTF7) 3.5.3 Nov 14 2018 / 23:49:50 (61d5abf00) MSP API: 1.40

# mixer
mixer HEX6X

# feature
feature GPS
feature TELEMETRY
feature LED_STRIP
feature ESC_SENSOR

# serial
serial 0 2 115200 57600 0 115200
serial 1 64 115200 57600 0 115200
serial 2 16384 115200 57600 0 115200
serial 5 2048 115200 57600 0 115200
serial 6 1024 115200 57600 0 115200

# led
led 0 10,15:ESD:CIW:10
led 1 15,7:ED:CIW:6
led 2 10,0:NED:CIW:4
led 3 5,0:NWD:CIW:4
led 4 0,7:WD:CIW:2
led 5 5,15:SWD:CIW:10

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 1 900 1625 0 0
aux 2 2 1 1625 1875 0 0
aux 3 5 6 1700 2100 0 0
aux 4 3 1 900 1375 0 0
aux 5 11 1 900 1125 0 0
aux 6 13 3 1300 2100 0 0
aux 7 28 4 1700 2100 0 0
aux 8 33 5 1700 2100 0 0

# adjrange
adjrange 0 0 2 900 2100 12 2 0 0

# rxfail
rxfail 5 s 1000
rxfail 6 s 1000
rxfail 7 s 2000
rxfail 9 s 2000

# master
set gyro_sync_denom = 2
set gyro_use_32khz = ON
set acc_trim_pitch = 14
set mag_declination = 2180
set serialrx_provider = FPORT
set serialrx_inverted = ON
set serialrx_halfduplex = ON
set blackbox_p_ratio = 0
set blackbox_device = NONE
set min_throttle = 1070
set dshot_idle_value = 200
set use_unsynced_pwm = OFF
set motor_pwm_protocol = DSHOT1200
set motor_pwm_rate = 480
set battery_meter = ESC
set gps_provider = UBLOX
set gps_sbas_mode = EGNOS
set gps_auto_baud = ON
set pid_process_denom = 1
set frsky_unit = METRIC
set osd_warn_core_temp = OFF
set osd_rssi_alarm = 30
set osd_cap_alarm = 1000
set osd_vbat_pos = 2466
set osd_rssi_pos = 2498
set osd_tim_2_pos = 405
set osd_flymode_pos = 2521
set osd_g_force_pos = 2201
set osd_throttle_pos = 2488
set osd_vtx_channel_pos = 33
set osd_crosshairs_pos = 2253
set osd_ah_sbar_pos = 2254
set osd_ah_pos = 2126
set osd_current_pos = 2474
set osd_gps_speed_pos = 2137
set osd_gps_sats_pos = 2113
set osd_altitude_pos = 2166
set osd_power_pos = 2508
set osd_esc_rpm_pos = 2455
set osd_stat_tim_1 = ON
set osd_stat_max_dist = ON
set osd_stat_max_alt = ON
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set vtx_channel = 5
set vtx_freq = 5820
set vcd_video_system = PAL
```

#### Code diff from 3.5.4

N/A (Just the stock binary in this case.)

## `ANYFCF7` diff

#### Config diff (RunCam on UART8)

```shell
# version
# Betaflight / ANYFCF7 (ANY7) 3.5.4 Dec 31 2018 / 02:48:55 (66d6d80f5) MSP API: 1.40

# resources
resource MOTOR 1 A03
resource MOTOR 2 C09
resource MOTOR 3 B04
resource MOTOR 4 B05
resource MOTOR 5 B08
resource MOTOR 6 A02
resource MOTOR 7 A00
resource MOTOR 8 B03
resource PPM 1 NONE
resource PWM 1 NONE
resource PWM 2 NONE
resource SERIAL_TX 1 NONE
resource SERIAL_RX 1 NONE

# mixer
mixer OCTOFLATX

# feature
feature GPS
feature RANGEFINDER
feature TELEMETRY
feature LED_STRIP
feature ESC_SENSOR

# serial
serial 1 1 115200 57600 0 115200
serial 2 64 115200 57600 0 115200
serial 3 32768 115200 57600 0 115200
serial 4 2048 115200 57600 0 115200
serial 5 1024 115200 57600 0 115200
serial 6 2 115200 57600 0 115200
serial 7 16384 115200 57600 0 115200

# led
led 0 3,14:SD:CIW:10
led 1 4,14:SD:CIW:10
led 2 5,15:SD:CIW:10
led 3 6,15:SD:CIW:10
led 4 1,10:WD:CIW:2
led 5 1,11:WD:CIW:2
led 6 2,12:WD:CIW:2
led 7 2,13:WD:CIW:2
led 8 2,5:WD:CIW:2
led 9 2,6:WD:CIW:2
led 10 1,7:WD:CIW:2
led 11 1,8:WD:CIW:2
led 12 6,3:ND:CIW:4
led 13 5,3:ND:CIW:4
led 14 4,4:ND:CIW:4
led 15 3,4:ND:CIW:4
led 16 11,4:ND:CIW:4
led 17 10,4:ND:CIW:4
led 18 9,3:ND:CIW:4
led 19 8,3:ND:CIW:4
led 20 13,8:ED:CIW:6
led 21 13,7:ED:CIW:6
led 22 12,6:ED:CIW:6
led 23 12,5:ED:CIW:6
led 24 12,13:ED:CIW:6
led 25 12,12:ED:CIW:6
led 26 13,11:ED:CIW:6
led 27 13,10:ED:CIW:6
led 28 8,15:SD:CIW:10
led 29 9,15:SD:CIW:10
led 30 10,14:SD:CIW:10
led 31 11,14:SD:CIW:10

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 1 900 1625 0 0
aux 2 2 1 1625 1875 0 0
aux 3 5 6 1700 2100 0 0
aux 4 3 1 900 1375 0 0
aux 5 11 1 900 1125 0 0
aux 6 13 3 1300 2100 0 0
aux 7 28 4 1700 2100 0 0
aux 8 33 5 1700 2100 0 0

# adjrange
adjrange 0 0 2 900 2100 12 2 0 0

# rxfail
rxfail 5 s 1000
rxfail 6 s 1000
rxfail 7 s 2000
rxfail 9 s 2000

# master
set align_mag = CW180
set mag_declination = 2180
set baro_i2c_device = 4
set max_check = 1950
set serialrx_provider = FPORT
set serialrx_inverted = ON
set serialrx_halfduplex = ON
set adc_device = 0
set blackbox_p_ratio = 0
set blackbox_device = NONE
set min_throttle = 1000
set dshot_idle_value = 400
set use_unsynced_pwm = ON
set motor_pwm_protocol = DSHOT1200
set motor_pwm_rate = 16000
set current_meter = ADC
set battery_meter = ESC
set ibata_scale = 800
set gps_provider = UBLOX
set gps_sbas_mode = EGNOS
set gps_auto_baud = ON
set pid_process_denom = 1
set frsky_unit = METRIC
set vtx_channel = 5
set vtx_freq = 5820
set vcd_video_system = PAL
set rangefinder_hardware = TFMINI
```

#### Code diff from 3.5.4

It does **not** work with stock Betaflight either. I'm mentioning this diff just to explain a few odd settings in the config.

```diff
diff --git a/src/main/target/ANYFCF7/target.c b/src/main/target/ANYFCF7/target.c
index 17dab5159..a0a9722ab 100644
--- a/src/main/target/ANYFCF7/target.c
+++ b/src/main/target/ANYFCF7/target.c
@@ -30,10 +30,8 @@
 const timerHardware_t timerHardware[USABLE_TIMER_CHANNEL_COUNT] = {
     DEF_TIM(TIM12, CH1, PB14, TIM_USE_PWM | TIM_USE_PPM,   0, 0 ), // S1_IN
     DEF_TIM(TIM12, CH2, PB15, TIM_USE_PWM,                 0, 0 ), // S2_IN
-    DEF_TIM(TIM8,  CH1, PC6,  TIM_USE_PWM,                 0, 0 ), // S3_IN DMA2_ST2 DMA2_ST2
-    DEF_TIM(TIM8,  CH2, PC7,  TIM_USE_PWM,                 0, 0 ), // S4_IN DMA2_ST3 DMA2_ST2
-    DEF_TIM(TIM8,  CH3, PC8,  TIM_USE_PWM,                 0, 0 ), // S5_IN DMA2_ST4 DMA2_ST2
-    DEF_TIM(TIM8,  CH4, PC9,  TIM_USE_PWM,                 0, 0 ), // S6_IN DMA2_ST7
+    DEF_TIM(TIM1,  CH3, PA10, TIM_USE_LED,                 0, 0 ),
+    DEF_TIM(TIM8,  CH4, PC9,  TIM_USE_MOTOR,               0, 0 ), // S6_IN DMA2_ST7
 
     DEF_TIM(TIM4,  CH3, PB8,  TIM_USE_MOTOR,               0, 0 ), // S10_OUT 1 DMA1_ST7
     DEF_TIM(TIM5,  CH3, PA2,  TIM_USE_MOTOR,               0, 0 ), // S6_OUT  2 DMA1_ST0
@@ -43,6 +41,6 @@ const timerHardware_t timerHardware[USABLE_TIMER_CHANNEL_COUNT] = {
     DEF_TIM(TIM5,  CH1, PA0,  TIM_USE_MOTOR,               0, 0 ), // S7_OUT DMA1_ST2
     DEF_TIM(TIM4,  CH4, PB9,  TIM_USE_MOTOR,               0, 0 ), // S5_OUT
     DEF_TIM(TIM9,  CH2, PE6,  TIM_USE_MOTOR,               0, 0 ), // S3_OUT
-    DEF_TIM(TIM2,  CH2, PB3,  TIM_USE_MOTOR | TIM_USE_LED, 0, 0 ), // S8_OUT DMA1_ST6
+    DEF_TIM(TIM2,  CH2, PB3,  TIM_USE_MOTOR,               0, 0 ), // S8_OUT DMA1_ST6
     DEF_TIM(TIM3,  CH1, PB4,  TIM_USE_MOTOR,               0, 0 ), // S9_OUT DMA1_ST4
 };
diff --git a/src/main/target/ANYFCF7/target.h b/src/main/target/ANYFCF7/target.h
index edebb21ea..313c174f6 100644
--- a/src/main/target/ANYFCF7/target.h
+++ b/src/main/target/ANYFCF7/target.h
@@ -55,12 +55,15 @@
 //#define MAG_HMC5883_ALIGN CW270_DEG_FLIP
 //#define MAG_HMC5883_ALIGN CW90_DEG
 
+#define USE_RANGEFINDER
+#define USE_RANGEFINDER_TF
+
 #define USE_BARO
 #define USE_BARO_MS5611
 #define USE_BARO_BMP280
 #define BARO_I2C_INSTANCE           (I2CDEV_2)
 
-#define USABLE_TIMER_CHANNEL_COUNT 16
+#define USABLE_TIMER_CHANNEL_COUNT 14
 
 #define USE_VCP
 #define USE_USB_DETECT
@@ -169,4 +172,4 @@
 #define TARGET_IO_PORTD 0xffff
 #define TARGET_IO_PORTE 0xffff
 
-#define USED_TIMERS  ( TIM_N(2) | TIM_N(3) | TIM_N(4) | TIM_N(5) | TIM_N(12) | TIM_N(8) | TIM_N(9) | TIM_N(10) | TIM_N(11))
+#define USED_TIMERS  ( TIM_N(1) | TIM_N(2) | TIM_N(3) | TIM_N(4) | TIM_N(5) | TIM_N(12) | TIM_N(8) | TIM_N(9) | TIM_N(10) | TIM_N(11))
```

jflyper, 2019-01-01:
@andrejpodzimek What is the last known version to work?

andrejpodzimek, 2019-01-02:
My RC Split 2 builds are new (finished this month), so they never had an older version. I'll try to find an older Betaflight where Split 2 works (Split 1 worked nearly OK in ~3.3.x, IIRC) and then bisect. Sadly I can't get back to this before February 11; business trip till then. :disappointed:

bw1129, 2019-01-13:
I am having the same issue with a runcam split 2s (as well as original runcam split). Have not been able to get it to work on anything post BF3.2.5. I'm pretty sure the issue arose with BF3.3.0. I hope someone follows up on this...

bw1129, 2019-01-13:
@jflyper I think the last version that worked with the runcam split was 3.2.5. The issue cam up for me in 3.3.0.

qratz, 2019-01-18:
I am running the Runcam Split Mini v1 on firmware version 2.1. Runcam Device controls do not work at all on a Heli-Nation Talon F7 (CLRACINGF7) with Betaflight 3.5.5

ashefpv, 2019-02-04:
I have a RunCam Split 2S with the latest firmware V3.1.0, and with the Bardwell FC and BetaFlight 3.5.4 and 3.5.5 I had no issue with the switches on my Taranis X9D Plus controlling recording, camera menu, or WiFi on/off. However, with the DALRC F722 and Betaflight 3.5.4, 3.5.5 (currently running 3.5.5 RDQ Patch # 8 (the OSD does not come up at all without this version)), but none of them work, even after 30 tries of any of the switches on the Taranis.

I also tried powering up the quad, selecting from the OSD menu FC "Save and Reboot" but the camera control switches still do not work.

I tried these settings on the FC to give the FC a greater number of chances to wake up the camera, but to no avail:

rcdevice_init_dev_attempts = 10
rcdevice_init_dev_attempt_interval = 5000

If there are any suggestions, or if there are patches/beta versions of BetaFlight to try I'd be glad to test them out.

rolial, 2019-02-19:
I have a RunCam Split 2S with the latest firmware V3.1.0, and with kakute F7 and BetaFlight 3.5.5 and it is not working either (Power / Shutter Button and WiFi / Mode Switch Button)
Is there any solution to the problem?

mikeller, 2019-02-20:
@Schrolli91: Please refrain from cross posting your issue across multiple issues. Look at the existing issues to see if there is one or more that matches your problem, if so, select the best matching one, and post there. If none matches open a new issue, supplying all of the information you are asked to provide.

Duplicate post removed from #7318.

rolial, 2019-02-20:
@mikeller ok, I got it!

closed 2019-03-29 by stale[bot]

andrejpodzimek, 2019-05-02:
Just FTR, with Betaflight 4.0.0 I've had the following RunCam Splits working:
* v1
* v2 (old, not the "S" one)
* mini v1

The behavior is still slightly odd and multiple channel changes might be needed to start / stop the recording, but one can at least guess this based on the camera beeps.

andrejpodzimek, 2019-09-16:
I've just retried most of my builds (Split 1, Split 2, Split 2S, Split Mini 1) and camera controls just work on certain builds and don't work on others (4.0.4, 4.0.6). All of my cameras are on F7s and it looks like `ANYFCF7` works (both 30.5&times;30.5 and 20&times;20) while `EXF722DUAL`, `OMNIBUSF7` and `KAKUTEF7` don't work. @azolyoung, any idea what could be wrong? Are there any interesting inversion settings on the F7 or the like?

andrejpodzimek, 2019-09-16:
To narrow down the space of possible issues, here's what I've just tried on `EXF722DUAL`:

```
set rcdevice_protocol_version=1
set rcdevice_feature=7
set rcdevice_init_dev_attempt_interval=5000
set rcdevice_init_dev_attempts=10
```

Sadly, none of that helped. :sob: (Values are stored normally though, so there's no #7318 any more.)

andrejpodzimek, 2024-03-30:
This is a voice from the past.[^0] I’ve discovered something important today: **It was the baud rate!** [This ancient bug](https://github.com/martinbudden/betaflight/issues/3) in a @martinbudden’s Betaflight clone provided a key hint: **`115200`**

[^0]: I have just disappeared for almost 5 years, for hard-to-explain reasons. This comment should be considered a time capsule from 2019, because my fleet is still stuck at Betaflight 4.0.x and it will take time for me to recover, get up to speed and back on track.

The latest RunCam Split 2 [firmware](https://www.runcam.com/download/runcamsplit2) 3.1.0[^1] (which is _still_ the latest as of **now**) and the latest Betaflight 4.0.x (i.e. Betaflight as of mid-2019) fixed the problem for `ANYFCF7`, but **not** for `KAKUTEF7`. Now that I got back to the issue, I tried the following:

* Resolder the camera from `RX3` / `TX3` to `RX4` / `TX4`, because STM32F7’s documentation appears to suggest there is a difference in these two ports’ hardware. That did **not** help. The UART3 → UART4 migration didn’t work.
  ```diff
  < serial 2 16384 115200 57600 0 115200
  > serial 3 16384 115200 57600 0 115200
  ```

* Try the wisdom from [the ancient bug](https://github.com/martinbudden/betaflight/issues/3), in a somewhat random manner:
  ```bash
  serial 3 16384 115200 115200 115200 115200
  save
  ```
  * Then disconnect or close `betaflight-configurator` **immediately**, before it reconnects!
  
  Notice the **`115200`** everywhere.

After that↑, rebooting the flight controller using just a battery works fine. The camera power button **works**.

Once you reconnect the FC to the confgurator, it stops working. Sometimes you see one of the numbers reverted to `57600` in `serial`. Sometimes this does **not** happen and the line stays the same, but RunCam no longer works.

The solution is to disable the camera, then `save`, then re-enable it using the magic line with `115200` everywhere, `save` again — and quickly disconnect or close `betaflight-configurator`, so it can’t ruin this again.

This concludes the `KAKUTEF7` story. Most of this hardware is now gone and obsolete. I have yet to try and reproduce the issue with _current_ Betaflight.

[^1]: 3.2.0 was for 2**S** only.