betaflight/betaflight #7318: Setting rcdevice_init_dev_attempt_interval appears to have no effect in Betaflight 3.5.4

andrejpodzimek, 2018-12-31:
I spotted this while poking into a RunCam Split v2 problem (#7317). No matter what value I set, it always flips back to 1000:

```
# get rcdevice_init_dev_attempt_interval
rcdevice_init_dev_attempt_interval = 1000
Allowed range: 500 - 5000

# set rcdevice_init_dev_attempt_interval=2000
rcdevice_init_dev_attempt_interval set to 1000

# get rcdevice_init_dev_attempt_interval
rcdevice_init_dev_attempt_interval = 1000
Allowed range: 500 - 5000

# set rcdevice_init_dev_attempt_interval=500
rcdevice_init_dev_attempt_interval set to 1000

# get rcdevice_init_dev_attempt_interval
rcdevice_init_dev_attempt_interval = 1000
Allowed range: 500 - 5000
```

azolyoung, 2019-01-03:
Hi andrejpodzimek, 
We have already purchased a piece of KAKUTE F7, it's on the way. After receiving it, we think the question will be solved quickly.Based on feedback that has been received but not resolved， basically there is a problem with the control on this flight controller.

qratz, 2019-01-18:
Same for me here: can't set rcdevice_init_dev_attempt_interval to anything except 1000. I am running the Runcam Split Mini v1 on firmware version 2.1 on a Heli-Nation Talon F7 (CLRACINGF7) with Betaflight 3.5.5.

azolyoung, 2019-01-21:
@qratz , here is a beta version firmware . This firmware has been tested by several friends in iNav and BetaFlight. It can make the Telemetry control function of RunCam Split work normally. After flashing the following firmware, please go to the CLI interface, enter the following commands:
set rcdevice_protocol_version = 1
set rcdevice_feature = 7
save

https://s3.amazonaws.com/rctmpfiles/rcsplit_remote_control_fws/betaflight_4.0.0_CLRACINGF7.hex

rolial, 2019-02-19:
I have a RunCam Split 2S with the latest firmware V3.1.0, and with kakute F7 and BetaFlight 3.5.5 and it is not working either (Power / Shutter Button and WiFi / Mode Switch Button)
Is there any solution to the problem?

rolial, 2019-02-20:
@azolyoung  Can you send a version to FC Kakute F7 AIO?
How should I proceed?

azolyoung, 2019-02-21:
Yeah, this is the link of the test firmware:
https://s3.amazonaws.com/rctmpfiles/rcsplit_remote_control_fws/betaflight_4.0.0_KAKUTEF7.hex

after flash, please type the following cli commands, and try again.
set rcdevice_protocol_version = 1
set rcdevice_feature = 7
save

Thanks for your test!

rolial, 2019-02-24:
It worked! Congratulations!!
Do you think it will be available in official versions?

I tested these two versions and both worked:

https://s3.amazonaws.com/rctmpfiles/rcsplit_remote_control_fws/betaflight_3.5.0_KAKUTEF7_test3.hex <https://s3.amazonaws.com/rctmpfiles/rcsplit_remote_control_fws/betaflight_3.5.0_KAKUTEF7_test3.hex>

https://s3.amazonaws.com/rctmpfiles/rcsplit_remote_control_fws/betaflight_4.0.0_KAKUTEF7.hex <https://s3.amazonaws.com/rctmpfiles/rcsplit_remote_control_fws/betaflight_4.0.0_KAKUTEF7.hex>
after flash, please type the following cli commands, and try again.
set rcdevice_protocol_version = 1
set rcdevice_feature = 7
save

Rodrigo Lima de Almeida
Tel: (55-16) 3911-7898 / (55-16) 98112-0203
E-mail: rodrigo@chopptime.com.br
Skype: rodrigolalmeida
www.chopptime.com.br
https://www.facebook.com/ChoppTimeRibeirao

BAIXE NOSSO APP:
Iphone: http://apple.co/2aP8p9c
Android: http://bit.ly/2aWymBd

qratz, 2019-04-02:
has this been merged into the 4.0 Release Candidates?

closed 2019-05-09 by stale[bot]