betaflight/betaflight #7321: Configurable GPS init parameters

haplm, 2018-12-31:
Current GPS initialisation defaults in [https://github.com/betaflight/betaflight/blob/master/src/main/io/gps.c](url) might be a bit too conservative. Using more channels, fix can be obtained faster. This was tested by someone I talked to who altered the defaults (increased both min and max) and reported that 3D fix was obtained considerably faster, with more satellites available during the flight.

Do you think that it would be possible to externalise the constants in ubloxGalileoInit into CLI parameters? Or perhaps detect those few widely used units like BN-180, BN-220, Matek, etc. and increase these constants if you are sure that FC is dealing with a reasonably modern GPS unit?

McGiverGim, 2019-01-02:
Maybe the person that made the change can make a pull request with the code to the repository to start the discussion about this. The GPS initialization code is totally hardcoded, so if it can be made in a configurable manner, why not?

McGiverGim, 2019-01-02:
I don't have the knowledge, but I never suspected there was a problem with the actual values. I have seen until 22 satellites fixed. Let's see if someone with more knowledge of this can give us some hints...

closed 2019-02-11 by stale[bot]