betaflight/betaflight #7324: ibus telemetry on software serial only showing one sensor

float23, 2019-01-01:
**Describe the bug**
A clear and concise description of what the bug is.
when i have ibus telemetry set on software serial only the firs sensor shows up on the remote when its on hardware serial all enabled sensors show up on the remote

**To Reproduce**
Steps to reproduce the behavior:
enable ibus telemetry on software serial

**Expected behavior**
to see all enabled sensors

**Flight controller configuration**
```# diff

# version
# Betaflight / OMNIBUSF4SD (OBSD) 3.5.4 Dec 17 2018 / 12:17:14 (66b8ea143) MSP API: 1.40

board_name OMNIBUSF4SD
manufacturer_id 

# name

# resources
resource MOTOR 5 NONE
resource MOTOR 6 NONE
resource SERVO 1 A08
resource SERIAL_TX 3 NONE
resource SERIAL_TX 11 A08
resource SERIAL_TX 12 B06
resource SERIAL_RX 3 NONE
resource SERIAL_RX 11 A01
resource I2C_SCL 2 B10
resource I2C_SDA 2 B11

# mixer

# servo
servo 0 1000 2100 1500 -45 7

# servo mix

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature SERVO_TILT
feature SOFTSERIAL
feature GPS
feature TELEMETRY
feature 3D
feature ESC_SENSOR

# beeper

# beacon

# map

# serial
serial 0 2 115200 57600 0 115200
serial 5 64 115200 57600 0 115200
serial 30 1024 115200 57600 0 115200
serial 31 4096 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 1 1300 2100 0 0
aux 1 1 0 900 1300 0 0
aux 2 2 0 1300 1700 0 0
aux 3 8 2 1700 2100 0 0
aux 4 13 5 1300 1700 0 0
aux 5 19 5 1700 2100 0 0
aux 6 26 1 900 2100 0 0

# adjrange

# rxrange

# vtx

# rxfail

# master
set acc_calibration = 133,121,231
set serialrx_provider = IBUS
set blackbox_p_ratio = 16
set dshot_idle_value = 0
set motor_pwm_protocol = DSHOT150
set ibatv_scale = 10
set ibatv_offset = 500
set 3d_deadband_low = 1455
set 3d_deadband_high = 1454
set 3d_neutral = 1500
set gps_provider = UBLOX
set pid_process_denom = 16
set ibus_sensor = 3,1,0,0,0,0,0,0,0,0,0,0,0,0,0
set osd_units = IMPERIAL
set osd_warn_core_temp = OFF
set osd_vbat_pos = 2056
set osd_tim_1_pos = 2080
set osd_tim_2_pos = 2048
set osd_flymode_pos = 2112
set osd_g_force_pos = 2368
set osd_throttle_pos = 2432
set osd_crosshairs_pos = 2253
set osd_ah_sbar_pos = 2254
set osd_ah_pos = 2126
set osd_gps_lon_pos = 2094
set osd_gps_lat_pos = 2126
set osd_gps_sats_pos = 2144
set osd_home_dir_pos = 2176
set osd_home_dist_pos = 2208
set osd_compass_bar_pos = 2282
set osd_altitude_pos = 2400
set osd_warnings_pos = 2411
set osd_avg_cell_voltage_pos = 2088
set osd_battery_usage_pos = 2062
set osd_disarmed_pos = 2282
set osd_esc_tmp_pos = 2304
set osd_esc_rpm_pos = 2443
set osd_core_temp_pos = 2336
set osd_stat_tim_1 = ON
set osd_stat_max_dist = ON
set osd_stat_min_rssi = OFF
set osd_stat_used_mah = OFF
set osd_stat_max_alt = ON
set vcd_video_system = NTSC

# profile
profile 0

# rateprofile
rateprofile 0
```

**Setup / Versions**
 - Flight controller omnibus f4 v3 
 - Betaflight / OMNIBUSF4SD (OBSD) 3.5.4
 - Other components flysky fs-i6s transmitter fs-ia6b receiver

McGiverGim, 2019-01-02:
The number of sensors is defined by:
```
set ibus_sensor = 3,1,0,0,0,0,0,0,0,0,0,0,0,0,0
```

By default, only:
```
IBUS_SENSOR_TYPE_TEMPERATURE (0x01),
IBUS_SENSOR_TYPE_RPM_FLYSKY (0x02),
IBUS_SENSOR_TYPE_EXTERNAL_VOLTAGE (0x03)
```
If I'm not wrong, you are seeing the temperature and external voltage. Right?

Clawzy, 2019-01-02:
Keen to follow as I'm having a related problem. I just moved my softserial from UART1 to UART3 and lost sensors. I am running OMNIBUSF4SD (OBSD) 3.3.3.

closed 2019-02-11 by stale[bot]