betaflight/betaflight #7327: Determine optimal gyro/pid rates

joelucid, 2019-01-02:
For discussion:

We currently prefer run the pid/gyro at 32/32k or 8/8k. As we introduce cpu intensive features a trade-off will need to be made between activation of features and loop times. We therefore need to determine the importance of running high loop rates to various goals.

The high level goals I can think of are (a) effective filtering, (b) low latency setpoint response and
(c) low latency response to external disturbances and

(a) effective filtering

It's worth noting that we already have a 256 hz digital lpf on the MPU 6000. We don't know the precise nature (e.g. order) of the low pass.

Both our dynamic and rpm notches don't seem to cover frequencies beyond 1khz, so a filter sampling rate of 2khz would be sufficient in terms of nyquist. We might have to run a sw lpf at a higher sampling frequency than 2khz to eliminate higher frequencies left over from the hw lpf. This should be tested.

So from a filtering perspective running one lpf at 4000hz and the other filters at 2000hz is very likely plenty. Possibly all filters can be run at 2khz. 

Obviously the filter loop frequency should not be lower than the pid loop frequency to present fresh samples to the pid loop and not add any delay.

(b) low latency setpoint response

Important in any trade-off is to compare the loop time latency to other latencies in the system:

The TX to FC latency of a Taranis using 8 channels is somewhere around 25ms. Motor rpm effectively follows motor signal like a pt1 filter with a cutoff of around 10hz which gives it a time constant of around 15ms. Now the pids overdrive the motors so we do get quicker response. But it still adds about 7ms latency to the sticks for a well tuned quad. RC filtering adds a bit more, call it 35ms total end-to-end latency.

Going from 32khz to 2khz loop times thus increases the end-to-end latency from sticks to quad by 1.3%. Clearly this is relatively a very small improvement.

(c) low latency external disturbance response

Here the relevant latencies are the motor response latency and the filter latency. Given that the long RC latency is not part of this picture higher loop rates have a larger impact. Still taking normal D delay of maybe 5ms and motor time constant of 15ms into account it's clear that any algorithms that lower filter delay or increase motor response have a much higher potential than ultra high loop rates, which likely bring less than 3% improvement.

For example the rpm filter reduces gyro latency by about 1.5ms and D latency by maybe 3ms. This shows how much more important our algorithms are to flight performance than loop time.

---

Given the above I'd tend to support 8k/8k, 4k/4k, 4k/2k and 2k/2k as preferred cycle times. 4k/2k is only needed if a sw lpf above 1khz is necessary. The higher filter rates at 4k and 8k are only there to avoid latency from up sampling at higher pid loop rates.

It will be essential to ensure that BF is written to allow lower loop times. The dynamic filter for example currently has issues at those due to increased latency since cycle time isn't taken into account in the state machine. At lower rates several filter stages can and should be run in one cycle.

I'd expect 2k/2k to become a sweet spot as our algorithms increase in complexity. That should give the F4 a long expected life time - even the lower level ones like the stm32f411.

fujin, 2019-01-02:
Fully Agree. I would like to see 32kHz gyro sampling and associated code dropped.

Increase in loop budgets will allow for simpler contributions especially of those new to the project.

All other points mentioned are accurate.

spatzengr, 2019-01-03:
One consideration is the DLPF moving up to ~4,000hz (from ~250hz) in 32k mode, which drops about 0.88ms of phase delay on the gyro signal.  You typically have to make this up with additional software filtering, but results vary based on your base vibration level in the quad.  However, you can do the same by going to "Experimental" mode for IMC gyro chips (and MPU6000 chips if we re-enable that feature) in 8k mode.

I typically will tell folks to tune their filters in 8k mode.  Then switch to 32k mode.  If they have to add any additional software filtering - which will typically induce more than 0.88ms of phase delay unless it is a Notch type filter - it is not worth it and to switch back.

It will surely cause a rumble to drop 32k mode support.  How much ROM space can be gained?

joelucid, 2019-01-04:
@spatzengr, in my attempts to use experimental mode with 8k I just got complete nonsense out of the chip. There was a lot of low frequency noise that could not be filtered out well at all. It seems like the chip does some processing that isn't really traditional filtering which removes these artifacts and that processing gets disabled in experimental mode. I believe @ctzsnooze had the same results.

Good point about using 32k to eliminate the 256 hz filter though. I'd argue that in that case you'd either need a software lpf to replace it which would need to run at 32k to allow high frequency filtering. But everything else - like dynamic notches, rpm filter etc could still be run at 2k, 4k or 8k.

Combine with https://github.com/betaflight/betaflight/issues/7326 and we could still use up to 500us for advanced filtering and pids without drawbacks. And that's the issue. I don't care about flash here. It's the super tight limit on cpu cycles for pids/filtering that results from 32k currently that will be an issue, soon.

fujin, 2019-01-04:
Yes, the ROM is not the concern: It's more that we spend a good amount of time aggressively optimizing changes in the context of 32kHz gyro sampling mode timing constraints, which could be energy better spent elsewhere instead of the current break 32khz -> fix 32khz -> break 32khz -> fix .. loop we're in.

joelucid, 2019-01-05:
@spatzengr, what you've left out in the graph is TX to FC latency (about 25ms for FrSky X). That's the biggest bang for the buck. It's partially due to the frame rate but there are others sources of delay there, too. Also RC Smoothing with the filter approach costs less than 9ms. I used to not run any RC smoothing but interestingly that is a problem with the rpm-filter since it creates noise side-bands that can't be filtered out based on rpm.

pulquero, 2019-01-05:
You could keep 32khz but just dont put any effort into optimizing it anymore - effectively depricate it (say remove it from configurator but leave the cli), see if hardware catches up. Arguably we've never had the hardware to comfortable run it, only recently with the advent of dual processor boards.

ItayBenNer, 2019-01-05:
Wat I can contribute is my expireance in Raceflight for the past 1.5 years , it was real responsive but it came with a big thumb rule 
Your quad must be a very very clean setup , so you have a really great expireance until...

The 32hz is really sensitive to any noise 
The benefits vs the real life senerios where not all your quads is going to be the cleanest build 

The need for 2k or 4K/4K as I use today and give me the best results , are the one of the reasons why I left the RF / FO frimwere (some of my quads using the revolt FC running BF and it is aowsom )

rav-rav, 2019-01-07:
One thing that should be evaluated is how well the dynamic filters work on lower sampling rates.
We are changing _constants_ here and I remember the filters needing a few samples to completely adjust to the new settings before being fully effective in the new cutoff.
I'd expect the effectiveness go way down if the update rate of the cutoff is close to the sampling rate of the data. A gyro rate of 4k should probably be just fine though.

spatzengr, 2019-01-07:
Here is what I want to vet in the next few months (winter, cut me a break); and someone is more than welcome to beat me to it.  Use the dual FC quad to do a flight with to FCs.  One at 8k mode and one at 32k mode.  Record raw and filtered vibs and see if we can see the difference in Phase Delay.

Potential Phase Delay reduction on 32k is compelling (see below).  Trouble is you can ONLY see this by comparing fully filtered gyro signal data on a dual FC in the stack rig, because the 0.88ms reduction going from 8k to 32k is in the raw noise, not in comparing raw vs. filtered.

Also Brian White developed a tool with MathLab to compare the PID Error of two logs.  Can switch between 8k and 32k on the same rig and run a bunch of prop wash inducing flights to compare logs and then average out.  Statistically, you need 30 samples, but .... we will have to deal with less if I'm doing it.  Maybe 5 to 10.

Also, @rav-rav : in the current code, the Dyn Notch disables for anything below 2k sampling and requires 3 samples regardless.  So at 2k sampling, you get a lower FFT sampling.  We have a test HEX out that goes a different route and need to compare that with 8k gyro rate tracking to see if the FFT tracks about the same or not.  Same method: 2 FCs on the same rig.  Same (or really damn close) flight data in the logs.

[image]

[image]

[image]

vkolotov, 2019-01-08:
Just so you know, I've recently tested latency between Taranis QX7 and R-XSR, the results are here: 

https://docs.google.com/spreadsheets/d/1cFSzIh9LWjcdg4LolqD-gMdTAGd3qDqY6YDe6nbOjGY

I'd say OpenTX latency is more around 30-35ms, e.g. not 25. 

ersky9x latency is a bit better - around 20ms.

Also OpenTX guys are working on a fix at the moment, it is not clear when it is going to be rolled out: https://github.com/opentx/opentx/issues/4640#issuecomment-451371167

Just thought this might help you to decide on optimal rates.

mujahid-abdulrahim, 2019-01-10:
I ran a series of tests this week to assess how flight controller loop rates affect the stability and response of an FPV quad.  I was using a QAV-X Charpu RTF running a LUXF4OSD board.  It was running a modified version of BetaFlight 3.4.0 which included provisions for injecting a frequency sweep disturbance into roll/pitch/yaw loops within pid.c.

This is broken-loop frequency response of the roll rate loop measured at loop rates ranging from 280Hz to 8kHz.  It is a measure of the performance and disturbance rejection capability of the control loop:
[image]
At low frequencies, the magnitude is above 0dB (1) because a small disturbance will result in a large control effort, meaning the controller will respond to and reject the effect of the disturbance.  As the frequency increases, the controller response will naturally decrease until it is only able to match the disturbance.  This is the frequency at which the magnitude plot crosses the 0dB line.  Above this frequency, the disturbance overwhelms the controller.

The difference between the responses is evident when zooming in on the region between the 0dB crossing and the -180deg crossing:
[image]
Magnitude changes a bit, but phase seems to change much more.  The frequency at which the phases crosses -180 deg more than doubles as the controller rate increases.

Most importantly, here is how the stability margins and crossover frequencies change with loop rate:
[image]

The shapes of the magnitude and phase responses determine the stability margins of the controller.  The gain margin measures how much you can increase the gain before the loop is unstable.  6dB means that the loop will be unstable if the gain is doubled.  The phase margin measures how much time delay can be incurred before instability.

Notice how both the gain and phase margin increase with loop rate.  All the margins and frequencies improve dramatically until 1kHz, then more slowly until 4kHz.  For this test, ,there is no improvement beyond 4kHz, and unfortunately, I wasn't able to test 16 or 32khz.

However...
(A) These tests were run at hover, and the results may be different for more dynamic flight conditions.
(B) A colleague pointed out that the differences in phase response *might* be related to changing the loop rate with a constant gain.  It may be possible to recover some performance at the lower loop rates, but I haven't tried.

joelucid, 2019-01-10:
@mujahid-abdulrahim, sounds and looks interesting! For the yet uninitiated do you have a pointer to or can you explain how your method works and exactly how the frequency and phase response was measured? Do you inject a disturbance into gyro before the filtering and then measure how the gyro reacts?

joelucid, 2019-01-10:
@spatzengr, I know you want to blame winter 😃. But couldn't you just have a 8k and 32k fc on your bench and use a loud speaker to sweep once over the relevant frequency range? That should give us the frequency and phase response of both the 8k and 32k filters which would be a really valuable piece of information!

joelucid, 2019-01-10:
@rav-rav 
> One thing that should be evaluated is how well the dynamic filters work on lower sampling rates.
We are changing constants here and I remember the filters needing a few samples to completely adjust to the new settings before being fully effective in the new cutoff.

Interestingly I update the rpm-filter at 1k and it works very well with a 4k loop rate. The rpm data is very noisy and I don't filter it that much in the hope that the adjustment delay you mention provides some averaging of its own. It seems to work or at least I haven't observed any side effects yet.

joelucid, 2019-01-10:
@vkolotov 
> I'd say OpenTX latency is more around 30-35ms, e.g. not 25.

I ran @bchwabe's fix on my Taranis so I'm probably in a somewhat improved situation. Looking forward to your tests with that fix.

closed 2019-02-16 by stale[bot]

andyp1per, 2019-06-08:
FWIW I have run some experiments on ArduPilot using DLPF option 7 on the ICM 20608 (your experimental option for 8k sampling). For me it gives pretty good results. Yes, there is more noise up around 500Hz or so (this option has a 3.6k DLPF), but it looks like you get better sensitivity lower down, you shave 1ms off the latency and software LPFs can remove the residual noise easily.

yoodog, 2019-09-02:
> FWIW I have run some experiments on ArduPilot using DLPF option 7 on the ICM 20608 (your experimental option for 8k sampling). For me it gives pretty good results. Yes, there is more noise up around [...]

So if you translate it to human language, 8 gyro loop brings noise at 500Hz but saves 1ms. Thus if LP filters can remove noise in less than 1ms than we do win a little with 8k vs 4k. Right?

it is not what @spatzengr graph showed - 8k vs 4k only less than 0.1ms difference. One of you got 10x error?

andyp1per, 2019-09-02:
@yoodog I'm only going off the spec'd characteristics for the filter. Yes you can easily do a software filter in 1ms - our filters typically add 2us per filter.

yoodog, 2019-09-02:
Am I wrong thinking that going from 4 (250us) to 8k (125us) saves 125 micro second = 0.125 ms ?

If we count all default BF filtering we get About 5ms delay. Thus going from 4 to 8 seems like a dumb idea, why do we even want 8k if it caused higher delay after all and added complexity?

joelucid, 2019-09-03:
> If we count all default BF filtering we get About 5ms delay. Thus going from 4 to 8 seems like a dumb idea, why do we even want 8k if it caused higher delay after all and added complexity?

Some quads have high frequency noise > 2khz. In those it makes sense to run the gyro filters at 8k. 8k4k is a good choice for this setup and still allows rpm filtering on f405. In the future we want to read the gyro at 8k and run only the gyro lpf at that frequency. Everything else at pid loop speed. Then even on F411 8k4k might be attainable.

With the rpm filter you can easily cut filter latency in half or better. So then at least running 2k vs 4k starts to make a bit of a difference. I think though that once we introduce more processing heavy functionality running 8k2k will still be very good. It should give the F411 a long life.