betaflight/betaflight #7330: No battery voltage in OSD, buzzer not working

gvhlaw, 2019-01-03:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:

**Expected behavior**
A clear and concise description of what you expected to happen.

**Flight controller configuration**
Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/)

**Setup / Versions**
 - Flight controller [what type is it, where was it bought from];
 - Other components [RX, VTX, brand / model for all of them, firmware version where applicable];
 - how are the different components wired up.

**Additional context**
Add any other context about the problem here.

gvhlaw, 2019-01-03:
I flashed a current build of 4.0. - and saw flashing 0.00 voltage in OSD.  Also, beeper did not work correctly.  Could be the result of recent DMA changes.  Hex is OminbusF4.

etracer65, 2019-01-03:
The instructions above are there for a reason. It would help if you followed them. Try resetting your settings with `defaults` and reconfigure (or re-flash with a full chip erase). There were some recent changes to the VBAT precision that may have a problem with old settings. If that doesn't solve it it would be helpful for you to try some previous builds (available in the configurator) to try and narrow down when the problem started for you.

gvhlaw, 2019-01-03:
> The instructions above are there for a reason. It would help if you followed them. Try resetting your settings with `defaults` and reconfigure (or re-flash with a full chip erase). There were some [...]

Thanks etracer65 - I followed the instructions to the extent applicable.  I reverted to an earlier build and the issues were not present.

fgiudice98, 2019-01-03:
@gvhlaw I think the problem with the new build is the voltage configuration: in your diff I see a voltage in 0.1V steps, from the #1321 it changed to 0.01V steps
What build is the earlier one without the problem?

etracer65, 2019-01-03:
@gvhlaw Did you try resetting to `defaults` like I suggested for the non-working build? Also, please provide the working build number(s) you used.

gvhlaw, 2019-01-03:
> @gvhlaw Did you try resetting to `defaults` like I suggested for the non-working build? Also, please provide the working build number(s) you used.

@etracer65  - I have not had a chance to do that yet - as I just posted, I am testing custom hex builds from Ken Mitchell  - but, when it warms-up, I will.  To answer your question, whenever I flash a new hex, I always do a full erase and, after flashing is completed, I click the "Reset All Settings" (twice) before pasting a diff.

etracer65, 2019-01-03:
I just tested with current master (build 1327) and I don't see any problems. The battery voltage shows up as expected in both the Configurator and in the OSD (tested on OMNIBUSF4SD). I suspect you just had a corrupted config from not doing a full chip erase when using the newer build.

Also, if you do a full chip erase there's no need to click the "Reset All Settings" button even once (same if you use the `defaults` CLI command).

Asizon, 2019-01-03:
This issue is foxed on PR #7315,try with the last 4.0 build and check it again please @gvhlaw

gvhlaw, 2019-01-03:
> This issue is fixed on PR #7315,try with the last 4.0 build and check it again please @gvhlaw

Happy to do so, and thanks!

gvhlaw, 2019-01-03:
> I just tested with current master (build 1327) and I don't see any problems. The battery voltage shows up as expected in both the Configurator and in the OSD (tested on OMNIBUSF4SD). I suspect you [...]

I always do a Full Chip Erase - Always.  I'll test again based on the comments from @Asizon

closed 2019-02-11 by stale[bot]