betaflight/betaflight #7334: Logging Custom / Debug Values with Blackbox

mujahid-abdulrahim, 2019-01-04:
I'm working on a research project in which I will need to log specific values from within pid.c.  Is there an easy way to include these values (two to five floats / fixed-point numbers) into the Blackbox log?

For the PX4 flight stack, I used a specific debug uORB topic.  Is there something similar for Betaflight's Blackbox?  

I will only need these values for specific flight tests, so it is absolutely acceptable to temporarily stuff them into fields of existing messages that I will not need.

The research focuses on handling qualities specifications for FPV quads, and one of the early questions that I want to help answer is the one brought up in Issue #7327.  I ran a previous experiment looking at loop and actuator rates for fixed-wing UAVs operating in turbulence, and I'd like to do the same here.

etracer65, 2019-01-04:
Betaflight has 4 available debug logging fields. However they are int16, not floats.

You could choose to reuse an existing debug mode (and comment out it's existing calls), or add a new one.

To add a new one you would add an identifier to src/main/build/debug.h in the `debugType_e` enumeration immediately before the final `DEBUG_COUNT`. Lets assume you want to call it `DEBUG_RESEARCH`. So it would look like this:
```
    DEBUG_RC_SMOOTHING_RATE,
    DEBUG_ANTI_GRAVITY,
    DEBUG_DYN_LPF,
    DEBUG_RX_SPEKTRUM_SPI,
    DEBUG_RESEARCH,
    DEBUG_COUNT
} debugType_e;
```
This will then be the identifier you use for logging in the code.

Next you need to name of the debug mode to the `debugModeNames` array in src/main/build/debug.c.  This will be what's used to select the `debug_mode` in the CLI.  So something like this:
```
    "RC_SMOOTHING_RATE",
    "ANTI_GRAVITY",
    "DYN_LPF",
    "RX_SPEKTRUM_SPI",
    "RESEARCH",
};
```
The you'll be able to enable this debugging mode in the CLI with:
```
set debug_mode = RESEARCH
```

Finally you need to add the logging statements to the code to record the required values. You can do this with:
```
DEBUG_SET(DEBUG_RESEARCH, 0, value1);
DEBUG_SET(DEBUG_RESEARCH, 1, value2);
DEBUG_SET(DEBUG_RESEARCH, 2, value3);
DEBUG_SET(DEBUG_RESEARCH, 3, value4);
```
where the values are `int16` data types. Then these values will be recorded in the blackbox log and visible in the debug traces.

mujahid-abdulrahim, 2019-01-06:
Your comment was extraordinarily specific and helpful.  Thanks for taking the time to respond with such detail. 

I implemented and enabled the debug mode logging and ran 10 flight tests of varying configuration.  I generated a frequency sweep (chirp) signal within pid.c and injected that as a disturbance into the roll, pitch, and yaw PID sum outputs (as three separate tests).  I ran that test at PID loop rates from ~250 Hz to 8 kHz to see how the gyro and PID rates affect the stability margins (gain and phase) and response frequencies.  

I ran all these tests a few hours before getting on an airplane to present the research results - so again, thank you very much for your help with this. 

I'm still processing the results, but I'll follow up on issue #7327 with the data.

closed 2019-01-06 by mujahid-abdulrahim