betaflight/betaflight #7335: GPS of switch

Gaz-B, 2019-01-04:
Hi I believe I'm correct that GPS can't be switched off without connecting to a PC. I found today flying in a bando trying to get enough signal to allow first arming is a bit of an issue having the GPS function switchable in the modes tab would be most helpful.

IllusionFpv, 2019-01-05:
Hi, check this https://github.com/betaflight/betaflight/pull/7320

Gaz-B, 2019-01-05:
That looks like it might be a work around to what I want. If flying in buildings with partial gaps in roofs I really want a complete off switch no danger of a failsafe trying rescue etc. At the moment my best option seems to be an OTG cable and speedyBee   to switch it off.

ijustwannafly, 2019-01-06:
This would be hard to accomodate IMO.

Turning off gps reverts the failsafe to drop if you set it to gps-rescue.  This means you have to reset the gps failsafe if you had turned off gps and now want to use it.

Setting min sats to 0 would allow instant arming but if the gps still didnt get a lock when you armed, what home position is recorded, and where would it go if you failsafed inside the building?

Gps-rescue IMO is kind of an all or none kind of deal.  Use gps all the time for gps rescue or dont use gps at all for gps rescue.  There's no in between.  You can still use gps for the data but again if you want to use the home arrow function you need (not should, need) to wait for a gps lock.

Gaz-B, 2019-01-08:
I'm hoping setting rescue will be fully incorporated into the next configurator so it can be switched on off at will and not via CLI as if I'm flying through trees I wan't it to drop if above them I wan't rescue. Having it on a switch I'm sure will be available at some point.

pkruger, 2019-01-25:
A simple option just to turn it off (and keep it off) will also be useful.  If you have GPS on your racing quad this can cause issues with arming, if it is sitting on the starting block and doesn't have enough satellites.

closed 2019-03-03 by stale[bot]