betaflight/betaflight #7336: SmartAudio 2.1 compatibility

fujin, 2019-01-04:
**Describe the bug**
SmartAudio 2.1 protocol not working.

**To Reproduce**
1) Use TBS Unify HV Race v2
2) Configure SmartAudio
3) Does not work

**Expected behavior**
SmartAudio 2.1 should work.

**Flight controller configuration**
n/a

**Setup / Versions**
 - Flight controller: n/a
 - Other components: TBS Unify HV Race V2
 - SmartAudio 2.1 protocol to spare UART.

**Additional context**
https://github.com/betaflight/betaflight/blob/master/src/main/interface/smartaudio_protocol.c#L39
https://github.com/betaflight/betaflight/blob/master/src/main/interface/smartaudio_protocol.c#L169

wind0r, 2019-01-04:
SettingResponse changed and contains new values:
```typedef struct smartaudioSettingsResponseFramev2_s {
    smartaudioFrameHeader_t header;
    uint8_t channel;
    uint8_t power;
    uint8_t operationMode;
    uint16_t frequency;
    uint8_t dBm;
    uint8_t powerLevelAmount;
    uint8_t dbm1;
    uint8_t dbm2;
    uint8_t dbm3;
    uint8_t dbm4;
    uint8_t crc;
} __attribute__((packed)) smartaudioSettingsResponseFramev2_t;```

Problem here in the futur will be the powerLevelAmount and so the dynamic length of this package.

moregeld, 2019-01-12:
Is this a CLI command?

wind0r, 2019-01-13:
@moregeld how do you think so?
why should there be a cli command for smartaudio 2.1. that does not make any sense.

this just is an issue to keep track

moregeld, 2019-01-13:
Ah okay.  Wasn't too sure if that was a fix or what not. All good. Thanks.  Fingers crossed this gets resolved in 4.0.

wind0r, 2019-01-13:
tbs send samples to fujin. i think he will make it work when he receives them

Hydroculture, 2019-01-15:
Would be so nice to have SA2.1 working. Got two of those little vtxs installed. Looking forward to be able testing it out.

mariniam, 2019-01-19:
Same here, just purchased TBS nano pro 32, cant get SA to work for the life of me. Go into the SA submenu and all I see is stats

dengorman, 2019-01-19:
Any eta?

Jaypeto, 2019-01-23:
FlightOne has already released a fix when will it be available in BetaFlight ?

wind0r, 2019-01-24:
When it is done.
This also is not a "fix".

mariniam, 2019-01-24:
fyi, got this reply from TBS support, might be useful while waiting:

_Betaflight team has not implemented 2.1 SA yet this explains why it's not working. 

The only way to make it work is via Crossfire Rx. Wires it on channel 4 signal on Micro V2/Nano RX on 2.88 firmware. Set SmartAudio on output channel 4 in the Crossfire settings. 

VTX setting is inside Crossfire lua script under Nano RX. 

​If you're using Crossfire 8ch Div rx or Nano Diversity rx it will be wired on channel 3 or 8 on 2.88 firmware_

fujin, 2019-01-25:
See #7461. Still WIP.

localidiot, 2019-01-26:
Thanks for work around. Works on CRAZYBEEF4DX.

closed 2019-02-13 by mikeller

matbhe, 2019-05-05:
#7979  There are probably still a few problems..

PR #7461 (2019-02-13): SmartAudio: Handle SA2.1 GET SETTINGS response.
Thanks to hardware support from TBS, I have been able to establish bare minimum backwards compatibility with SmartAudio 2.0. 

**Tested**:
- [x] Set band
- [x] Set channel
- [x] Set power from Betaflight power table (0,1,2) - 400mW not available without dBm power level code.
- [x] Softserial
- [x] Confirm/deny Hobbywing G2 UART6_TX problem

**Hardware**:
- [x] [TBS Unify PRO32 Nano](https://www.team-blacksheep.com/products/prod:unifypro32_nano)
- [X] [TBS Unify Pro 5G8 HV - RACE 2](https://www.team-blacksheep.com/products/prod:unify_pro_hv_race2_ [... cut]